RIOT-OS/RIOT #19524: sx1262 LoRa chip support

simonescaccia, 2023-04-30:
#### Description
The example provided for using LoRaWAN only supports the sx1272 and sx1276 LoRa chips, also written in RIOT-OS [documentation](https://api.riot-os.org/group__pkg__semtech-loramac.html).

Why not provide support also for the sx1262 LoRa chip? 
Are there existing RIOT-OS solutions to work with LoRaWAN using the sx1262 LoRa chip?  

My current board is the Heltec WiFi LoRa 32 (V3), the new version of the Heltec WiFi LoRa 32 (V2) which has embedded the sx1276 LoRa chip. Could this change in the LoRa chip be a trend also in other manufacturing companies?

aabadie, 2023-04-30:
Thanks for your question. Actually the Semtech Loramac package can be used with sx1272, sx1276, sx1261, sx1262, llcc68 (but not with SF12 for this particular chip). It also works with the stm32wl family of MCUs.
Your problem is just that documentation wasn't updated in #16238.

aabadie, 2023-04-30:
> Could this change in the LoRa chip be a trend also in other manufacturing companies?

The sx127x radios were the original radio manufactured by Semtech. sx126x radios are just more recent and have the same behavior.

simonescaccia, 2023-04-30:
Nice! Thank you for the clarification

PR #19525 (2023-05-02): semtech-loramac: extend list of supported radio with sx1261, sx1262 and sx1268

### Contribution description

Fixes #19524 

The documentation/README update was overlooked in #16238 

### Testing procedure

Just read the modified files.

### Issues/PRs references

#19524, #16238
closed 2023-05-02 by bors[bot]