RIOT-OS/RIOT #19559: MTD: Not working on native

Ollrogge, 2023-05-08:
#### Description

I am trying to implement support for my CTAP implementation for a native target. As there is no flash I tried to use the [_native_flash](https://github.com/RIOT-OS/RIOT/blob/b93328d14b4f8999110b2ad609137f18e365eb9a/cpu/native/periph/flashpage.c#LL31C15-L31C15) array instead, combined with the `mtd_flashpage_driver`. 

This however does not work. The [out_of_bounds](https://github.com/RIOT-OS/RIOT/blob/b93328d14b4f8999110b2ad609137f18e365eb9a/drivers/mtd/mtd.c#L32) checks in `mtd.c` fail and even when uncommenting these I get assertion failures in the [native flashpage driver](https://github.com/RIOT-OS/RIOT/blob/b93328d14b4f8999110b2ad609137f18e365eb9a/cpu/native/periph/flashpage.c#L56).

#### Steps to reproduce the issue

Pushed the changes I made to my [repository](https://github.com/Ollrogge/RIOT/tree/mtd_issues).
To reproduce simply run `tests/sys_fido2_ctap` for NATIVE target.

maribu, 2023-05-22:
@Ollrogge: Did @kfessel answer help you to solve the issue?

Ollrogge, 2023-05-25:
> @Ollrogge: Did @kfessel answer help you to solve the issue?

Yes kind of :) Using a native mtd device which uses a file instead of a buffer on the stack seems to work. Thank you