RIOT-OS/RIOT #19578: example/gnrc_border_router cannot answer after some time

3JIouHoCoK, 2023-05-11:
#### Description

After some time gnrc_border_router stops sending and answering by stdio (ethos). But I can ping this board from the another board.

#### Steps to reproduce the issue
Prepare 2 boards: gnrc_border_router and gnrc_networking (just for "ping" utility)

Connect the BR to PC by ethos. Run from ethos folder: 
`$ sudo ./start_network.sh /dev/ttyUSB0 tap0 bbbb::/64`

After just start ping the gnrc_networking node from linux console. By default 64-bytes ping packet will be sended with 1s delay and 1s timeout. For example: 
`$ ping bbbb::fc9e:48d0:a8de:14f5`

#### Expected results
Ping packets should receive successfully infinite.

#### Actual results
After some packets, in my case from 1000 to 20000, linux console starts write:
```
64 bytes from bbbb::fc9e:48d0:a8de:14f5: icmp_seq=1741 ttl=63 time=123 ms
64 bytes from bbbb::fc9e:48d0:a8de:14f5: icmp_seq=1742 ttl=63 time=124 ms
64 bytes from bbbb::fc9e:48d0:a8de:14f5: icmp_seq=1743 ttl=63 time=125 ms
64 bytes from bbbb::fc9e:48d0:a8de:14f5: icmp_seq=1744 ttl=63 time=1142 ms
From fd00:dead:beef::1 icmp_seq=1793 Destination unreachable: Address unreachable
From fd00:dead:beef::1 icmp_seq=1794 Destination unreachable: Address unreachable
From fd00:dead:beef::1 icmp_seq=1795 Destination unreachable: Address unreachable
From fd00:dead:beef::1 icmp_seq=1796 Destination unreachable: Address unreachable
From fd00:dead:beef::1 icmp_seq=1797 Destination unreachable: Address unreachable
```
Ethos console:
```
uhcp_client(): no reply received
uhcp_client(): no reply received
uhcp_client(): no reply received
uhcp_client(): no reply received
```
Also ethos console doesn't answer on input messages at all. 

But after long time (about 10 hours) BR starts works normally without reseting. 
If BR board will be reseted, it starts works normally.
#### Boards
STM32WLE5CCU6 with LoRa (RAK3172 module) + 802.15.4 HAL driver (SubMAC)
#### Versions

Operating system: Ubuntu 22.04.1 on WSL2 
RIOT OS master 11.05.2023

3JIouHoCoK, 2023-05-11:
I'm not sure but #16398 seems a bit similar

benpicco, 2023-05-11:
Does it work if you switch from ethos to slip? Should work the same, except that you set `UPLINK=slip` in the border router example.

3JIouHoCoK, 2023-05-12:
Hello, @benpicco 

I'd tried run BR on slip and start ping from the linux host, I've got the same issue but after only 32-37 packets. 

If I have try to send ping from slip console after this error, I get:

```
ping bbbb::fc9e:48d0:a8de:14f5
error: packet buffer full
error: packet buffer full
error: packet buffer full
```
<details>
<summary>pktbuf utility:</summary>

```
pktbuf
packet buffer: first byte: 0x20002e10, last byte: 0x20004610 (size: 6144)
  position of last byte used: 6136
=========== chunk   0 (0x20002e10 size:  800) ===========
00000000  00  00  00  00  28  2E  00  20  06  00  00  00  01  00  00  00
00000010  00  00  00  00  00  00  00  00  55  48  43  50  00  40  00  00
00000020  10  2E  00  20  48  2E  00  20  08  00  00  00  01  00  00  00
00000030  04  00  00  00  00  00  00  00  30  39  30  39  00  0E  5D  21
00000040  30  2E  00  20  68  2E  00  20  28  00  00  00  01  00  00  00
00000050  02  00  00  00  00  00  00  00  60  00  00  00  00  0E  11  40
00000060  FE  80  00  00  00  00  00  00  00  00  00  00  00  00  00  02
00000070  FF  15  00  00  00  00  00  00  00  00  00  00  00  00  AB  CD
00000080  A8  2E  00  20  30  30  00  20  28  00  00  00  01  00  00  00
00000090  02  00  00  00  00  00  00  00  00  00  00  00  58  30  00  20
000000A0  40  00  00  00  01  00  00  00  00  D0  48  9E  FE  7A  70  3A
000000B0  00  00  00  00  D8  2E  00  20  08  00  00  00  01  00  00  00
000000C0  03  9E  48  D0  A8  DE  14  F5  85  00  7D  35  00  00  00  00
000000D0  C0  2E  00  20  F8  2E  00  20  28  00  00  00  01  00  00  00
000000E0  02  2F  00  20  40  00  00  00  60  00  00  00  00  08  3A  FF
000000F0  FE  80  00  00  00  00  00  00  00  00  00  00  00  00  00  02
00000100  FF  02  00  00  00  00  00  00  00  00  00  00  00  00  00  02
00000110  85  00  7D  35  00  00  00  00  55  48  43  50  00  40  00  00
00000120  85  00  7D  35  00  00  00  00  A8  2F  00  20  38  31  00  20
00000130  28  00  00  00  01  00  00  00  02  DE  12  64  DA  C5  A4  2B
00000140  00  00  00  00  30  2F  00  20  08  00  00  00  01  00  00  00
00000150  03  80  00  00  00  00  00  00  50  2F  00  20  80  2F  00  20
00000160  28  00  00  00  01  00  00  00  02  BB  00  00  00  00  00  00
00000170  60  00  00  00  00  08  3A  FF  FE  80  00  00  00  00  00  00
00000180  00  00  00  00  00  00  00  02  FF  02  00  00  00  00  00  00
00000190  00  00  00  00  00  00  00  02  00  00  00  00  60  31  00  20
000001A0  40  00  00  00  01  00  00  00  00  D0  48  9E  FE  7A  70  3A
000001B0  08  30  00  20  C0  30  00  20  28  00  00  00  01  00  00  00
000001C0  02  9E  48  D0  A8  DE  14  F5  A0  31  00  20  90  32  00  20
000001D0  28  00  00  00  01  00  00  00  02  30  00  20  28  00  00  00
000001E0  B8  31  00  20  50  33  00  20  28  00  00  00  01  00  00  00
000001F0  02  DE  12  64  DA  C5  A4  2B  00  00  00  00  E8  30  00  20
00000200  40  00  00  00  01  00  00  00  00  00  00  00  00  00  00  00
00000210  30  39  30  39  00  0E  5D  21  55  48  43  50  00  40  00  00
00000220  60  00  00  00  00  40  3A  3F  BB  BB  00  00  00  00  00  00
00000230  FC  9E  48  D0  A8  DE  14  F5  FD  EA  DB  EE  00  0F  00  00
00000240  00  00  00  00  00  00  00  01  81  00  60  17  F8  E6  02  2A
00000250  43  AD  5D  64  00  00  00  00  1C  90  0E  00  00  00  00  00
00000260  10  11  12  13  14  15  16  17  18  19  1A  1B  1C  1D  1E  1F
00000270  20  21  22  23  24  25  26  27  28  29  2A  2B  2C  2D  2E  2F
00000280  30  31  32  33  34  35  36  37  00  00  00  00  20  2F  00  20
00000290  08  00  00  00  01  00  00  00  03  DE  12  64  DA  C5  A4  2B
000002A0  30  39  30  39  00  0E  5D  21  55  48  43  50  00  40  00  00
000002B0  60  00  00  00  00  40  3A  3F  BB  BB  00  00  00  00  00  00
000002C0  FC  9E  48  D0  A8  DE  14  F5  FD  EA  DB  EE  00  0F  00  00
000002D0  00  00  00  00  00  00  00  01  81  00  FF  19  F8  E6  02  2D
000002E0  47  AD  5D  64  00  00  00  00  87  8A  00  00  00  00  00  00
000002F0  10  11  12  13  14  15  16  17  18  19  1A  1B  1C  1D  1E  1F
00000300  20  21  22  23  24  25  26  27  28  29  2A  2B  2C  2D  2E  2F
00000310  30  31  32  33  34  35  36  37  30  39  30  39  00  0E  5D  21
[... 325 lines cut ...]
000000A0  00  00  00  00  00  00  AB  CD  00  00  00  00  B8  30  00  20
000000B0  06  00  00  00  01  00  00  00  00  45  00  20  08  00  00  00
000000C0  60  00  00  00  00  40  3A  3F  BB  BB  00  00  00  00  00  00
000000D0  FC  9E  48  D0  A8  DE  14  F5  FD  EA  DB  EE  00  0F  00  00
000000E0  00  00  00  00  00  00  00  01  81  00  D7  88  F8  E6  02  4F
000000F0  69  AD  5D  64  00  00  00  00  88  F9  04  00  00  00  00  00
00000100  10  11  12  13  14  15  16  17  18  19  1A  1B  1C  1D  1E  1F
00000110  20  21  22  23  24  25  26  27  28  29  2A  2B  2C  2D  2E  2F
00000120  30  31  32  33  34  35  36  37  B0  44  00  20  28  31  00  20
00000130  08  00  00  00  01  00  00  00  04  00  00  00  00  00  00  01
00000140  30  45  00  20  60  45  00  20  28  00  00  00  01  00  00  00
00000150  02  45  00  20  10  00  00  00  60  00  00  00  00  0E  11  40
00000160  FE  80  00  00  00  00  00  00  00  00  00  00  00  00  00  02
00000170  FF  15  00  00  00  00  00  00  00  00  00  00  00  00  AB  CD
00000180  60  00  00  00  00  40  3A  3F  BB  BB  00  00  00  00  00  00
00000190  FC  9E  48  D0  A8  DE  14  F5  FD  EA  DB  EE  00  0F  00  00
000001A0  00  00  00  00  00  00  00  01  81  00  BD  7B  F8  E6  02  51
000001B0  6B  AD  5D  64  00  00  00  00  9F  04  06  00  00  00  00  00
000001C0  10  11  12  13  14  15  16  17  18  19  1A  1B  1C  1D  1E  1F
000001D0  20  21  22  23  24  25  26  27  28  29  2A  2B  2C  2D  2E  2F
000001E0  30  31  32  33  34  35  36  37
~ unused: 0x200045f0 (next: (nil), size:   32) ~
```

</details>

benpicco, 2023-05-12:
Ah this is unfortunately a known issue - the subMAC depends on `gnrc_netif_pktq` which leaks memory #17924

3JIouHoCoK, 2023-05-12:
This issue with memory leaks only occurs with SLIP. 

With ethos (after the issue is occured) I can ping BR from the remote node => pktbuf is not full.

benpicco, 2023-05-12:
That's interesting - does #18066 make a difference here?

3JIouHoCoK, 2023-05-15:
hmm, with `PREFIX_CONF=dhcpv6` and /48 prefix this issue doesn't occur at all! 

```
^C--- bbbb::2:fc9e:48d0:a8de:14f5 ping statistics ---
65571 packets transmitted, 64559 received, 1.54337% packet loss, time 8096101ms
rtt min/avg/max/mdev = 120.116/123.347/187.100/1.381 ms, pipe 2, ipg/ewma 123.472/123.229 ms

```

3JIouHoCoK, 2023-05-19:
Hello @benpicco 

After a couple of days of testing I found that packet buffer also on ethos can be filled and not cleared. But it happens much more rare than on slip. Random `pktbuf` command after 20 minutes of `ping -A` :

<details>
<summary>pktbuf</summary>

```
> pktbuf
pktbuf
packet buffer: first byte: 0x200037d0, last byte: 0x20004fd0 (size: 6144)
  position of last byte used: 3528
=========== chunk   0 (0x200037d0 size: 2056) ===========
00000000  30  38  00  20  D0  38  00  20  51  00  00  00  01  00  00  00
00000010  00  00  00  00  00  00  00  00  00  00  00  00  00  38  00  20
00000020  14  00  00  00  01  00  00  00  FF  C9  00  59  11  40  FE  80
00000030  06  06  06  00  00  00  00  80  9A  E8  CE  A4  E4  33  A6  63
00000040  C3  8E  5D  0F  00  00  A4  63  E8  37  00  20  A0  38  00  20
00000050  28  00  00  00  01  00  00  00  02  C7  00  01  00  0A  00  03
00000060  18  38  00  20  C8  38  00  20  08  00  00  00  01  00  00  00
00000070  04  8F  04  B2  00  15  5D  C3  00  00  00  00  C8  39  00  20
00000080  40  00  00  00  01  00  00  00  00  20  00  1A  00  19  00  00
00000090  48  38  00  20  A0  39  00  20  28  00  00  00  01  00  00  00
000000A0  02  39  00  20  30  00  00  00  60  38  00  20  90  38  00  20
000000B0  08  00  00  00  01  00  00  00  FF  39  00  20  18  00  00  00
000000C0  00  00  07  00  00  00  00  80  00  00  07  00  00  00  00  80
000000D0  60  0C  05  C9  00  59  11  40  FE  80  00  00  00  00  00  00
000000E0  00  00  00  00  00  00  00  01  FE  80  00  00  00  00  00  00
000000F0  A4  63  C3  FF  FE  8E  5D  0F  02  23  02  22  00  59  8A  B6
00000100  07  0E  EB  C7  00  01  00  0A  00  03  00  01  A6  63  C3  8E
00000110  5D  0F  00  02  00  0E  00  01  00  01  2B  8F  04  B2  00  15
00000120  5D  C3  9F  F6  00  19  00  29  00  19  00  07  00  00  27  10
00000130  00  00  4E  20  00  1A  00  19  00  00  75  30  00  00  9C  40
00000140  40  BB  BB  85  42  BB  FF  00  02  00  00  00  00  00  00  00
00000150  00  00  00  00  B0  16  00  00  00  00  00  00  A8  3A  00  20
00000160  40  00  00  00  01  00  00  00  00  8E  5D  0F  3A  40  FD  00
00000170  28  39  00  20  80  3A  00  20  28  00  00  00  01  00  00  00
00000180  02  3A  00  20  50  00  00  00  40  39  00  20  98  38  00  20
00000190  08  00  00  00  01  00  00  00  FF  39  00  20  18  00  00  00
000001A0  00  00  00  00  20  3B  00  20  40  00  00  00  01  00  00  00
000001B0  00  3A  00  20  20  00  00  00  70  39  00  20  F8  3A  00  20
000001C0  28  00  00  00  01  00  00  00  02  3A  00  20  08  00  00  00
000001D0  60  0C  7A  6E  00  40  3A  3F  FD  00  DE  AD  BE  EF  00  00
000001E0  00  00  00  00  00  00  00  01  BB  BB  85  42  BB  FF  00  02
000001F0  B8  5C  1A  25  64  CB  D6  F9  80  00  E1  15  6B  03  1C  D9
00000200  7D  F4  66  64  00  00  00  00  C4  80  09  00  00  00  00  00
00000210  10  11  12  13  14  15  16  17  18  19  1A  1B  1C  1D  1E  1F
00000220  20  21  22  23  24  25  26  27  28  29  2A  2B  2C  2D  2E  2F
00000230  30  31  32  33  34  35  36  37  88  39  00  20  20  3A  00  20
00000240  08  00  00  00  01  00  00  00  FF  6E  00  40  3A  40  FD  00
00000250  00  00  07  00  00  00  00  80  00  00  00  00  00  3C  00  20
00000260  40  00  00  00  01  00  00  00  00  3A  00  20  48  00  00  00
00000270  28  3A  00  20  D8  3B  00  20  28  00  00  00  01  00  00  00
00000280  02  00  00  00  00  00  10  11  40  3A  00  20  70  3A  00  20
00000290  08  00  00  00  01  00  00  00  FF  3A  00  20  18  00  00  00
000002A0  00  00  07  00  00  00  00  80  00  00  07  00  00  00  00  80
000002B0  60  0C  7A  6E  00  40  3A  3F  FD  00  DE  AD  BE  EF  00  00
000002C0  00  00  00  00  00  00  00  01  BB  BB  85  42  BB  FF  00  02
000002D0  B8  5C  1A  25  64  CB  D6  F9  80  00  24  33  6B  03  1C  DA
000002E0  7D  F4  66  64  00  00  00  00  7F  62  0B  00  00  00  00  00
000002F0  10  11  12  13  14  15  16  17  18  19  1A  1B  1C  1D  1E  1F
00000300  20  21  22  23  24  25  26  27  28  29  2A  2B  2C  2D  2E  2F
00000310  30  31  32  33  34  35  36  37  00  00  07  00  00  00  00  80
00000320  00  00  07  00  00  00  00  80  60  0C  7A  6E  00  40  3A  3F
00000330  FD  00  DE  AD  BE  EF  00  00  00  00  00  00  00  00  00  01
00000340  BB  BB  85  42  BB  FF  00  02  B8  5C  1A  25  64  CB  D6  F9
00000350  80  00  2B  50  6B  03  1C  DB  7D  F4  66  64  00  00  00  00
00000360  76  44  0D  00  00  00  00  00  10  11  12  13  14  15  16  17
00000370  18  19  1A  1B  1C  1D  1E  1F  20  21  22  23  24  25  26  27
00000380  28  29  2A  2B  2C  2D  2E  2F  30  31  32  33  34  35  36  37
[... 87 lines cut ...]
~ unused: 0x200040a8 (next: 0x20004120, size:   16) ~
=========== chunk   2 (0x200040b8 size:  104) ===========
00000000  60  0C  7A  6E  00  40  3A  3F  FD  00  DE  AD  BE  EF  00  00
00000010  00  00  00  00  00  00  00  01  BB  BB  85  42  BB  FF  00  02
00000020  B8  5C  1A  25  64  CB  D6  F9  80  00  DC  7B  6B  03  1C  E3
00000030  7E  F4  66  64  00  00  00  00  C4  10  0D  00  00  00  00  00
00000040  10  11  12  13  14  15  16  17  18  19  1A  1B  1C  1D  1E  1F
00000050  20  21  22  23  24  25  26  27  28  29  2A  2B  2C  2D  2E  2F
00000060  30  31  32  33  34  35  36  37
~ unused: 0x20004120 (next: 0x20004200, size:  120) ~
=========== chunk   3 (0x20004198 size:  104) ===========
00000000  60  0C  7A  6E  00  40  3A  3F  FD  00  DE  AD  BE  EF  00  00
00000010  00  00  00  00  00  00  00  01  BB  BB  85  42  BB  FF  00  02
00000020  B8  5C  1A  25  64  CB  D6  F9  80  00  4A  99  6B  03  1C  E4
00000030  7E  F4  66  64  00  00  00  00  54  F2  0E  00  00  00  00  00
00000040  10  11  12  13  14  15  16  17  18  19  1A  1B  1C  1D  1E  1F
00000050  20  21  22  23  24  25  26  27  28  29  2A  2B  2C  2D  2E  2F
00000060  30  31  32  33  34  35  36  37
~ unused: 0x20004200 (next: (nil), size: 3536) ~
>
```

</details>

Can we say that it's the same problem with `gnrc_netif_pktq` memory leaks?