RIOT-OS/RIOT #19788: RISC-V toolchain detection delays all makefile targets by several seconds

nmeum, 2023-07-03:
#### Description

For the RISC-V architecture, RIOT's Makefiles test various triplet names. This test is performed every time make is invoked and can take several seconds each time. The code is here:

https://github.com/RIOT-OS/RIOT/blob/e0a81fe945ddc5064b8bff61b50e5a0e06d5a530/makefiles/arch/riscv.inc.mk#L15-L38

#### Steps to reproduce the issue

```
$ make BOARD=hifive1 -C examples/hello-world clean
$ # At this point there are no files to clean anymore in examples/hello-world
$ time make BOARD=hifive1 -C examples/hello-world clean
make: Entering directory '/home/soeren/src/RIOT/examples/hello-world'
make: Leaving directory '/home/soeren/src/RIOT/examples/hello-world'
    0m03.02s real     0m02.30s user     0m00.71s system
```

this 3 second delay happens on every make invocation.

#### Expected results

`make clean` on a clean directory should be instant.

#### Actual results

There is a 3s delay for every make invocation (also build etc).

This is due to the toolchain selection code above compare this with:

```
$ time make TARGET_ARCH_RISCV=riscv-none-elf BOARD=hifive1 -C examples/hello-world/ clean
make: Entering directory '/home/soeren/src/RIOT/examples/hello-world'
make: Leaving directory '/home/soeren/src/RIOT/examples/hello-world'
    0m00.16s real     0m00.10s user     0m00.06s system
```

#### Versions

N/A.

---

CC: @maribu

maribu, 2023-07-03:
For the specific issue just add `export TARGET_ARCH_RISCV=riscv-none-elf` to your `~/.profile` / `~/.bashrc` / `~/.zshrc` or whatever. When you don't specify what toolchain to use, the fallback is auto-detection.

For the general the build system is crazy, slow, and a particularly bad example for technical dept, feature creep, and bloat: Everyone is hoping that we'll do a clean cut at some point.

I'm fully aware that there is no need to detect the toolchain, resolve features and dependencies, and detect whether to use `unzip` or `7z` for unpack`, `curl or `wget` to download etc. when just doing `make clean`... That is more a general issue than specific to the toolchain detection.

nmeum, 2023-07-03:
> For the specific issue just add `export TARGET_ARCH_RISCV=riscv-none-elf` to your `~/.profile` / `~/.bashrc` / `~/.zshrc` or whatever. When you don't specify what toolchain to use, the fallback is auto-detection.

I am, aware that one can do that but if that's the recommended setup then the automatic toolchain finding code should IMHO be disabled and modified to just print an error if `TARGET_ARCH_RISCV` is not set because it is unbearably slow and quite confusing for new users. I was made aware of this by a student who uses RIOT with RISC-V in a WSL environment and on his machine this takes ~10s every time and he didn't manage to figure out why it was that slow. It's simply a bad experience for new users as-is.

> That is more a general issue than specific to the toolchain detection.

The detection of the other tools is nowhere near as slow as the RISC-V toolchain autodetection code.

closed 2023-07-06 by bors[bot]