RIOT-OS/RIOT #19805: Not able to compile RIOT on samr21-xpro boards choosing serial number

LizzieDE110, 2023-07-06:
#### Description
Not able to flash binary on samr21-xpro boards choosing serial number.

#### Steps to reproduce the issue

I am trying to work with 2 samr21-xpro boards, however, I can't flash binary on the board I want when I am choosing a Serial number.

Below, commands I have executed : 
```shell
git clone https://github.com/RIOT-OS/RIOT.git
cd RIOT/
git checkout 2023.04-branch
make list-ttys 
```
This last command shows the 2 boards I use with their Serial number. 
To show you the bug, I used the hello-world example : 
```shell
cd examples/hello-world
make flash term BOARD=samr21-xpro SERIAL=ATML2127031800003035 
```
It respond : 
```shell 
/home/dev/RIOT/dist/tools/usb-serial/find-tty.sh: ligne 10: ./ttys.py: Aucun fichier ou dossier de ce type
/home/dev/RIOT/makefiles/boards/sam0.inc.mk:11: *** Did not find a device with serial ATML2127031800003035. Arrêt.
```
However, without Serial parameter, RIOT works as expected : 
```shell 
make flash term BOARD=samr21-xpro
```

#### Expected results

I can choose which board I would like to flash using serial number.
#### Actual results

When I try to specify serial number, it respond : 
```shell 
/home/dev/RIOT/dist/tools/usb-serial/find-tty.sh: ligne 10: ./ttys.py: Aucun fichier ou dossier de ce type
/home/dev/RIOT/makefiles/boards/sam0.inc.mk:11: *** Did not find a device with serial ATML2127031800003035. Arrêt.
```
#### Versions

OS : Ubuntu 22.04.2 LST
Build environment : 
    - native gcc: gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
    - arm-none-eabi-gcc: arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)

dylad, 2023-07-11:
Hello @LizzieDE110, thanks for your report. I can confirm the issue on  my side.
I'll provide a fix during the week.

dylad, 2023-07-11:
@LizzieDE110 Note that you can use `DEBUG_ADAPTER_ID` instead of `SERIAL`.
`SERIAL`seems to exist only for backwards compatibility nowadays so if you're fine with switching to `DEBUG_ADAPTER_ID` this would solve your issue.

commit b482a713 (2023-07-13): Merge #19823

19823: dist/tools/usb-serial: call ttys.py with its path r=maribu a=dylad

### Contribution description
This PR provides a fix for #19805
When calling `SERIAL=xxx make flash` the command failed because the script `dist/tools/usb-serial/find-ttys.sh` tries to call `ttys.py` but it doesn't find it. So pass the full path to `ttys.py` to get rid of the problem.

### Testing procedure
Connect at least two boards to your computer
run `make list-ttys ` to get the ID of your boards
Try to flash one of them with (replace SERIAL with the one you get with the previous command):
`BOARD=xxx S [... cut]
commit e3e09f04 (2023-07-14): Merge #19831

19831: dist/tools/usb-serial: call ttys.py with its path [backport 2023.07] r=benpicco a=maribu

# Backport of #19823

### Contribution description
This PR provides a fix for #19805
When calling `SERIAL=xxx make flash` the command failed because the script `dist/tools/usb-serial/find-ttys.sh` tries to call `ttys.py` but it doesn't find it. So pass the full path to `ttys.py` to get rid of the problem.

### Testing procedure
Connect at least two boards to your computer
run `make list-ttys ` to get the ID of your boards
Try to flash one of them with (replace SERIAL with the one you  [... cut]
PR #19831 (2023-07-14): dist/tools/usb-serial: call ttys.py with its path [backport 2023.07]
# Backport of #19823

### Contribution description
This PR provides a fix for #19805
When calling `SERIAL=xxx make flash` the command failed because the script `dist/tools/usb-serial/find-ttys.sh` tries to call `ttys.py` but it doesn't find it. So pass the full path to `ttys.py` to get rid of the problem.

### Testing procedure
Connect at least two boards to your computer
run `make list-ttys ` to get the ID of your boards
Try to flash one of them with (replace SERIAL with the one you get with the previous command):
`BOARD=xxx [... cut]
closed 2023-07-13 by bors[bot]