RIOT-OS/RIOT #19829: Integer overflow in icmpv6 echo packet creation

mchesser, 2023-07-14:
#### Description

It is possible for an integer overflow of `len`/`data_len` to occur as part of `gnrc_icmpv6_echo_send`/`gnrc_icmpv6_echo_build` when computing the total length of the packet including the header i.e. `data_len + sizeof(imcpv6_echo_t)` at [gnrc_icmpv6_echo.c:34](https://github.com/RIOT-OS/RIOT/blob/d7dba6206b783d09e3fef392caa6e3b65c15f00e/sys/net/gnrc/network_layer/icmpv6/echo/gnrc_icmpv6_echo.c#L34)

Depending on the value used this results in either either a null pointer dereference at [gnrc_icmpv6.c:136](https://github.com/RIOT-OS/RIOT/blob/d7dba6206b783d09e3fef392caa6e3b65c15f00e/sys/net/gnrc/network_layer/icmpv6/gnrc_icmpv6.c#L136) (if `len == (SIZE_MAX - 8)`) or a buffer overflow at [gnrc_icmpv6_echo.c:181](https://github.com/RIOT-OS/RIOT/blob/d7dba6206b783d09e3fef392caa6e3b65c15f00e/sys/net/gnrc/network_layer/icmpv6/echo/gnrc_icmpv6_echo.c#L181) (if `(SIZE_MAX - 8) < len <= SIZE_MAX`).

It's probably more an issue of API misuse than anything -- but it is pretty easy to accidentally trigger through the shell interface (see below):

#### Steps to reproduce the issue

It’s pretty easy to trigger in the `gnrc_networking` [example](https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_networking)  because the -s options allows negative numbers. Meaning we can run the following to get a segfault:

```
main(): This is RIOT! (Version: 2023.07-devel-693-g561e1)
RIOT network stack example application
All up, running the shell now
> ping -s -7 ::1
ping -s -7 ::1
Segmentation fault (core dumped)
```

krzysztof-cabaj, 2023-09-27:
I think we could close that issue. Reported bug was fixed by PR #19927 which was successfully merged into RIOT master.

miri64, 2023-09-27:
Yepp, the trick to auto-close is to add „Fixes #<issue number>“ to the OP or commit message btw. :-)

closed 2023-09-27 by miri64