RIOT-OS/RIOT #19905: Support for Riotee boards

chrysn, 2023-08-25:
#### Description

The Riotee module is an SMD module housing both an nRF52840 and an MSP430FR5962, which the original firmware only uses for "suspend-to-disk". It features all the hardware needed to run on energy harvesting: an MPP tracker, voltage regulators, and voltage monitoring to initiate suspend-to-disk.

As a full board it is sold as the "Riotee board", which adds a USB RP2040 CMSIS-DAP debugger (with some custom extras to switch around the power supply) and feather-style pins, and has wings for sensors, capacitors and solar panels.

#### Notes

* The nRF can be accessed either through openocd or through pyocd. Their riotee-probe tool provides options to write the MSP's firmware, and given it's a thin tool on top of pyocd, I suspect that should also give direct MSP430 access.
* For a first working port it should be possible to write software on the nRF52 alone -- it'll just brown-out when its time comes, and boot up again. We'll see how good our power management really is.
* Using the MSP's original firmware should just work as persistent RAM, possibly with some protocol reverse engineering from their code.
* Porting to the MSP should be possible independently, and will just require shutting down the nRF. I have zero experience with that platform. If it gets ported, ideally the same application (like reading sensors) should work on either chip transparently. (Obviously this one has no radio, and different sizes and speeds).
* Running RIOT on both at the same time should be possible. Until we have very flexible multicore message passing, the applications would need to treat the SPI between them as a custom bus.

#### Quickstart

You can already play around with it by running `USEMODULE+=stdio_rtt make -C examples/nimble_heart_rate_sensor BOARD=nrf52840dongle PROGRAMMER=openocd OPENOCD_DEBUG_ADAPTER=dap all flash term`. It may be necessary to use the provided riotee-probe software to `riotee-probe target-power --on` before, as that example probably exceeds the solar power budget.

See screenshots at https://chaos.social/@chrysn/110950515515409728.

#### Useful links

Full board description on their crowdfunding campaign at https://www.crowdsupply.com/nessie-circuits/riotee, docs at https://www.riotee.nessie-circuits.de/. Software and hardware (with the notable exception of the module itself) are open at https://github.com/NessieCircuits/. The riotee-probe tool can be installed through `pip install riotee-probe` (as always, I recommend doing that in a venv).

---

CC'ing @geissdoerfer who is behind the board in case you're interested in keeping an eye on this. If not: Sorry for the noise, I trust you'll find the unsubscribe button.

geissdoerfer, 2023-08-27:
Great to see first steps toward running Riot-OS on Riotee!
> 
>     * I didn't get the UART pass-through to work by setting the right pins. Then again, I'm not a big fan of serial debug anyway -- and stdio_rtt works just as expected.
Did you have difficulties configuring UART on Riot-OS side or is there anything that wasn't quite clear on Riotee side?
> 
>     * The nRF can be accessed either through openocd or through pyocd. Their riotee-probe tool provides options to write the MSP's firmware, and given it's a thin tool on top of pyocd, I suspect that should also give direct MSP430 access.
> 
With *direct access*, do you mean proper debugging with a tool like gdb? That would indeed be possible by just adding a few more spy-bi-wire primitives in the Riotee Probe Firmware and then implementing a probe in [msp-debug](https://github.com/dlbeer/mspdebug), for example.

> 
> See screenshots at https://chaos.social/@chrysn/110950515515409728.

The link seems broken.

chrysn, 2023-08-27:
> Did you have difficulties configuring UART on Riot-OS side or is there anything that wasn't quite clear on Riotee side?

Frankly, I don't know. I set up the UART to pins D0 / P0.21 and D1 / P0.8, connected to the ttyACM0 UART, it didn't work, and then I thought twice about that I don't really like those anyway (I'm using the thing that is an actual debugger that auto-configures well and has pretty direct RAM access, so why bother going through a physical UART that I'd have to set matching baud rates on and pray it never overflows, when RTT is just picking the right module away?), so I switched to the simpler thing.

> With direct access, do you mean proper debugging with a tool like gdb?

Yes -- but that works pretty fine out of the box, as both OpenOCD and pyOCD give gdb access, so `make debug` in RIOT dropped me into a debugger shell.

> Link broken

No clue how that happened -- https://chaos.social/@chrysn/110950792138810265 is the right one, original post is fixed.

geissdoerfer, 2023-08-28:
> Frankly, I don't know. I set up the UART to pins D0 / P0.21 and D1 / P0.8, connected to the ttyACM0 UART, it didn't work, and then I thought twice about that I don't really like those anyway (I'm [...]

I see. Having an active SWD session doesn't really go well with intermittency, so I often rely on UART for that scenario. Anyway if you have troubles with the USB-UART bridge, feel free to open an issue.
> 
> > With direct access, do you mean proper debugging with a tool like gdb?
> 
> Yes -- but that works pretty fine out of the box, as both OpenOCD and pyOCD give gdb access, so `make debug` in RIOT dropped me into a debugger shell.

Right, debugging is fully supported for the nRF52. But for the MSP430, debugging is currently not supported. We only provide a limited set of CMSIS-DAP vendor commands that allow uploading firmware to the MSP430.

> 
> > Link broken
> 
> No clue how that happened -- https://chaos.social/@chrysn/110950792138810265 is the right one, original post is fixed.

closed 2026-10-18 by basilfx