RIOT-OS/RIOT #19913: KConfig options from external packages

tvanfossen, 2023-09-05:
#### Description
I use several applications with RIOT, and I am beginning to look at adding Kconfig options to a few custom packages. Each application exists as its own repository and can have an "app.config" file containing selected Kconfig options as a checked in resource.

This is specifically an issue for fresh builds, that might occur within a typical CI/CD loop, build pipeline, etc.

I noticed this case exists for RIOTs LVGL implementation specifically, where I can have a custom app.config that lives with an application. LVGL has its own Kconfig that lives within the LVGL repository itself. In RIOT, the Kconfig options from the LVGL repository are included via osource. This works great to include external Kconfig options, assuming that LVGL package has been populated in RIOTs build directory.

Enabling options to be preserved across different apps while using the same core RIOT package would be greatly useful.

Would it be possible for checkout of packages to occur prior to genconfig.py running? I believe this would resolve the core issue?

#### Steps to reproduce the issue
If I have saved an app.config to use as an artifact to instruct what options to use when building the app via a build pipeline, because the Kconfig for which the app.config has options from (LVGL) has not been checked out by RIOT yet, build errors will occur - this is easily repeatable by generating a config file for an application, deleting the RIOT build directory, and then attempting to rebuild from scratch will throw the error

#### Expected results
I would like to be able to use a pre-generated config file based on selected options in an application

#### Actual results
[image]

Build options which do not exist at genconfig.py time result in errors
#### Versions
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "22.04.3 LTS (Jammy Jellyfish)"
                   Kernel: Linux 6.2.0-31-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: ccache version 4.5.1
                    cmake: cmake version 3.22.1
                 cppcheck: missing
                  doxygen: 1.9.1
                      git: git version 2.34.1
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.12.0+dev-00245-g63f4e7c72 (2023-06-12-08:50)
                   python: missing
                  python2: Python 2.7.18
                  python3: Python 3.10.12
                   flake8: 5.0.4 (mccabe: 0.7.0, pycodestyle: 2.9.1, pyflakes: 2.5.0) CPython 3.10.12 on
               coccinelle: spatch version 1.1.1 compiled with OCaml version 4.13.1

tvanfossen, 2023-09-06:
[image]
[image]

By exporting pkg-prepare from the top level Makefile.include and then calling pkg-prepare ahead the kconfig external module/pkg collection, this allows external packages to be checked out prior to running genconfig.py - in turn this allows for stored app.config options that exist only in external packages to be visible when genconfig creates out.config/autoconf.h

If this solution is acceptable, I can start a PR

MrKevinWeiss, 2023-09-11:
Thanks for the PR, I will comment there.