RIOT-OS/RIOT #20071: [tracking] Bugs found by the peripheral selftest

maribu, 2023-11-09:
This is intended to track bugs found by the peripheral selftest, so that the bugs are not only found but eventually also fixed :)

- [x] Incorrect Arduino Mapping on Nucleo64 boards (see https://github.com/RIOT-OS/RIOT/pull/19935)
- [ ] Incomplete Arduino Mapping on Nucleo64 boards
    - [ ] `nucleo-f070rb`: Missing peripheral configuration for SPI
    - [ ] `nucleo-f302r8`: Missing peripheral configuration for the Arduino SPI bus (the provided bus(es) is not compatible with any Arduino shields, as it is connected to the wrong pins)
    - [ ] `nucleo-l152re`: Missing peripheral configuration for the Arduino SPI bus (the provided bus(es) is not compatible with any Arduino shields, as it is connected to the wrong pins)
- [ ] nRF5x: `uart_poweroff()` is not powering off UART (see https://github.com/RIOT-OS/RIOT/pull/19926)
- [ ] `nucleo-f104rb`: I2C test failing
    - **Beware:** The STM32F1 cannot enable the internal pull on the pins unless in GPIO mode and the selftesting shield has no external pull ups. Connecting an I2C breakout board with external pull ups (e.g. SHT1x / BMx280 / ATC24xxx / ... boards) to the female 4 pin socket J2 (or a resistor) can work around this.
    - Even with the work around in place, the I2C test still fails
- `nucleo-f303re` (or rather STM32F3): SPI clock polarity on idle level not correct
- `nucleo-f446re` (or rather STM32F4): SPI transfer is faster than possible with the requested SPI clock
    - Either the SPI frequency is not honored, or the function returns before the transfer is complete

krzysztof-cabaj, 2023-11-09:
@maribu I could provide configuration for SPI in `nucleo-f070rb` using ONLY datasheet - I do not have access to this board.

krzysztof-cabaj, 2023-11-14:
Hmmm ... I'm looking at pinout of [nucleo-l152](https://doc.riot-os.org/group__boards__nucleo-l152re.html) - and Arduino D11, D12, D13 are connected to PA7, 6 and 5. This is configured in periph_conf.h and is compatible with datasheet (page 40). What I'm missing?

maribu, 2023-11-14:
> Hmmm ... I'm looking at pinout of [nucleo-l152](https://doc.riot-os.org/group__boards__nucleo-l152re.html) - and Arduino D11, D12, D13 are connected to PA7, 6 and 5. This is configured in periph_conf.h and is compatible with datasheet (page 40). What I'm missing?

Indeed, that looks all fine.

I guess I confused it with `nucleo-l433rc` which indeed has configured the incorrect bus. I'll update the tracking accordingly.

maribu, 2023-11-15:
So that I have this handy later on, I paste the crucial part of the test output on the `nucleo-f446re` here:

```
Starting test for GPIO-IRQ (both-edges) at tests/periph/selftest_shield/main.c:588
FAILURE in tests/periph/selftest_shield/main.c:596
FAILURE in tests/periph/selftest_shield/main.c:601
FAILURE in tests/periph/selftest_shield/main.c:606
FAILURE in tests/periph/selftest_shield/main.c:611
FAILURE in tests/periph/selftest_shield/main.c:616
FAILURE in tests/periph/selftest_shield/main.c:628
FAILURE in tests/periph/selftest_shield/main.c:632
FAILURE in tests/periph/selftest_shield/main.c:637
FAILURE in tests/periph/selftest_shield/main.c:642
[FAILED]
```

Note: This failure is sporadic.