RIOT-OS/RIOT #20093: Bug in drivers/lpsxxx sensor

mchalko, 2023-11-20:
#### Description

Readings of temperature on LPSxxx sensor is wrong

#### Steps to reproduce the issue

Use an example to temperature of the sensor using lps331

#### Expected results

Normal temperature

#### Actual results

Wrong temperature

#### Versions

Bug introduces 6 months ago in a commit

----

The driver is trying to read 16 bit signed value into 32 bit signed integer, without doing sign extension. That's why the readings are corrupted. Also it would be a good idea to do a sw reset of the sensor during init(), so all the registers have default configuration.

KRVPerera, 2023-11-20:
I think just using int16_t here instead of int32_t fixes this.

https://github.com/RIOT-OS/RIOT/blob/e51dc0937d2bd550152fedcc46794c3de2afb714/drivers/lpsxxx/lpsxxx.c#L134

KRVPerera, 2023-11-20:
I also believe we need to change like

```diff
index 7f84c4fd98..3c7aa13347 100644
--- a/drivers/lpsxxx/lpsxxx.c
+++ b/drivers/lpsxxx/lpsxxx.c
@@ -131,8 +131,8 @@ int lpsxxx_init(lpsxxx_t *dev, const lpsxxx_params_t * params)
 int lpsxxx_read_temp(const lpsxxx_t *dev, int16_t *temp)
 {
     uint8_t tmp;
-    int32_t val = 0;
-    uint16_t res = TEMP_BASE;      /* reference value -> see datasheet */
+    int16_t val = 0;
+    int16_t res = TEMP_BASE;      /* reference value -> see datasheet */
 
     i2c_acquire(DEV_I2C);
     if (i2c_read_reg(DEV_I2C, DEV_ADDR, LPSXXX_REG_TEMP_OUT_L, &tmp, 0) < 0) {
@@ -150,7 +150,7 @@ int lpsxxx_read_temp(const lpsxxx_t *dev, int16_t *temp)
     i2c_release(DEV_I2C);
     val |= ((uint16_t)tmp << 8);
 
-    DEBUG("[lpsxxx] read_temp: raw data %08" PRIx32 "\n", val);
+    DEBUG("[lpsxxx] read_temp: raw data %08" PRIx16 "\n", val);
 
     /* convert val to c°C */
     val *= 100;
```

closed 2023-11-27 by benpicco