RIOT-OS/RIOT #20127: sys/hashes: SHA-384, SHA-512/224 and SHA-512/256 support

mguetschow, 2023-11-30:
As a follow-up for #19969, implement the three FIPS specified truncated versions of SHA-512 in `sys/hashes`. For unified access, add frontend support for the PSA Crypto API, too.

Teufelchen1, 2024-03-18:
Could you give a short status update? The soft feature freeze is on the 25th ;)

mguetschow, 2024-03-18:
Sure, I have an almost finished implementation at https://github.com/mguetschow/RIOT/tree/hashes-sha512-truncated including the necessary PSA Crypto glue code. Still outstanding:
- SHA-384 seems to fail on the nrf52840dk cryptocell for some reason, although it should be supported according to the cryptocell API. This needs to be further investigated.
- The PSA compilation tests in `tests/sys/psa_crypto_*` should be adapted to the change in `examples/psa_crypto` (they use a symlinked `main.c`)

If you wanted, I could split this in two PRs, one that adds truncated SHA512 support to `sys/hashes` and a second one that adds support for it to PSA crypto. The first one would be ready to be merged, while the second one would require some more work.

Teufelchen1, 2024-03-18:
> If you wanted, ..

I think that's up to you!

> I could split this in two PRs

The linked branch has 4 commits with around ~1800 edits. Could be enough to justify the overhead of splitting it in two PRs.

> one that adds truncated SHA512 support to sys/hashes and a second one that adds support for it to PSA crypto

Both are probably minor enough to still get them in during soft-freeze :>

Teufelchen1, 2024-04-29:
Everything is checked off! :tada:

Can this issue be closed?

mguetschow, 2024-04-29:
I would keep it open until #20598 is merged, which is currently hold back by #20545. The former should close the issue automatically as soon as it is merged.

PR #20598 (2024-05-17): sys/psa_crypto: SHA-{384,512/{224,256}}
### Contribution description

Since #20484, RIOT contains software-implementations for the most commonly used truncated versions of SHA-512. This PR adds glue code to enable usage via the PSA Crypto API.

Sidenote: Although the [nRF5 SDK](https://infocenter.nordicsemi.com/topic/sdk_nrf5_v17.1.0/group__crys__hash.html) documents support for SHA-384, it appears not to be supported on the [nrf52840 CRYPTOCELL](https://infocenter.nordicsemi.com/topic/ps_nrf52840/cryptocell.html), as it returns `CRYS_HASH_ILLEGAL_OPERATION_MODE_ERROR`. This is also confirmed  [... cut]
closed 2024-05-17 by mguetschow