RIOT-OS/RIOT #20164: [Rust]: Missing `sock_udp_ep_t` during compiling `riot-wrappers`

NguyenLe1605, 2023-12-11:
I am trying to write a rust module inside RIOT, however, I keep getting the bug of missing `sock_udp_ep_t`. I had tried various gnrc module and sock networking module, but still not figure out how can I get this to compile.
My  `Makefile.include` and output of `make all term` looks like below
```Makefile
USEMODULE += rust_riotmodules
USEMODULE += gnrc_sock_udp
USEMODULE += gnrc_ipv6

FEATURES_REQUIRED += rust_target
```
```bash
"make" -C /media/duich1605/extras/projects/iot/rcoap/RIOT/sys/rust_riotmodules_standalone
    Updating crates.io index
   Compiling riot-sys v0.7.9 (https://github.com/RIOT-OS/rust-riot-sys#b4bd4bde)
   Compiling riot-wrappers v0.8.1 (https://github.com/RIOT-OS/rust-riot-wrappers#fbc6cfe8)
error[E0412]: cannot find type `sock_udp_ep_t` in crate `riot_sys`
     --> /home/duich1605/.cargo/git/checkouts/rust-riot-wrappers-155be8faf0b21fa1/fbc6cfe/src/socket.rs:3:39
      |
3     | pub struct UdpEp(pub(crate) riot_sys::sock_udp_ep_t);
      |                                       ^^^^^^^^^^^^^ help: a struct with a similar name exists: `sock_ip_ep_t`
      |
     ::: /media/duich1605/extras/projects/iot/rcoap/RIOT/examples/hello-rcoap/bin/native/target/i686-unknown-linux-gnu/release/build/riot-sys-df19dc7138e46ef2/out/bindings.rs:25953:1
      |
25953 | pub struct sock_ip_ep_t {
      | ----------------------- similarly named struct `sock_ip_ep_t` defined here

error[E0412]: cannot find type `sock_udp_ep_t` in crate `riot_sys`
     --> /home/duich1605/.cargo/git/checkouts/rust-riot-wrappers-155be8faf0b21fa1/fbc6cfe/src/socket.rs:24:21
      |
24    | impl From<riot_sys::sock_udp_ep_t> for UdpEp {
      |                     ^^^^^^^^^^^^^ help: a struct with a similar name exists: `sock_ip_ep_t`
      |
     ::: /media/duich1605/extras/projects/iot/rcoap/RIOT/examples/hello-rcoap/bin/native/target/i686-unknown-linux-gnu/release/build/riot-sys-df19dc7138e46ef2/out/bindings.rs:25953:1
      |
25953 | pub struct sock_ip_ep_t {
      | ----------------------- similarly named struct `sock_ip_ep_t` defined here

error[E0412]: cannot find type `sock_udp_ep_t` in crate `riot_sys`
     --> /home/duich1605/.cargo/git/checkouts/rust-riot-wrappers-155be8faf0b21fa1/fbc6cfe/src/socket.rs:25:27
      |
25    |     fn from(ep: riot_sys::sock_udp_ep_t) -> Self {
      |                           ^^^^^^^^^^^^^ help: a struct with a similar name exists: `sock_ip_ep_t`
      |
     ::: /media/duich1605/extras/projects/iot/rcoap/RIOT/examples/hello-rcoap/bin/native/target/i686-unknown-linux-gnu/release/build/riot-sys-df19dc7138e46ef2/out/bindings.rs:25953:1
      |
25953 | pub struct sock_ip_ep_t {
      | ----------------------- similarly named struct `sock_ip_ep_t` defined here

error[E0412]: cannot find type `sock_udp_ep_t` in module `riot_sys::inline`
   --> /home/duich1605/.cargo/git/checkouts/rust-riot-wrappers-155be8faf0b21fa1/fbc6cfe/src/socket.rs:30:29
    |
30  | impl From<riot_sys::inline::sock_udp_ep_t> for UdpEp {
    |                             ^^^^^^^^^^^^^ help: a struct with a similar name exists: `sock_ip_ep_t`
    |
   ::: /media/duich1605/extras/projects/iot/rcoap/RIOT/examples/hello-rcoap/bin/native/target/i686-unknown-linux-gnu/release/build/riot-sys-df19dc7138e46ef2/out/riot_c2rust_replaced.rs:950:1
    |
950 | pub struct sock_ip_ep_t {
    | ----------------------- similarly named struct `sock_ip_ep_t` defined here

error[E0412]: cannot find type `sock_udp_ep_t` in module `riot_sys::inline`
   --> /home/duich1605/.cargo/git/checkouts/rust-riot-wrappers-155be8faf0b21fa1/fbc6cfe/src/socket.rs:31:35
    |
31  |     fn from(ep: riot_sys::inline::sock_udp_ep_t) -> Self {
    |                                   ^^^^^^^^^^^^^ help: a struct with a similar name exists: `sock_ip_ep_t`
    |
   ::: /media/duich1605/extras/projects/iot/rcoap/RIOT/examples/hello-rcoap/bin/native/target/i686-unknown-linux-gnu/release/build/riot-sys-df19dc7138e46ef2/out/riot_c2rust_replaced.rs:950:1
    |
950 | pub struct sock_ip_ep_t {
    | ----------------------- similarly named struct `sock_ip_ep_t` defined here

error[E0412]: cannot find type `sock_udp_ep_t` in crate `riot_sys`
[... 112 lines cut ...]
9 |         unsafe { riot_sys::macro_SOCK_IPV6_EP_ANY().into() }
  |                                                     ^^^^ the trait `From<riot_sys::inline::sock_ip_ep_t>` is not implemented for `UdpEp`
  |
  = help: the trait `From<{type error}>` is implemented for `UdpEp`
  = note: required for `riot_sys::inline::sock_ip_ep_t` to implement `Into<UdpEp>`

error[E0277]: the trait bound `UdpEp: From<riot_sys::inline::sock_ip_ep_t>` is not satisfied
  --> /home/duich1605/.cargo/git/checkouts/rust-riot-wrappers-155be8faf0b21fa1/fbc6cfe/src/socket.rs:15:53
   |
15 |         unsafe { riot_sys::macro_SOCK_IPV4_EP_ANY().into() }
   |                                                     ^^^^ the trait `From<riot_sys::inline::sock_ip_ep_t>` is not implemented for `UdpEp`
   |
   = help: the trait `From<{type error}>` is implemented for `UdpEp`
   = note: required for `riot_sys::inline::sock_ip_ep_t` to implement `Into<UdpEp>`

Some errors have detailed explanations: E0277, E0412.
For more information about an error, try `rustc --explain E0277`.
error: could not compile `riot-wrappers` (lib) due to 17 previous errors
warning: build failed, waiting for other jobs to finish...
make[3]: *** [/media/duich1605/extras/projects/iot/rcoap/RIOT/makefiles/cargo-targets.inc.mk:56: /media/duich1605/extras/projects/iot/rcoap/RIOT/examples/hello-rcoap/bin/native/target/i686-unknown-linux-gnu/release/librust_riotmodules_standalone.a] Error 101
make[2]: *** [/media/duich1605/extras/projects/iot/rcoap/RIOT/Makefile.base:31: ALL--/media/duich1605/extras/projects/iot/rcoap/RIOT/sys/rust_riotmodules_standalone] Error 2
make[1]: *** [/media/duich1605/extras/projects/iot/rcoap/RIOT/Makefile.base:31: ALL--/media/duich1605/extras/projects/iot/rcoap/RIOT/sys] Error 2
make: *** [/media/duich1605/extras/projects/iot/rcoap/RIOT/examples/hello-rcoap/../../Makefile.include:760: application_hello-world.module] Error 2
```
```

OlegHahm, 2023-12-13:
Could you provide a pointer to your repository to reproduce this?

NguyenLe1605, 2023-12-14:
Here is the link for my applications: [examples](https://github.com/NguyenLe1605/RIOT/tree/rcoap/examples/hello-rcoap). For context, I try to reproduce the COAP source code but in rust. Here are the [include](https://github.com/NguyenLe1605/RIOT/blob/rcoap/sys/include/rcoap.h) header file, and the for the rust [module](https://github.com/NguyenLe1605/RIOT/tree/rcoap/sys/rcoap).

OlegHahm, 2023-12-16:
You're module is not include the dependency for `sock_udp`.

With
```diff
diff --git a/sys/rcoap/Makefile.dep b/sys/rcoap/Makefile.dep
index 8637ecd735..caf449c9a0 100644
--- a/sys/rcoap/Makefile.dep
+++ b/sys/rcoap/Makefile.dep
@@ -1,4 +1,5 @@
 USEMODULE += rust_riotmodules
+USEMODULE += sock_udp
 USEMODULE += gnrc_sock_udp
 USEMODULE += gnrc_ipv6
 
diff --git a/sys/rcoap/src/lib.rs b/sys/rcoap/src/lib.rs
index 8d86300f87..2bf2b7a982 100644
--- a/sys/rcoap/src/lib.rs
+++ b/sys/rcoap/src/lib.rs
@@ -6,7 +6,7 @@
 
 pub const CONFIG_RCOAP_NOPTS_MAX: u32 = 16;
 
-pub use riot_wrappers::riot_sys::socket_udp_ept_t;
+pub use riot_wrappers::riot_sys::sock_udp_ep_t;
 pub type rcoap_method_flags_t = u16;
 #[repr(C, packed)]
 #[derive(Debug, Copy, Clone)]
 ```
it does build for me.

closed 2023-12-18 by NguyenLe1605