RIOT-OS/RIOT #20233: Add driver for Honeywell ABP2 series pressure sensors

dpcrd, 2024-01-05:
#### Description

I would like to use the ABP2 series pressure sensors. This series covers a wide range of sensors, but they all share a common software interface, leveraging the driver code.

They come with 2 interfaces : either SPI or I2C.

### Useful links

- Product page : https://sps.honeywell.com/us/en/campaigns/sensing-and-iot/abp2-series
- Datasheet : http://prod-edam.honeywell.com/content/dam/honeywell-edam/sps/siot/en-us/products/sensors/pressure-sensors/board-mount-pressure-sensors/basic-abp2-series/documents/sps-siot-abp2-series-datasheet-32350268-en.pdf

dpcrd, 2024-01-05:
As the interface of these sensors is either SPI or I2C, what is the best strategy to handle dependencies ? For example, `FEATURES_REQUIRED += periph_i2c` in Makefile.dep doesn't make sense if you want to use the SPI flavor.

dpproto, 2024-02-02:
I'm taking over this issue from another GitHub account.

I mostly wrote the SPI driver. I need to do some clean-up.

In the meantime, any idea of how to support SPI and I2C in the same driver the clean way ?

maribu, 2024-02-15:
> In the meantime, any idea of how to support SPI and I2C in the same driver the clean way ?

This depends a bit on the use case. If it makes no sense to have multiple instances of the same device with different interfaces attached, one could go do it the same as e.g. the BMP280/BME280 driver did with the `bmx280_spi` module selecting the SPI flavor and the `bmx280_i2c` selecting the I2C flavor of the driver.

If it would make sense to allow using both, IMO the cleanest would be to split the code over three modules, e.g. `abp2_spi` for the SPI communication, `abp2_i2c` for the I2C communication, and `abp2_common` for conversions, logic etc. The idea would be that `abp2_i2c` and `abp2_spi` would be as lean as possible and most of the code would be shared between the two.

dpproto, 2024-02-16:
I got a similar answer from a maintainer of the project in the forum.
https://forum.riot-os.org/t/how-to-write-a-driver-for-a-sensor-with-either-spi-or-i2c-interface/4102/2

closed 2025-05-04 by maribu