RIOT-OS/RIOT #20280: sys/ztimer: callback in the process context

svobedkov, 2024-01-21:
#### Description
`ztimer->callback()` allows you to call functions only in the interrupt context.

Could you make possible 'callback()' in the process context?

chrysn, 2024-01-23:
A callback in the process context would need a caller in the process context.

ztimer callbacks are a relatively low-level (but thus performant) tool; they are supposed to send messages, unlock mutexes or otherwise interact with atomic data structures.

What do you want to achieve / what do you need an in-process callback for? Depending on your application, it may be suitable to defer work of your application until it has been unblocked by the timer, or to use the event thread.

benpicco, 2024-01-25:
You might want to look into [timeout events](https://doc.riot-os.org/event_2timeout_8h.html). 
They are executed in the context of the event thread, e.g. 

```C
static void _event_cb(void *ctx)
{
    printf("This is executed in thread %u\n", thread_getpid());
}

int main(void)
{
    static event_timeout_t event_timeout;
    static event_callback_t event_cb = EVENT_CALLBACK_INIT(_event_cb, 0);

    event_timeout_ztimer_init(&event_timeout, ZTIMER_MSEC, EVENT_PRIO_MEDIUM, &event_cb.super);
    event_timeout_set(&event_timeout, 500);
}
```

maribu, 2024-01-29:
timeout events is implementing the requested feature

closed 2024-01-29 by maribu