RIOT-OS/RIOT #20332: Deletion of old Wiki Pages: Meeting notes from 2013/14

Teufelchen1, 2024-02-02:
Hej 🐐 

I propose to delete those notes from the wiki. 

- https://github.com/RIOT-OS/RIOT/wiki/Minutes-from-face-to-face-meeting-@-FU-Berlin-on-August-29,-2013
- https://github.com/RIOT-OS/RIOT/wiki/Minutes-from-face-to-face-meeting-@-FU-Berlin-on-November-12,-2013
- https://github.com/RIOT-OS/RIOT/wiki/Minutes-from-face-to-face-meeting-@-FU-Berlin-on-September-27,-2013
- https://github.com/RIOT-OS/RIOT/wiki/Minutes-from-face-to-face-meeting-@-FU-Berlin-on-February-04,-2014
- https://github.com/RIOT-OS/RIOT/wiki/Virtual-Meeting-on-December-03,-2014
- https://github.com/RIOT-OS/RIOT/wiki/Virtual-Meeting-on-November-19,-2014
- https://github.com/RIOT-OS/RIOT/wiki/Virtual-Meeting-on-October-20,-2014
- https://github.com/RIOT-OS/RIOT/wiki/Virtual-testing-meeting-on-November-19,-2014

kfessel, 2024-02-21:
there is no reason to delete them as the page leading to them documents that this is the old documentation process and newer meeting are documented in the forum 
https://github.com/RIOT-OS/RIOT/wiki/Meetings#virtual-meetings

kfessel, 2024-03-26:
notes from 2013/14 are still documentation of the meeting happening and the meeting results

e.g.: In order to keep it simple, it was decided to use the master branch as the main development branch and create branches for the releases.

is a simple decision that was made in  August 29, 2013 and that is documented in the meeting notes.

> The wiki is full with "dead pages", that is, pages nobody looks at ever

probaly but they also don't hurt anyone who does not look at them especially meeting notes --  they are documentation of the process -- happening in for eg 2013

kfessel, 2024-03-26:
but maybe we we should cleanup that wiki frontpage