RIOT-OS/RIOT #20414: Allow resizing of the side-nav (in Firefox)

Remmirad, 2024-02-22:
#### Description

At the moment it is not possible to resize the side-nav/tree-nav of the RIOT-doxygen website in Firefox (might apply to other browsers). When the tree is expanded this makes reading it a bit cumbersome.  

This is caused by https://github.com/RIOT-OS/RIOT/blob/543c62f4e8ab0761e1f12743b2044ec7e7d251ac/doc/doxygen/src/js/riot-doxy.js#L47 being called even when resizing the side-nav not only the window. Line `64` then always resets the width. Adding `if ($(window).width() == window_before) {return;}` to the beginning of `resize_handler` fixes that. 

Is the side-nav not being resizable a deliberate decision or a bug (in which case I would like to fix it)?

#### Steps to reproduce the issue
Open https://doc.riot-os.org/ and try to resize the side-nav on the left (using firefox)

#### Expected results
Side-nav resizes accordingly

#### Actual results
Side-nav remains in the same width

maribu, 2024-03-27:
I think this is a bug. I can confirm that the proposed fix solves the issue.

Would you mind to open a PR for that?

Remmirad, 2024-03-27:
Thanks for checking this. I'll open a PR.

closed 2024-03-27 by benpicco