RIOT-OS/RIOT #20451: C2Rust failes on native

chrysn, 2024-03-08:
(Issue moved over from forum, see <https://forum.riot-os.org/t/compiling-rust-example-programs-for-native-rust2c-error/4127> for original entry by @SzymonKubica)

#### Description

Hello,

I’m working on a project which ports over a rust-based eBPF VM to run on RIOT. So far I’ve been testing it out on the nucleo-f439zi stm32 board and everything was working fine. I have installed c2rust as suggested from the github repo (not directly from cargo).

The problem is that when I try to compile my project for native, c2rust fails with the following error:

```
Running C2Rust on /home/szymon/Projects/ebpf-on-microcontrollers/mibpf/RIOT/examples/rust-hello-world/bin/native/target/i686-unknown-linux-gnu/release/build/riot-sys-2c2c5674e877dcb9/out/compile_commands.json
  Transpiling riot-c2rust.h
  cargo:warning=C2Rust failed with error code exit status: 101, exiting

  --- stderr
  In file included from riot-c2rust.h:79:
  In file included from /home/szymon/Projects/ebpf-on-microcontrollers/mibpf/RIOT/core/lib/include/rmutex.h:28:
  /home/szymon/Projects/ebpf-on-microcontrollers/mibpf/RIOT/core/include/mutex.h:130:10: fatal error: 'stddef.h' file not found
  #include <stddef.h>
           ^~~~~~~~~~
  1 error generated.
  Error while processing /home/szymon/Projects/ebpf-on-microcontrollers/mibpf/RIOT/examples/rust-hello-world/bin/native/target/i686-unknown-linux-gnu/release/build/riot-sys-2c2c5674e877dcb9/out/riot-c2rust.h.
  warning: Missing child 108724094825960 of node AstNode { tag: TagCallExpr, children: [Some(108724094825960)], loc: SrcSpan { fileid: 110, begin_line: 384, begin_column: 11, end_line: 384, end_column: 33 }, type_id: Some(108724091303168), rvalue: RValue, macro_expansions: [], macro_expansion_text: None, extras: [] }
  Exported Clang AST was invalid. Check warnings above for unimplemented features.
  --> /home/szymon/Projects/ebpf-on-microcontrollers/mibpf/RIOT/sys/include/net/gnrc/netif/internal.h:384:11
   [-Wclang-ast]
  warning: Missing child 108724094826208 of node AstNode { tag: TagCallExpr, children: [Some(108724094826208), Some(108724094826240)], loc: SrcSpan { fileid: 110, begin_line: 387, begin_column: 16, end_line: 387, end_column: 43 }, type_id: Some(108724091303168), rvalue: RValue, macro_expansions: [], macro_expansion_text: None, extras: [] }
  Exported Clang AST was invalid. Check warnings above for unimplemented features.
```

The same issue is present when I try to compile any of the rust example programs. I suspect it has something to do with my OS as I’m using arch linux and when installing c2rust I needed to explicitly specify the llvm installation dir.

Has anyone seen a similar issue before? Thank you

chrysn, 2024-03-08:
Haven't seen this yet; AFAIK @maribu is the RIOT developer who has the most experience with distros that use musl libc.

It may also be that this is an issue with the include paths and how they are adjusted between GCC and LLVM.

Please try both these:
* Build with BUILD_IN_DOCKER=1
* Build with TOOLCHAIN=llvm

SzymonKubica, 2024-03-08:
I have already tried setting TOOLCHAIN=llvm, however this hasn't changed anything. 
To give you some additional context, in order to get c2rust to even compile during installing, I had to explicitly set
```
LLVM_TOOLCHAIN_DIR=/usr/lib
```
And then compile c2rust using the version from the git repo. 
It seems like the multilib setup on arch isn't ideal, or there is some problem with my configuration.
I don't have access to my main machine right now, once I have it, I will test the build in docker option

As for my setup, I'm pretty sure I have all versions of clang and llvm from 14 to 17 installed. Not sure why the compilation process isn't able to find any of them.

SzymonKubica, 2024-03-08:
Speaking of compile_commands.json, it could have something to do with this. 
I had some issues with riot-wrappers and riot-sys after syncing to the latest version of RIOT master. 
It kept complaining that I need to set RIOT_USEMODULE, even though the build.rs of  the latest riot sys should not fail at that point. 
In order to fix that issue, I had to add the RIOT_USEMODULE line back to this file:
https://github.com/RIOT-OS/RIOT/commit/9a59d59bd26b0376aa69a1a0c2dc2760aaa460e4

SzymonKubica, 2024-03-08:
I will look into the above once I'm back and figure out if it is causing the issue.
Although I did clone a fresh RIOT repo as well and just compiled the rust-hello-word example there and it failed with the
same exception.

SzymonKubica, 2024-03-08:
Ok so I have tried running with BUILD_IN_DOCKER=1 however there seems to be an issue with the latest version of riot-wrappers requiring a newer version of rustc (1.75) than the one present in the docker image. Do you know how to deal with this issue?

chrysn, 2024-03-08:
docker pull riotdocker

should fix that detail easily.

SzymonKubica, 2024-03-08:
Thank you for suggesting this, I did that, however it didn't seem to affect the actual image used for docker build:
```
Launching build container using image "docker.io/riot/riotbuild:latest".
```
I have pulled this image as 'riotdocker' wasn't found: `docker pull kaspar030/riotdocker`. Is this the right one? I assume I would have to override it in the config somewhere. I looked through the RIOT build in docker docs, however there doesn't seem to be an easy way to change the image.

SzymonKubica, 2024-03-08:
Ok, pulling the last image did the trick. Now I was able to compile the rust-hello-word example and it did execute correctly.
Thanks a lot for your help. 

I tried using the same setup for my main project, but got the issue where the rust doesn't have the proper toolchain installed:
```
 i686-unknown-linux-gnu
```
It is interesting that this one is targeted as I have that one installed locally. I assume it isn't present in docker. I will investigate that.

maribu, 2024-03-09:
`BUILD_IN_DOCKER=1` uses `docker.io/riot/riotbuild` to build. So you should go for

```
docker pull docker.io/riot/riotbuild
```

to update.

You can use a different container via `DOCKER_IMAGE=<img_name>` (if I recall the name correctly), but I think @chrysn was not suggesting to do that but just recalled the docker image name incorrectly.

chrysn, 2024-03-09:
Sorry, still didn't have time to look into this in full, just dumping anything I have on short notes:

>  think chrysn was not suggesting to do that but just recalled the docker image name incorrectly.

Indeed; I just sent the remark from mobile without looking up the details.

>  It kept complaining that I need to set RIOT_USEMODULE, even though the build.rs of the latest riot sys should not fail at that point.

That should be fixed already because the latest riot-sys has been uploaded to crates.io. RIOT uses riot-sys/-wrappers with a patch.crates-io to go directly to git. Tracked in https://github.com/RIOT-OS/RIOT/issues/20452 for not letting this happen again.

SzymonKubica, 2024-03-09:
Thank you @chrysn for opening the other issue, I will look into this and provide additional context where applicable. 
As for the current state of my build-in-docker. I did manage to pull the correct image and get the rust-hello-world  and rust-gcoap examples to compile. Thanks a lot for helping me fix this issue. My project still doesn't work, but I think the issue lies somewhere deeper (either in the rust toolchain for i686 or c2rust) you can see the issue below  but it's probably unrelated to the original problem:
```
/usr/bin/ld: /data/riotbuild/riotproject/mibpf-server/bin/native/mibpf_server/mibpf_server-fcf3a3ec338093ea.mibpf_server.9ede0b52c9482aa-cgu.0.rcgu.o:(.data.DW.ref.rust_eh_personality[DW.ref.rust_eh_personality]+0x0): undefined reference to `rust_eh_personality'
collect2: error: ld returned 1 exit status

```
I just attached it here in case you have seen it before.
Does this look familiar?

I did some digging and found that this missing eh_personality problem shouldn't be there if Cargo.toml specifies in profile.release `panic="abort"`

A possible workaround that I found online was to define this function manually like so:
```
    #[lang = "eh_personality"]
    extern "C" fn eh_personality() {}
```
The problem is that it requires nightly rust to allow for using language features. Do you know how I can override the toolchain used by the build-in-docker process? I tried using a custom toolchain.toml file, however it then complains that it doesn't have the target installed:
```
rustup target add i686-unknown-linux-gnu
```
Is there a way of adding new targets to the rust toolchain that is present in the docker image?

SzymonKubica, 2024-03-09:
From the RIOT side, my setup works now. Thank you @chrysn and @maribu for helping me out.
Excellent support right here, thank you for all of your contributions to RIOT.  

For anyone reading this issue in the future:

## Problem
RIOT examples of rust programs (rust-hello-world, rust-gcoap) fail to build because of link-time error in c2rust on Arch Linux.
## Solution
Use the BUILD_IN_DOCKER=1 TOOLCHAIN=llvm environment variables when compiling the program
If compilation fails because of an incompatible version of rustc, pull the latest docker image required for the build process:
`docker pull riot/riotbuild`. If after that you get some weird issues with `eh_personality`, read this issue https://github.com/rust-lang/rust/issues/106864 and try solutions suggested there. The thing that worked for me was adding a dummy implementation of that missing function to my main lib.rs file:
```
#[no_mangle]
extern "C" fn rust_eh_personality() {}
```

mcr, 2024-08-16:
The underlying problem is that the LLVM needs some gcc headers.  GCC knows to look in the compiler-specific directories, but LLVM does not.
https://github.com/rust-lang/rust-bindgen/issues/2612 is another example of this problem.
https://clangd.llvm.org/guides/system-headers talks more about this, but despite the test at the end working, my bindgen run does not work yet.

mcr, 2024-08-16:
make BOARD=native64 all
seems to work better, but fails at 

````
/bin/ld: /corp/projects/pandora/riotos/RIOT/examples/rust-gcoap/bin/native64/periph/timer.o: in function `timer_start':
/corp/projects/pandora/riotos/RIOT/cpu/native/periph/timer.c:205: undefined reference to `timer_settime'
/bin/ld: /corp/projects/pandora/riotos/RIOT/examples/rust-gcoap/bin/native64/periph/timer.o: in function `timer_init':
/corp/projects/pandora/riotos/RIOT/cpu/native/periph/timer.c:117: undefined reference to `timer_create'
/bin/ld: /corp/projects/pandora/riotos/RIOT/cpu/native/periph/timer.c:124: undefined reference to `timer_delete'
collect2: error: ld returned 1 exit status
````

I am on an Ubuntu 20.04 LTS platform. (wont move to 22 due to gnome3 garbage)

mcr, 2024-08-16:
````
sudo apt install gcc-multilib
````

got me to the same place as native64 did (ztimer errors) things for me on Ubuntu20, once I realized that I was building in 32-bit mode.

maribu, 2024-08-17:
> make BOARD=native64 all seems to work better, but fails at
>
> ```
> /bin/ld: /corp/projects/pandora/riotos/RIOT/examples/rust-gcoap/bin/native64/periph/timer.o: in function `timer_start': [...]

That looks like an `FEATURES_REQUIRES += periph_timer` is missing in the `Makefile`.

Maybe we should consider a maintainable option to better error messages on modules/features not selected but used.

crasbe, 2026-08-31:
It seems as if this is related to the version of LLVM, because a very similar issue is haunting me in https://github.com/RIOT-OS/riotdocker/pull/286 with LLVM 17.

Following errors occur:

On `native`:
```
buechse@skyleaf:~/RIOTstuff/riot-fresh/RIOT$ BUILD_IN_DOCKER=1 DOCKER_IMAGE=docker.io/library/riotbuild-resolute make -C examples/lang_support/official/rust-hello-world
make: Entering directory '/home/buechse/RIOTstuff/riot-fresh/RIOT/examples/lang_support/official/rust-hello-world'
using BOARD="native64" as "native" on a 64-bit system
Launching build container using image "docker.io/library/riotbuild-resolute".
docker run --rm --tty --user $(id -u):$(id -g) --platform linux/amd64 -v '/home/buechse/RIOTstuff/riot-fresh/RIOT:/data/riotbuild/riotbase:delegated' -v '/home/buechse/.cargo/registry:/data/riotbuild/.cargo/registry:delegated' -v '/home/buechse/.cargo/git:/data/riotbuild/.cargo/git:delegated' -e 'TZ=Europe/Berlin' -e 'RIOTBASE=/data/riotbuild/riotbase' -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' -e 'BUILD_DIR=/data/riotbuild/riotbase/build' -e 'BUILD_IN_DOCKER=0' -e 'RIOTPROJECT=/data/riotbuild/riotbase' -e 'RIOTCPU=/data/riotbuild/riotbase/cpu' -e 'RIOTBOARD=/data/riotbuild/riotbase/boards' -e 'RIOTMAKE=/data/riotbuild/riotbase/makefiles'     -e 'DISABLE_MODULE=' -e 'DEFAULT_MODULE=' -e 'FEATURES_REQUIRED=' -e 'FEATURES_BLACKLIST=' -e 'FEATURES_OPTIONAL=' -e 'USEMODULE=' -e 'USEPKG='  -w '/data/riotbuild/riotbase/examples/lang_support/official/rust-hello-world/' 'docker.io/library/riotbuild-resolute' make
using BOARD="native64" as "native" on a 64-bit system
Building application "hello-world" for "native64" with CPU "native".

"make" -C /data/riotbuild/riotbase/boards compile-commands
"make" -C /data/riotbuild/riotbase/boards/common/init compile-commands
"make" -C /data/riotbuild/riotbase/boards/native64 compile-commands
"make" -C /data/riotbuild/riotbase/boards/common/native compile-commands
"make" -C /data/riotbuild/riotbase/boards/common/native/drivers compile-commands
"make" -C /data/riotbuild/riotbase/core compile-commands
"make" -C /data/riotbuild/riotbase/core/lib compile-commands
"make" -C /data/riotbuild/riotbase/cpu/native compile-commands
"make" -C /data/riotbuild/riotbase/cpu/native/periph compile-commands
"make" -C /data/riotbuild/riotbase/cpu/native/stdio_native compile-commands
"make" -C /data/riotbuild/riotbase/drivers compile-commands
"make" -C /data/riotbuild/riotbase/drivers/periph_common compile-commands
"make" -C /data/riotbuild/riotbase/sys compile-commands
"make" -C /data/riotbuild/riotbase/sys/auto_init compile-commands
"make" -C /data/riotbuild/riotbase/sys/libc compile-commands
"make" -C /data/riotbuild/riotbase/sys/preprocessor compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/stdio compile-commands
   Compiling proc-macro2 v1.0.93
   Compiling unicode-ident v1.0.14
   Compiling memchr v2.7.4
   Compiling glob v0.3.2
   Compiling libc v0.2.169
   Compiling rustix v0.38.43
   Compiling prettyplease v0.2.35
   Compiling linux-raw-sys v0.4.15
   Compiling either v1.13.0
   Compiling cfg-if v1.0.0
   Compiling regex-syntax v0.8.5
   Compiling bitflags v2.7.0
   Compiling serde v1.0.217
   Compiling minimal-lexical v0.2.1
   Compiling serde_json v1.0.135
   Compiling bindgen v0.69.5
   Compiling home v0.5.11
   Compiling ryu v1.0.18
   Compiling itoa v1.0.14
   Compiling log v0.4.22
   Compiling autocfg v1.4.0
   Compiling shlex v1.3.0
   Compiling lazy_static v1.5.0
   Compiling lazycell v1.3.0
   Compiling syn v1.0.109
   Compiling rustc-hash v1.1.0
   Compiling nb v1.1.0
   Compiling byteorder v1.5.0
   Compiling void v1.0.2
   Compiling heapless v0.8.0
   Compiling coap-numbers v0.2.4
   Compiling cty v0.2.2
   Compiling c2rust-asm-casts v0.2.0
   Compiling stable_deref_trait v1.2.0
   Compiling riot-wrappers v0.9.2 (https://github.com/RIOT-OS/rust-riot-wrappers#6258817d)
   Compiling mutex-trait v0.2.0
   Compiling bare-metal v1.0.0
   Compiling hex v0.4.3
[... 580 lines cut ...]
  --- stderr
  In file included from riot-c2rust.h:87:
  /data/riotbuild/riotbase/sys/include/vfs.h:58:10: fatal error: 'sys/statvfs.h' file not found
     58 | #include <sys/statvfs.h> /* for struct statvfs */
        |          ^~~~~~~~~~~~~~~
  1 error generated.
  Error while processing /data/riotbuild/riotbase/examples/lang_support/official/rust-hello-world/bin/hifive1b/target/riscv32imac-unknown-none-elf/release/build/riot-sys-7c79d41e466a0054/out/riot-c2rust.h.

  thread 'main' (342) panicked at c2rust-transpile/src/c_ast/conversion.rs:950:22:
  Type conversion not implemented for TagTypeUnknown expecting 3
  stack backtrace:
     0: __rustc::rust_begin_unwind
     1: core::panicking::panic_fmt
     2: <c2rust_transpile::c_ast::conversion::ConversionContext>::visit_node
     3: <c2rust_transpile::c_ast::conversion::ConversionContext>::new
     4: c2rust_transpile::transpile_single
     5: <core::iter::adapters::map::Map<core::slice::iter::Iter<alloc::rc::Rc<c2rust_transpile::compile_cmds::CompileCmd>>, c2rust_transpile::transpile::{closure#5}> as core::iter::traits::iterator::Iterator>::fold::<(), core::iter::traits::iterator::Iterator::for_each::call<core::result::Result<(std::path::PathBuf, alloc::vec::Vec<(&str, alloc::vec::Vec<&str>)>, indexmap::set::IndexSet<c2rust_transpile::ExternCrate>), ()>, <alloc::vec::Vec<core::result::Result<(std::path::PathBuf, alloc::vec::Vec<(&str, alloc::vec::Vec<&str>)>, indexmap::set: [... cut ...]
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
make: *** [/data/riotbuild/riotbase/makefiles/cargo-targets.inc.mk:63: /data/riotbuild/riotbase/examples/lang_support/official/rust-hello-world/bin/hifive1b/target/riscv32imac-unknown-none-elf/release/librust_hello_world.a] Error 101
make: *** [/home/buechse/RIOTstuff/riot-fresh/RIOT/makefiles/docker.inc.mk:405: ..in-docker-container] Error 2
make: Leaving directory '/home/buechse/RIOTstuff/riot-fresh/RIOT/examples/lang_support/official/rust-hello-world'
```

crasbe, 2026-08-31:
I have no idea what the first issue is about, but the second one seems to be that LLVM can't find `sys/statvfs.h`, which is a very similar issue to the original issue where it couldn't find `stddef.h`.

crasbe, 2026-08-31:
What's also odd is the fact that `riotbase/sys/include/vfs.h` is evaluated in the first place, because as far as I know, that example does not use `vfs`, which is why the `posix_headers` module is not included, which is why `sys/posix/include/sys/statvfs.h` is not included (or rather `RIOTBASE/sys/posix/include` as an include path.

crasbe, 2026-08-31:
Okay, the error on `native` is the same error as documented here https://github.com/RIOT-OS/riotdocker/pull/283

Maybe I did the testing with the newer version of LLVM too? I don't remember 🫠
Otherwise the failing build would've already bitten us in the Murdock runs.

crasbe, 2026-09-09:
With the newer versions, the hifive1b build still fails, but native succeeds:

<details><summary>Full log for the hifive1b:</summary>

```
buechse@skyleaf:~/RIOTstuff/riot-upstream3/RIOT$ BUILD_IN_DOCKER=1 DOCKER_IMAGE=5ebf563ef282 BOARD=hifive1b make -C examples/lang_support/official/rust-hello-world/
make: Entering directory '/home/buechse/RIOTstuff/riot-upstream3/RIOT/examples/lang_support/official/rust-hello-world'
Launching build container using image "5ebf563ef282".
docker run --rm --tty --user $(id -u):$(id -g) --platform linux/amd64 -e 'RUST_BACKTRACE=full' -v '/home/buechse/RIOTstuff/riot-upstream3/RIOT:/data/riotbuild/riotbase:delegated' -v '/home/buechse/.cargo/registry:/data/riotbuild/.cargo/registry:delegated' -v '/home/buechse/.cargo/git:/data/riotbuild/.cargo/git:delegated' -e 'TZ=Europe/Berlin' -e 'RIOTBASE=/data/riotbuild/riotbase' -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' -e 'BUILD_DIR=/data/riotbuild/riotbase/build' -e 'BUILD_IN_DOCKER=0' -e 'RIOTPROJECT=/data/riotbuild/riotbase' -e 'RIOTCPU=/data/riotbuild/riotbase/cpu' -e 'RIOTBOARD=/data/riotbuild/riotbase/boards' -e 'RIOTMAKE=/data/riotbuild/riotbase/makefiles' -v '/home/buechse/.gitcache:/data/riotbuild/gitcache:delegated' -e 'GIT_CACHE_DIR=/data/riotbuild/gitcache'    -e 'BOARD=hifive1b' -e 'DISABLE_MODULE=' -e 'DEFAULT_MODULE=' -e 'FEATURES_REQUIRED=' -e 'FEATURES_BLACKLIST=' -e 'FEATURES_OPTIONAL=' -e 'USEMODULE=' -e 'USEPKG='  -w '/data/riotbuild/riotbase/examples/lang_support/official/rust-hello-world/' '5ebf563ef282' make
Building application "hello-world" for "hifive1b" with CPU "fe310".

"make" -C /data/riotbuild/riotbase/boards compile-commands
"make" -C /data/riotbuild/riotbase/boards/common/init compile-commands
"make" -C /data/riotbuild/riotbase/boards/hifive1b compile-commands
"make" -C /data/riotbuild/riotbase/core compile-commands
"make" -C /data/riotbuild/riotbase/core/lib compile-commands
"make" -C /data/riotbuild/riotbase/cpu/fe310 compile-commands
"make" -C /data/riotbuild/riotbase/cpu/fe310/periph compile-commands
"make" -C /data/riotbuild/riotbase/cpu/fe310/vendor compile-commands
"make" -C /data/riotbuild/riotbase/cpu/riscv_common compile-commands
"make" -C /data/riotbuild/riotbase/cpu/riscv_common/periph compile-commands
"make" -C /data/riotbuild/riotbase/drivers compile-commands
"make" -C /data/riotbuild/riotbase/drivers/periph_common compile-commands
"make" -C /data/riotbuild/riotbase/sys compile-commands
"make" -C /data/riotbuild/riotbase/sys/auto_init compile-commands
"make" -C /data/riotbuild/riotbase/sys/div compile-commands
"make" -C /data/riotbuild/riotbase/sys/libc compile-commands
"make" -C /data/riotbuild/riotbase/sys/malloc_thread_safe compile-commands
"make" -C /data/riotbuild/riotbase/sys/newlib_syscalls_default compile-commands
"make" -C /data/riotbuild/riotbase/sys/preprocessor compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/stdio compile-commands
"make" -C /data/riotbuild/riotbase/sys/stdio_uart compile-commands
   Compiling proc-macro2 v1.0.93
   Compiling unicode-ident v1.0.14
   Compiling memchr v2.7.4
   Compiling glob v0.3.2
   Compiling libc v0.2.169
   Compiling prettyplease v0.2.35
   Compiling serde v1.0.217
   Compiling regex-syntax v0.8.5
   Compiling minimal-lexical v0.2.1
   Compiling cfg-if v1.0.0
   Compiling either v1.13.0
   Compiling bindgen v0.72.0 (https://github.com/chrysn-pull-requests/rust-bindgen?branch=extended-floats#3fe9bdc1)
   Compiling serde_json v1.0.135
   Compiling log v0.4.22
   Compiling bitflags v2.7.0
   Compiling shlex v1.3.0
   Compiling syn v1.0.109
   Compiling itoa v1.0.14
   Compiling ryu v1.0.18
   Compiling rustc-hash v2.1.3
   Compiling autocfg v1.4.0
   Compiling nb v1.1.0
   Compiling byteorder v1.5.0
   Compiling heapless v0.8.0
   Compiling void v1.0.2
   Compiling c2rust-asm-casts v0.2.0
   Compiling riot-wrappers v0.9.2 (https://github.com/RIOT-OS/rust-riot-wrappers#61c794b0)
   Compiling coap-numbers v0.2.4
   Compiling stable_deref_trait v1.2.0
   Compiling mutex-trait v0.2.0
   Compiling rand_core v0.6.4
   Compiling hex v0.4.3
   Compiling bare-metal v1.0.0
   Compiling embedded-hal v1.0.0
   Compiling pin-utils v0.1.0
   Compiling rust_riotmodules v0.1.0 (/data/riotbuild/riotbase/sys/rust_riotmodules)
   Compiling libloading v0.8.6
[... 646 lines cut ...]
    Finished `release` profile [optimized] target(s) in 13.61s
"make" -C /data/riotbuild/riotbase/boards
"make" -C /data/riotbuild/riotbase/boards/common/init
"make" -C /data/riotbuild/riotbase/boards/native64
"make" -C /data/riotbuild/riotbase/boards/common/native
"make" -C /data/riotbuild/riotbase/boards/common/native/drivers
"make" -C /data/riotbuild/riotbase/core
"make" -C /data/riotbuild/riotbase/core/lib
"make" -C /data/riotbuild/riotbase/cpu/native
"make" -C /data/riotbuild/riotbase/cpu/native/periph
"make" -C /data/riotbuild/riotbase/cpu/native/stdio_native
"make" -C /data/riotbuild/riotbase/drivers
"make" -C /data/riotbuild/riotbase/drivers/periph_common
"make" -C /data/riotbuild/riotbase/sys
"make" -C /data/riotbuild/riotbase/sys/auto_init
"make" -C /data/riotbuild/riotbase/sys/libc
"make" -C /data/riotbuild/riotbase/sys/preprocessor
"make" -C /data/riotbuild/riotbase/sys/stdio
/usr/bin/x86_64-linux-gnu-ld.bfd: warning: /data/riotbuild/riotbase/examples/lang_support/official/rust-hello-world/bin/native64/hello-world.elf has a LOAD segment with RWX permissions
   text    data     bss     dec     hex filename
  29168    1096   59160   89424   15d50 /data/riotbuild/riotbase/examples/lang_support/official/rust-hello-world/bin/native64/hello-world.elf
make: Leaving directory '/home/buechse/RIOTstuff/riot-upstream3/RIOT/examples/lang_support/official/rust-hello-world'
```

</details>

crasbe, 2026-09-09:
The compilation of `rust-gcoap` seems to cause a kernel panic for the transpiler:
`native64` is not supported by that example, referring to a bug in the Docker image?

https://github.com/RIOT-OS/RIOT/blob/681273ecd3988d3d61fc06fbab0ecffd075b8d3d/examples/lang_support/official/rust-gcoap/Makefile#L4-L5

When I comment that out, it builds just fine though with #22683 applied and https://github.com/RIOT-OS/riotdocker/pull/286

The run was disabled here: https://github.com/RIOT-OS/RIOT/pull/21531

I'm not quite sure what the issue was there though 🤔 

<details><summary>Full log for hifive1b</summary>

```
buechse@skyleaf:~/RIOTstuff/riot-upstream3/RIOT$ BUILD_IN_DOCKER=1 DOCKER_IMAGE=5ebf563ef282 BOARD=hifive1b make -C examples/lang_support/official/rust-gcoap
make: Entering directory '/home/buechse/RIOTstuff/riot-upstream3/RIOT/examples/lang_support/official/rust-gcoap'
Launching build container using image "5ebf563ef282".
docker run --rm --tty --user $(id -u):$(id -g) --platform linux/amd64 -e 'RUST_BACKTRACE=full' -v '/home/buechse/RIOTstuff/riot-upstream3/RIOT:/data/riotbuild/riotbase:delegated' -v '/home/buechse/.cargo/registry:/data/riotbuild/.cargo/registry:delegated' -v '/home/buechse/.cargo/git:/data/riotbuild/.cargo/git:delegated' -e 'TZ=Europe/Berlin' -e 'RIOTBASE=/data/riotbuild/riotbase' -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' -e 'BUILD_DIR=/data/riotbuild/riotbase/build' -e 'BUILD_IN_DOCKER=0' -e 'RIOTPROJECT=/data/riotbuild/riotbase' -e 'RIOTCPU=/data/riotbuild/riotbase/cpu' -e 'RIOTBOARD=/data/riotbuild/riotbase/boards' -e 'RIOTMAKE=/data/riotbuild/riotbase/makefiles' -v '/home/buechse/.gitcache:/data/riotbuild/gitcache:delegated' -e 'GIT_CACHE_DIR=/data/riotbuild/gitcache'    -e 'BOARD=hifive1b' -e 'DISABLE_MODULE=' -e 'DEFAULT_MODULE=' -e 'FEATURES_REQUIRED=' -e 'FEATURES_BLACKLIST=' -e 'FEATURES_OPTIONAL=' -e 'USEMODULE=auto_init_gnrc_netif constfs gcoap gnrc_icmpv6_echo gnrc_ipv6_default gnrc_netapi_callbacks netdev_default saul_default vfs ztimer ztimer_msec ztimer_sec ztimer_usec' -e 'USEPKG='  -w '/data/riotbuild/riotbase/examples/lang_support/official/rust-gcoap/' '5ebf563ef282' make
Building application "rust_gcoap" for "hifive1b" with CPU "fe310".

"make" -C /data/riotbuild/riotbase/boards compile-commands
"make" -C /data/riotbuild/riotbase/boards/common/init compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/boards/hifive1b compile-commands
make[2]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/core compile-commands
make[2]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/core/lib compile-commands
make[2]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/cpu/fe310 compile-commands
"make" -C /data/riotbuild/riotbase/cpu/fe310/periph compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/cpu/fe310/vendor compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/cpu/riscv_common compile-commands
"make" -C /data/riotbuild/riotbase/cpu/riscv_common/periph compile-commands
make[4]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/drivers compile-commands
"make" -C /data/riotbuild/riotbase/drivers/netdev compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/drivers/periph_common compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/drivers/saul compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/drivers/saul/init_devs compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys compile-commands
"make" -C /data/riotbuild/riotbase/sys/auto_init compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/div compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/event compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/evtimer compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/fmt compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/frac compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/fs/constfs compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/iolist compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/libc compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/luid compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/malloc_thread_safe compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /data/riotbuild/riotbase/sys/net/application_layer/gcoap compile-commands
[... 1239 lines cut ...]
"make" -C /data/riotbuild/riotbase/sys/net/gnrc/transport_layer/udp
"make" -C /data/riotbuild/riotbase/sys/net/link_layer/eui_provider
"make" -C /data/riotbuild/riotbase/sys/net/link_layer/l2util
"make" -C /data/riotbuild/riotbase/sys/net/netif
[... cut ...]
"make" -C /data/riotbuild/riotbase/sys/ztimer
/usr/bin/x86_64-linux-gnu-ld.bfd: warning: /data/riotbuild/riotbase/examples/lang_support/official/rust-gcoap/bin/native64/rust_gcoap.elf has a LOAD segment with RWX permissions
   text    data     bss     dec     hex filename
 331407    7664  141808  480879   7566f /data/riotbuild/riotbase/examples/lang_support/official/rust-gcoap/bin/native64/rust_gcoap.elf
make: Leaving directory '/home/buechse/RIOTstuff/riot-upstream3/RIOT/examples/lang_support/official/rust-gcoap'
```

</details>

chrysn, 2026-09-09:
21531 was mostly "we see it is breaking, we don't know how to fix it, so we exclude it"; if it works now, let's re-enable it.

But I think we now have two separate issues that are both not this none any more.

crasbe, 2026-09-09:
That is probably true, but IMO it doesn't make sense to create a new issue.

<hr/>

Some more debugging: Adding `USEMODULE += vfs` to the example Makefile just causes the `rust-hello-world` error to become the same error as the `rust-gcoap` error.

The `rust-riot-sys` unconditionally includes `vfs.h`:
https://github.com/RIOT-OS/rust-riot-sys/blob/main/riot-c2rust.h#L87

...and `vfs.h` unconditionally includes `sys/statvfs.h`:
https://github.com/RIOT-OS/RIOT/blob/681273ecd3988d3d61fc06fbab0ecffd075b8d3d/sys/include/vfs.h#L58

Maybe the first include has to be guarded with `#if MODULE_VFS`, but idk how to build `rust-riot-sys`.

<hr/>

Claude says that the other issue with the data types is a somewhat known problem because the `__rvv_int8mf8x2_t` tuples were added/extended with newer revisions of LLVM. So this is really a RISC-V and C2Rust specific issue rather than a RIOT issue, but it would mean that we have to (temporarily) disable Rust support for RISC-V.

crasbe, 2026-09-10:
I created a PR in `C2Rust` that adds the required functionality for the vector tuple types: https://github.com/immunant/c2rust/pull/2008

closed 2026-09-10 by Enoch247