RIOT-OS/RIOT #20537: doc: nesting of member groups is not allowed

Enoch247, 2024-04-03:
#### Description

Doxygen does not allow for [member groups](https://www.doxygen.nl/manual/grouping.html#memgroup) to be nested. It is a common idiom that we wrap whole files in a member group so that the cmd `@ingroup` can be applied to everything in the file. This creates problems when member groups are then actually used in the file. Current behavior of Doxygen seems to be that instead of maintaining a stack to know which group it should place doc elements into, it simply treats the group opening cmd as an indication to set the current group and the closing cmd to exit all groups (ie exit back to top-level/no group).

#### Steps to reproduce the issue

I'm working on an example that I will post here.

#### Expected results

#### Actual results

#### Versions

- current master
- 2024.04
- probably affects many more versions

kfessel, 2024-04-03:
are you sure that's what is happening?

i think there is a difference between **group** ( `@ingroup @defgroup @weakgroup @addtogroup `) that we are using for files 

and **membergroup**s `@name`

we can have a member group inside a group

but opening a member group cancels automatic membergroups

kfessel, 2024-04-03:
`@{` and `@}` are blocks these can be nested but they don't do much if they a not with other commands by the doxgen example they apply they documentation text to all elements inside.

Enoch247, 2024-04-03:
I just added steps to reproduce. I think it should prove that our use of nesting is breaking things. This isn't the best example in that we probably actually are okay with the outputted doc in this case, but it is the example I have in-front of me and understand best at this time.

Enoch247, 2024-04-03:
Another ancillary thing I noticed is that there seems to be alot of misunderstanding surrounding use of the member groups exacerbated by copy-pasting. I'm not actually sure that the member group used [here](https://github.com/RIOT-OS/RIOT/blob/a9e7b441a7be21a90a26bbd199afa060f3646559/cpu/stm32/include/candev_stm32.h#L22) does anything.

Enoch247, 2024-04-03:
I believe the `@addtogroup` supports use of the opening and closing braces, but `@ingroup` doe not, maybe we could replace our use of ingroup with addgroup?

kfessel, 2024-04-03:
you are probably rigth, in that there is missing understanding of these blocks, groups and member-group things and a lot of copy and paste.

Enoch247, 2024-04-03:
> you are probably rigth, in that there is missing understanding of these blocks, groups and member-group things and a lot of copy and paste.

I know I am guilty of it.

Enoch247, 2024-04-03:
I believe
``` c
/**
 * @ingroup foo
 * @{
*/

/** @} */
```

creates a member group (between the braces) and places that member group into the topic group foo. Where-as:

``` c
/**
 * @addtogroup foo
 * @{
*/

/** @} */
```

add everything between the braces to the group foo.

kfessel, 2024-04-03:
> I believe the `@addtogroup` supports use of the opening and closing braces, but `@ingroup` does not, maybe we could replace our use of ingroup with addgroup?

`@addgroup id title` adds the things it is documenting to an allready existing group if exiting else it creates that group giving it the title
`@ingroup id` only adds to an existing group
`@defgroup id title` creates a new group can't be done twice  (overwites addgroup titles)

all of them can  have blockes open

kfessel, 2024-04-03:
> > ```c
> /**
>  * @ingroup foo
>  * @{
> */
> 
> /** @} */
> ```
> creates a member group (between the braces) and places that member group into the topic group foo. Where-as:

no this puts things in a group that is existing 

> ```c
> /**
>  * @addtogroup foo
>  * @{
> */
> 
> /** @} */
> ```
> adds everything between the braces to the group foo.

either add to an existing group or creates a new one

Enoch247, 2024-04-03:
> `@addgroup id title` adds the things it is documenting to an allready existing group if exiting else it creates that group giving it the title `@ingroup id` only adds to an existing group `@defgroup [...]

I agree with everything above, except that the statement that `@ingroup` supports braces. I think we are unwitting creating a member group and then adding the member group to the topic group. I can't find anything to support the statment that ingroup supports brace syntax.

Enoch247, 2024-04-03:
hmm. I may be wrong

> To avoid putting [\ingroup](https://www.doxygen.nl/manual/commands.html#cmdingroup) commands in the documentation for each member you can also group members together by the open marker @{ before the group and the closing marker @} after the group. The markers can be put in the documentation of the group definition or in a separate documentation block.
(https://www.doxygen.nl/manual/grouping.html)

kfessel, 2024-04-03:
at least we wouldn't be alone with that 

https://github.com/zephyrproject-rtos/zephyr/blob/bba8641354bfbf83743c032c378a791b941ad7fc/include/zephyr/ipc/icmsg.h#L23-27

kfessel, 2024-04-03:
doxygen also reads some blocking/grouping from the comments and some from the source code

Enoch247, 2024-04-03:
I too was suspicious of the `#ifndef DOXYGEN` statement, but could find explanation of why they could be the root cause. I am going to experiment a bit more and see if I can't get some more facts to work from.

Enoch247, 2024-04-03:
Ok, so I tried removing all the inner member groups, still none of the comment blocks showed up in the parent topic.

Next, I replaced `ingroup` with `addtogroup`, and everything showed up in the parent topic.

Next, I put all the inner member groups back, leaving the `addtogroup` in-palce, and everything continued to show in the parent.

Finally, I restored the file back to its original state and added a `defgroup` right before the `ingroup`. `git diff` shows:

``` patch
diff --git a/cpu/stm32/include/periph/cpu_spi.h b/cpu/stm32/include/periph/cpu_spi.h
index 3fa42e11d4..26a49975a4 100644
--- a/cpu/stm32/include/periph/cpu_spi.h
+++ b/cpu/stm32/include/periph/cpu_spi.h
@@ -8,6 +8,7 @@
  */
 
 /**
+ * @defgroup        cpu_stm32_periph_spi SPI
  * @ingroup         cpu_stm32
  * @{
  *

```
And now the parent topic (CPU/STM32) contains the new SPI topic with everything in the file showing under the new topic group.

Enoch247, 2024-04-03:
I now re-read the previous quote:

> To avoid putting [\ingroup](https://www.doxygen.nl/manual/commands.html#cmdingroup) commands in the documentation for each member you can also group members together by the open marker @{ before the [...]

and see that the use of braces is referred to "put in the documentation of the group definition" (`defgroup`), "or in a separate documentation block" (not really sure what they mean here but they don't really tie it back to `ingroup` here). I think this is just a poor choice of words here and what they are trying to say is that you can use braces (with other commands) if you don't want to plaster every comment block with an `ingroup` cmd.

kfessel, 2024-04-04:
I think to understand the choice of command words and their why you would need to dig through the history of doxygen and maybe javadoc.

Thanks for investigating

Enoch247, 2024-04-17:
I have updated this issue's title and description to describe the real problem that is occurring. A very simple approach to fixing this is replacing `ingroup` with `addtogroup` wherever `ingroup` is used without a `defgroup`. This approach would not add any new document groups and honor (I believe) what the original authors intended in their doxgen blocks. However, I believe heavy use of `addtogroup` could lead to maintainability problems.

The other approach is to add a `defgroup` everywhere their is in `ingroup` without a `defgroup`. This approach would add new topic groups to the doc.

kfessel, 2024-04-18:
> A very simple approach to fixing this is replacing ingroup with addtogroup wherever ingroup

addtogroup would need a group title i think -not sure it seems to be optional- and the first addtogroup with title would define the group title  

I would like to make @miri64 (does a lot of documentation things) and @maribu (most recent edits in the /doc folder) aware of this issue.

@Enoch247 may be a/some PR(s) would having one or some file(s) changed in the suggested ways would help to illustrate (please link this issue)

I think there also was some thought in the air of having another renderer (MkDocs or [Sphinx](https://docs.readthedocs.io/en/stable/intro/getting-started-with-sphinx.html?) ) for the documentation output. We certainly will want to avoid compatibility issues

maribu, 2024-04-18:
Regarding the use of Sphinx: We would use [breathe](https://breathe.readthedocs.io/), which parses the XML output of Doxygen, to generate the API doc. Thus, switching to breathe would only fix the "website looks ugly" issue. Any issues with Doxygen would stay by thay switch.

Enoch247, 2025-04-08:
If this ever gets fixed, we will also need to update the example in the codding conventions under the [documentation section](https://github.com/RIOT-OS/RIOT/blob/2025.01/CODING_CONVENTIONS.md#documentation).

maribu, 2025-04-08:
I think this might be an issue with the use of `@file`, which explicitly tells Doxygen that something should apply to the whole file. We have quite a few headers where both `@{` ... `@}` and `@file` is used, but only one of them would actually be needed.

Enoch247, 2025-04-12:
I believe the issue to be improper use of `ingroup` and not a bug in doxygen. Its been a while since I looked at this problem, but I recall that while doxygen's documentation of these cmds is a bit unclear leading up to their misuse. I think the tool is doing exactly what it's doc claims it will do when used in this way.

albert-github, 2025-04-17:
Which version of doxygen is used?
See also my comments at https://github.com/doxygen/doxygen/issues/11539#issuecomment-2813516622 and the comment from @Enoch247 (in https://github.com/RIOT-OS/RIOT/issues/20537#issuecomment-2798787886)

albert-github, 2025-04-17:
@Enoch247 

> `ingroup` syntax does not support use of braces

is not correct, the grouping by means if `@{` / `@}` is honored.

- It is correct that `ingroup` is used for creating hierarchies of `defgroup` / `addtogroup` / `weakgroup` definitions of groups