RIOT-OS/RIOT #20604: BlackMagicProbe does not recognize nRF52 device

chrysn, 2024-04-21:
#### Description

Something in the bmp.py script goes wrong and can not enumerate the targets.

#### Steps to show that all relevant hardware works

* Connect a BlackMagicProbe (I use v2.3 with the latest firmware v1.10.2) to a thingy:52
* in examples/saul

```
$ make BOARD=thingy52 all
[...]
$ gdb-multiarch bin/thingy52/saul_example.elf
[...]
(gdb) target extended-remote /dev/ttyACM0
[...]
(gdb) monitor swdp_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
(gdb) attach 1
[...]
(gdb) load
Loading section .text, size 0x6460 lma 0x0
Loading section .ARM.exidx, size 0x8 lma 0x6460
Loading section .eh_frame, size 0x88 lma 0x6468
Loading section .relocate, size 0xc0 lma 0x64f0
Start address 0x0000082c, load size 26032
Transfer rate: 41 KB/sec, 867 bytes/write.
```

So connectivity is there and all is fine.

#### Steps to reproduce the issue

In the same examples/saul,

```
$ make BOARD=thingy52 all flash term PROGRAMMER=bmp
[...]
/home/chrysn/git/RIOT/dist/tools/bmp/bmp.py  flash /home/chrysn/git/RIOT/examples/saul/bin/thingy52/saul_example.elf
found following Black Magic GDB servers:
        [/dev/ttyACM0] Serial: 81D6649A <- default 
connecting to [/dev/ttyACM0]...
connecting successful.
scanning using SWD...
Traceback (most recent call last):
  File "/home/chrysn/git/RIOT/dist/tools/bmp/bmp.py", line 270, in <module>
    gdbmi = connect_to_target(port)
            ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chrysn/git/RIOT/dist/tools/bmp/bmp.py", line 243, in connect_to_target
    assert len(targets) > 0, "no targets found"
           ^^^^^^^^^^^^^^^^
AssertionError: no targets found
make: *** [/home/chrysn/git/RIOT/examples/saul/../../Makefile.include:849: flash] Error 1
```

#### Expected results

BMP support is there, so it should just work.

#### Versions

Current RIOT HEAD, all requirements of bmp installed in a fresh venv.

#### Next steps

I'm a bit at a loss. Pinging @maz3max and @maribu who might have more insights.

maribu, 2024-04-21:
To be honest, I have not used SWD boards without an integrated debugger for some time, so I didn't have the need to use BMP for ages.

Chances are good that this regression has been introduced some time ago. My first guess would be checking if it is a breaking API change in pygdbmi. In 9 out of 10 cases of a python script starting to fail at runtime and versions not being pinned, it is a breaking API change in one of the modules.

crasbe, 2024-04-21:
I don't have a BMP, but I have an idea how we could debug this:
Simply add a "print(res)" after this line: https://github.com/RIOT-OS/RIOT/blob/master/dist/tools/bmp/bmp.py#L238 and post the console output here.

This prints pygdbmi's interpretation of the gdp response. This should let us isolate if the fault lies in pygdbmi or in the bmp.py script.

closed 2024-05-10 by benpicco