RIOT-OS/RIOT #20743: Allow custom FEATURE to be defined from an external source

HendrikVE, 2024-06-11:
#### Description

We are using the RIOT repository as a base for our apps and custom boards. While it is possible to add external `MODULE`s and `BOARD`s, as far as I understand it's not possible to add custom `FEATURE`s. In case some of my boards provide a custom feature I'd like to be able to add it to the list of available features, because otherwise the build would fail as the buildsystem does not recognize the given feature. But since this list (`RIOT/makefiles/features_existing.inc.mk`) is hardcoded and auto-generated there is currently no way to do that as far as I can see.

maribu, 2024-06-28:
That is indeed true.

It would be possible to also `-include` an `features_existing.inc.mk` within the board's folder, or for each folder in `EXTERNAL_BOARD_DIRS` to extend this.

Could you elaborate a bit more your use case?

To me, it is not obvious how a custom feature provided in a board would be useful. After all, only that board would know about that custom feature. And usually the value of features is that code in RIOT can depend on features or unlock extra functionality (via `FEATURES_OPTIONAL`) based on a feature - but that won't work for features RIOT is not aware of anyway.

(Just to rule out this is that this is not an [XY problem](https://en.wikipedia.org/wiki/XY_problem). If there is indeed value in allowing external boards to add custom features, we should implement this.)

HendrikVE, 2024-06-28:
My use case is to allow a common API function that boards can optionally provide via the `board.h`, e.g. an extended initialization function for testing environments. It's not a hardware but a software feature of a board that can be shared throughout the repository. So if I have an application, which works for several boards, that wants to call this extended board feature conditionally it needs to know whether it is available for the compiled board. 

Theoretically one could do it as described [here](https://stackoverflow.com/a/43854514), but that's not very clean.

As a workaround the `Makefile.features` of my boards can add a custom feature with
`CFLAGS += -DFEATURES_PROVIDED_BOARD_<feature_name>`
which will then be respected in the applications.