RIOT-OS/RIOT #20759: Nightly Rust features fails with BUILD_IN_DOCKER

WataNekko, 2024-06-23:
#### Description

Building Rust examples on native with docker (BUILD_IN_DOCKER=1) without `[profile.release] panic = "abort"` fails with
```rs
   Compiling riot-wrappers v0.8.4
error[E0554]: `#![feature]` may not be used on the stable release channel
  --> /data/riotbuild/.cargo/registry/src/index.crates.io-6f17d22bba15001f/riot-wrappers-0.8.4/src/lib.rs:19:5
   |
19 |     feature(lang_items)
   |     ^^^^^^^^^^^^^^^^^^^
```

Seems like the `riot/riotbuild` docker image is not using nightly toolchain. Forcing nightly with `rust-toolchain.toml`
```toml
[toolchain]
channel = "nightly"
```
Gives
```sh
info: syncing channel updates for 'nightly-x86_64-unknown-linux-gnu'
error: could not create temp file /opt/rustup/.rustup/tmp/9rdon3zhu42fd5p8_file: Permission denied (os error 13)
Error: No Rust libraries are installed for the board's CPU. Run
    $ rustup target add i686-unknown-linux-gnu
or set `CARGO_OPTIONS=-Zbuild-std=core`.
```

Is nightly not supported in docker and I have to add `panic = "abort"` as a workaround?
(Rust examples here works fine. I was trying to run https://gitlab.com/etonomy/riot-module-examples cc @chrysn)

chrysn, 2024-06-23:
Thanks for reporting this; setting a toolchain should work in docker --
for "should" as in "it'd be good if", not as in "it worked before" (I
never tested it).

That aside, the panic mode needs to be set to "abort" on native to avoid
the need for a nightly. Mind you that a panic is aborting the current
thread in RIOT anyway for lack of unwinding, and due to how the panic
handlers in riot-wrappers are implemented. Setting panic=abort merely
also tells the compiler that there is really no unwinding going on,
possibly allowing some more optimizations and avoiding the need for a
nightly-only eh_personality, so there is nothing gained by leaving panic
on its default value except brevity of the Cargo.toml.

WataNekko, 2024-06-24:
Thanks for your response. Okay, I’m now clear about setting panic is recommended. But if I were to use some nightly features and build with docker, how should I go about setting the toolchain for the docker image? I assume it would be best to configure this somewhere in the `riot/riotbuild` image upstream.

crasbe, 2026-10-06:
This has been solved in the meantime.

closed 2026-10-06 by crasbe