RIOT-OS/RIOT #20791: [Tracking] Fix failures of test-on-iotlab

mguetschow, 2024-07-18:
### Description

Several tests have been failing for quite some time on specific boards in the [`test-on-iotlab` action](https://github.com/RIOT-OS/RIOT/actions/workflows/test-on-iotlab.yml). They are not strictly part of the [release-specs task02](https://github.com/RIOT-OS/Release-Specs/tree/master/02-tests) as of now, but do indicate bugs in RIOT for specific boards. This issue collects constantly failing test-board combinations and should be used to coordinate fixing.

```
ERROR:dwm1001:Tests failed: 3
Failures during test:
- [tests/periph/timer_periodic](tests/periph/timer_periodic/test.failed)
- [tests/pkg/lz4](tests/pkg/lz4/test.failed)
- [tests/pkg/uzlib](tests/pkg/uzlib/test.failed)

ERROR:iotlab-m3:Tests failed: 3
Failures during test:
- [tests/drivers/mtd_flashpage](tests/drivers/mtd_flashpage/test.failed)
- [tests/net/gcoap_dns](tests/net/gcoap_dns/test.failed)
- [tests/pkg/libschc](tests/pkg/libschc/test.failed)

ERROR:nrf52dk:Tests failed: 5
Failures during test:
- [tests/pkg/lz4](tests/pkg/lz4/test.failed)
- [tests/pkg/uzlib](tests/pkg/uzlib/test.failed)
- [tests/sys/congure_abe](tests/sys/congure_abe/test.failed)
Failures during test.flash:
- [tests/sys/cb_mux_bench](tests/sys/cb_mux_bench/test.flash.failed)
- [tests/sys/xtimer_hang](tests/sys/xtimer_hang/test.flash.failed)

ERROR:samr34-xpro:Tests failed: 4
Failures during compilation:
- [tests/turo_txt](tests/turo_txt/compilation.failed)
Failures during test:
- [tests/drivers/mtd_raw](tests/drivers/mtd_raw/test.failed)
- [tests/periph/rtt](tests/periph/rtt/test.failed)
- [tests/sys/dbgpin](tests/sys/dbgpin/test.failed)

ERROR:b-l072z-lrwan1:Tests failed: 1
Failures during test:
- [tests/pkg/libschc](tests/pkg/libschc/test.failed)
```

The failure during compilation on the `samr34-xpro` was apparently due to insufficient space on the action runner. Not sure how to cope with this.

### Useful links

The list was initially compiled from the [Workflow Action on the `2024.07-RC1`](https://github.com/RIOT-OS/RIOT/actions/runs/9953273322), and has been compared to several recent `master` and the [`2024.04-RC2` workflow](https://github.com/RIOT-OS/RIOT/actions/runs/8876192411).

MrKevinWeiss, 2024-07-18:
I think `tests/periph/timer_periodic` and `tests/pkg/libschc` have always been flakey or failing with IoTLabs. I should look at `tests/turo_txt`.  I thought the mtd ones were fixed but maybe not...  I can take a closer look on Monday :)

mguetschow, 2026-10-19:
Update: current failures on https://github.com/RIOT-OS/RIOT/actions/runs/38188477365/job/114615898316

```
ERROR:dwm1001:Tests failed: 7
Failures during test:
- [examples/lang_support/community/micropython](examples/lang_support/community/micropython/test.failed)
- [tests/sys/congure_abe](tests/sys/congure_abe/test.failed)
- [tests/sys/congure_quic](tests/sys/congure_quic/test.failed)
- [tests/sys/congure_reno](tests/sys/congure_reno/test.failed)
- [tests/sys/congure_test](tests/sys/congure_test/test.failed)
- [tests/sys/heap_cmd](tests/sys/heap_cmd/test.failed)
- [tests/turo](tests/turo/test.failed)

ERROR:iotlab-m3:Tests failed: 3
Failures during test:
- [examples/advanced/psa_crypto](examples/advanced/psa_crypto/test.failed)
- [tests/drivers/mtd_flashpage](tests/drivers/mtd_flashpage/test.failed)
- [tests/pkg/libschc](tests/pkg/libschc/test.failed)

ERROR:nrf52dk:Tests failed: 3
Failures during test:
- [examples/lang_support/community/micropython](examples/lang_support/community/micropython/test.failed)
Failures during test.flash:
- [examples/advanced/psa_crypto](examples/advanced/psa_crypto/test.flash.failed)
- [tests/pkg/flatbuffers](tests/pkg/flatbuffers/test.flash.failed)

ERROR:nrf52840dk:Tests failed: 1
Failures during test:
- [tests/sys/psa_crypto_persistent_storage](tests/sys/psa_crypto_persistent_storage/test.failed)

ERROR:samr21-xpro:Tests failed: 1
Failures during test:
- [examples/advanced/psa_crypto](examples/advanced/psa_crypto/test.failed)

ERROR:b-l072z-lrwan1:Tests failed: 3
Failures during test:
- [tests/periph/rtt_min](tests/periph/rtt_min/test.failed)
- [tests/pkg/libschc](tests/pkg/libschc/test.failed)
- [tests/turo](tests/turo/test.failed)

ERROR:b-l475e-iot01a:Tests failed: 2
Failures during test:
- [tests/drivers/mtd_flashpage](tests/drivers/mtd_flashpage/test.failed)
- [tests/pkg/libschc](tests/pkg/libschc/test.failed)
```

`nucleo-wl55jc` seems to have no available boards (?) and `samr34-xpro` seem to be having major issues (a lot of flash failures, some test failures).

miri64, 2026-10-19:
> ```
> - [tests/sys/congure_abe](tests/sys/congure_abe/test.failed)
> - [tests/sys/congure_quic](tests/sys/congure_quic/test.failed)
> - [tests/sys/congure_reno](tests/sys/congure_reno/test.failed)
> - [tests/sys/congure_test](tests/sys/congure_test/test.failed)
> ```

Congure has always been a bit flaky too. Will take a look into the `libschc` tests.

crasbe, 2026-10-20:
I have a `b-l072z-lrwan1`, `dwm1001`, `nrf52840dk` and `nucleo-wl55jc` here. Can do some testing/vibe fixing.

miri64, 2026-10-20:
> 14\. I have a `b-l072z-lrwan1`, `dwm1001`, `nrf52840dk` and `nucleo-wl55jc` here. Can do some testing/vibe fixing.

Let's see what is still broken on RC1 on IoT-LAB.

miri64, 2026-10-20:
(from my experience: Copilot can also be really helpful to understand the errors in the workflow)

miri64, 2026-10-20:
~~Mh, is the `test_utils_interactive_sync` module still included for any of the tests (for the once I checked so far, I did not see it)?~~ checked for native... for that it is of course excluded 🤦‍♀️

miri64, 2026-10-20:
> ~~Mh, is the `test_utils_interactive_sync` module still included for any of the tests (for the once I checked so far, I did not see it)?~~ checked for native... for that it is of course excluded 🤦‍♀️ 

However, _some_ of the still failing tests actually do not include it. Will provide a follow-up for #22942.

crasbe, 2026-10-20:
> Will take a look into the `libschc` tests.

This fails because the interrupts are blocked for too long so that the UART transmission cuts out. The issue lies in `periph_rtt` though because `rtt_set_alarm()` waits for a flag to be set. I'll see if I can come up with a solution and a test case for it.

crasbe, 2026-10-20:
> This fails because the interrupts are blocked for too long so that the UART transmission cuts out. The issue lies in `periph_rtt` though because `rtt_set_alarm()` waits for a flag to be set. I'll [...]

Some might even say... *it's a race condition* 🏎️🏁

miri64, 2026-10-20:
> I have a `b-l072z-lrwan1`, `dwm1001`, `nrf52840dk` and `nucleo-wl55jc` here. Can do some testing/vibe fixing.

`nucleo-wl55jc` seems to be broken on IoT-LAB anyway. Can you vibe fix that? 🤡

crasbe, 2026-10-20:
>> I have a `b-l072z-lrwan1`, `dwm1001`, `nrf52840dk` and `nucleo-wl55jc` here. Can do some testing/vibe fixing.
>
> `nucleo-wl55jc` seems to be broken on IoT-LAB anyway. Can you vibe fix that? 🤡

🔴 I'm sorry Dave, I'm afraid I can't do that.

miri64, 2026-10-20:
> 18\. However, _some_ of the still failing tests actually do not include it. Will provide a follow-up for [#22942](https://github.com/RIOT-OS/RIOT/pull/22942).

See https://github.com/RIOT-OS/RIOT/pull/22951

miri64, 2026-10-20:
> This fails because the interrupts are blocked for too long so that the UART transmission cuts out. The issue lies in `periph_rtt` though because `rtt_set_alarm()` waits for a flag to be set. I'll see if I can come up with a solution and a test case for it.

There is also a failure in `tests/periph/rtt` on some boards. Could this also be the case?

miri64, 2026-10-20:
[Failures during `2026.10-RC1`](https://github.com/RIOT-OS/RIOT/actions/runs/39443044269):

`dwm1001`
- Failures during test:
  + <details><summary><tt>tests/build_system/test_tools</tt></summary>

    ```
    make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory -C ./tests/build_system/test_tools test
    shellping
    ssh -t rici@toulouse.iot-lab.info 'socat - tcp:dwm1001-100.toulouse.iot-lab.info:20000'
    shellping
    main(): This is RIOT! (Version: buildtest)
    Running 'tests_tools' application
    shellpong
    shellpong
    true this should not be echoed
    shellping
    toupper lowercase
    shellpong
    
    Traceback (most recent call last):
    File "/home/runner/work/RIOT/RIOT/tests/build_system/test_tools/tests/01-run.py", line 83, in <module>
    sys.exit(run(testfunc))
    File "/home/runner/work/RIOT/RIOT/dist/pythonlibs/testrunner/__init__.py", line 31, in run
    testfunc(child)
    File "/home/runner/work/RIOT/RIOT/tests/build_system/test_tools/tests/01-run.py", line 76, in testfunc
    _test_clean_output(child)
    File "/home/runner/work/RIOT/RIOT/tests/build_system/test_tools/tests/01-run.py", line 55, in _test_clean_output
    assert retline.strip() == 'LOWERCASE'
    AssertionError
    make: *** [/home/runner/work/RIOT/RIOT/makefiles/tests/tests.inc.mk:32: test] Error 1
    
    Return value: 2
    ```

    </details>
  + <details><summary><tt>tests/periph/wdt</tt></summary>

    ```
    make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory -C ./tests/periph/wdt test
    Timeout in expect script at "child.expect_exact(">")" (tests/periph/wdt/tests/01-run.py:51)
    
    make: *** [/home/runner/work/RIOT/RIOT/makefiles/tests/tests.inc.mk:32: test] Error 1
    
    Return value: 2
    ```

    </details>
  + <details><summary><tt>tests/pkg/edhoc_c</tt></summary>

    ```
    make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory -C ./tests/pkg/edhoc_c test
    pkg/edhoc-c is deprecated and will be removed after the 2027.04 Release!
    
    
    pkg/edhoc-c is deprecated and will be removed after the 2027.04 Release!
    ssh -t rici@toulouse.iot-lab.info 'socat - tcp:dwm1001-100.toulouse.iot-lab.info:20000'
    main(): This is RIOT! (Version: buildtest)
    [initiator]: load ephemeral key: ONLY FOR TESTING
    [initiator]: preset cid: ONLY FOR TESTING
    [responder]: load ephemeral key: ONLY FOR TESTING
    [responder]: preset cid: ONLY FOR TESTING
    Starting the shell
    > main():
    
    >
    
    > Timeout in expect script at "child.expect_exact('>')" (tests/pkg/edhoc_c/tests/01-run.py:49)
    
    make: *** [/home/runner/work/RIOT/RIOT/makefiles/tests/tests.inc.mk:32: test] Error 1
[... 2231 lines cut ...]

    ```
    make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory -C ./tests/drivers/mtd_flashpage test
    ssh -t rici@saclay.iot-lab.info 'socat - tcp:st-iotnode-10.saclay.iot-lab.info:20000'
    s to start test, r to print it is ready
    r
    READY
    s
    START
    main(): This is RIOT! (Version: buildtest)
    ......
    mtd_flashpage_tests.test_mtd_aux_slot (tests/drivers/mtd_flashpage/main.c 287) exp 24 was 0
    
    run 6 failures 1
    { "threads": [{ "name": "main", "stack_size": 1536, "stack_used": 592 }]}
    Timeout in expect script at "child.expect(r'OK \((\d+) tests\)', timeout=timeout)" (dist/pythonlibs/testrunner/__init__.py:59)
    
    make: *** [/home/runner/work/RIOT/RIOT/makefiles/tests/tests.inc.mk:32: test] Error 1
    
    Return value: 2
    ```

    </details>

Obviously, the flash failures are unrelated to the release, so I restarted the tests for that particular board.

miri64, 2026-10-20:
Most of these should be fixed by either #22951 and #22953.

miri64, 2026-10-20:
#22958 is a workaround for the `samr38-xpro` issues upstream.

miri64, 2026-10-20:
> `nucleo-wl55jc` seems to be broken on IoT-LAB anyway. Can you vibe fix that? 🤡

And see #22956 for that :-)

mguetschow, 2026-11-04:
Failing tests on `master` (a0ba4982ffc5d275cf07ab11b1fcbd3f0c0377a9) from https://github.com/RIOT-OS/RIOT/actions/runs/41366367265

`dwm1001`
- Failures during test:
  + <details><summary><tt>tests/build_system/test_tools</tt></summary>

    ```
    make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory -C ./tests/build_system/test_tools test
    ssh -t rici@toulouse.iot-lab.info 'socat - tcp:dwm1001-100.toulouse.iot-lab.info:20000'
    shellping
    shellping
    shellping
    main(): This is RIOT! (Version: buildtest)
    Running 'tests_tools' application
    shellpong
    shellpong
    shellpong
    true this should not be echoed
    shellping
    toupper lowercase
    
    Traceback (most recent call last):
    File "/home/runner/work/RIOT/RIOT/tests/build_system/test_tools/tests/01-run.py", line 85, in <module>
    sys.exit(run(testfunc))
    ~~~^^^^^^^^^^
    File "/home/runner/work/RIOT/RIOT/dist/pythonlibs/testrunner/__init__.py", line 31, in run
    testfunc(child)
    ~~~~~~~~^^^^^^^
    File "/home/runner/work/RIOT/RIOT/tests/build_system/test_tools/tests/01-run.py", line 78, in testfunc
    _test_clean_output(child)
    ~~~~~~~~~~~~~~~~~~^^^^^^^
    File "/home/runner/work/RIOT/RIOT/tests/build_system/test_tools/tests/01-run.py", line 55, in _test_clean_output
    assert retline.strip() == 'LOWERCASE'
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    AssertionError
    make: *** [/home/runner/work/RIOT/RIOT/makefiles/tests/tests.inc.mk:32: test] Error 1
    
    Return value: 2
    ```

    </details>
  + <details><summary><tt>tests/sys/congure_abe</tt></summary>

    ```
    make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory -C ./tests/sys/congure_abe test
    ....EEF
    ======================================================================
    ERROR: test_init (__main__.TestCongUREABEDefaultInitTests.test_init)
    Test the initial congestion state.
    ----------------------------------------------------------------------
    Traceback (most recent call last):
    File "/home/runner/work/RIOT/RIOT/tests/sys/congure_abe/tests/01-run.py", line 190, in test_init
    self.assertEqual(state['consts']['init_mss'], state['mss'])
    ~~~~~^^^^^^^^^^
    KeyError: 'consts'
    
    ======================================================================
    ERROR: test_setup (__main__.TestCongUREABEDefaultInitTests.test_setup)
    ----------------------------------------------------------------------
    Traceback (most recent call last):
    File "/home/runner/work/RIOT/RIOT/tests/sys/congure_abe/tests/01-run.py", line 147, in test_setup
    self.assertIsNotNone(state['consts'])
    ~~~~~^^^^^^^^^^
    KeyError: 'consts'
    
    ======================================================================
    FAIL: test_no_setup (__main__.TestCongUREABEWithoutSetup.test_no_setup)
    ----------------------------------------------------------------------
    Traceback (most recent call last):
    File "/home/runner/work/RIOT/RIOT/tests/sys/congure_abe/tests/01-run.py", line 128, in test_no_setup
[... 4707 lines cut ...]
    s
    START
    main(): This is RIOT! (Version: buildtest)
    Testing clock: ZTIMER_MSEC
    i: 0 time: 3009 offset: 1
    i: 1 time: 3108 offset: 1
    i: 2 time: 3208 offset: 0
    i: 3 time: 3309 offset: 1
    i: 4 time: 3408 offset: 1
    Testing clock: ZTIMER_USEC
    i: 0 time: 1894352 offset: 116
    Offset too large, failure
    i: 1 time: 1904368 offset: 16
    i: 2 time: 1914368 offset: 0
    i: 3 time: 1924368 offset: 0
    i: 4 time: 1934368 offset: 0
    Test failed!
    Timeout in expect script at "child.expect_exact("Test successful.")" (tests/sys/ztimer_periodic/tests/01-run.py:11)
    
    make: *** [/home/runner/work/RIOT/RIOT/makefiles/tests/tests.inc.mk:32: test] Error 1
    
    Return value: 2
    ```

    </details>