RIOT-OS/RIOT #20837: add CAN support to Nucleo L432KC (Nucleo 32)

donsez, 2024-08-27:
#### Description

The [Nucleo L432KC](https://os.mbed.com/platforms/ST-Nucleo-L432KC/) board is a Nucleo 32 board supported by RIOT OS.

This board supports CAN interface. However, CAN periph is not enabled into `Makefile.features` and the definition of the CAN pins is not correct for the STM32L432KC : `PA11/PA12` instead of `PB8/PB9` into `cpu/stm32/include/can_params.h`

#### Steps to reproduce the issue

The following patch is enough for using the CAN bus with [Nucleo L432KC](https://os.mbed.com/platforms/ST-Nucleo-L432KC/)

```
diff --git a/boards/nucleo-l432kc/Makefile.features b/boards/nucleo-l432kc/Makefile.features
index 1bf6b33cbc..a0c1c07920 100644
--- a/boards/nucleo-l432kc/Makefile.features
+++ b/boards/nucleo-l432kc/Makefile.features
@@ -2,6 +2,7 @@ CPU = stm32
 CPU_MODEL = stm32l432kc
 
 # Put defined MCU peripherals here (in alphabetical order)
+FEATURES_PROVIDED += periph_can
 FEATURES_PROVIDED += periph_i2c
 FEATURES_PROVIDED += periph_pwm
 FEATURES_PROVIDED += periph_rtc
diff --git a/cpu/stm32/include/can_params.h b/cpu/stm32/include/can_params.h
index d1c0e26912..8a6a58c7be 100644
--- a/cpu/stm32/include/can_params.h
+++ b/cpu/stm32/include/can_params.h
@@ -54,9 +54,15 @@ static const can_conf_t candev_conf[] = {
     	.rx_pin = GPIO_PIN(PORT_A, 11),
     	.tx_pin = GPIO_PIN(PORT_A, 12),
 #elif defined(CPU_FAM_STM32L4) || defined(CPU_FAM_STM32F4)
+#if  defined(BOARD_NUCLEO_L432KC)
+    	.rx_pin = GPIO_PIN(PORT_A, 11),
+    	.tx_pin = GPIO_PIN(PORT_A, 12),
+    	.af = GPIO_AF9,
+#else
     	.rx_pin = GPIO_PIN(PORT_B, 8),
     	.tx_pin = GPIO_PIN(PORT_B, 9),
     	.af = GPIO_AF9,
+#endif
 #else
     	.rx_pin = GPIO_PIN(PORT_D, 0),
     	.tx_pin = GPIO_PIN(PORT_D, 1),
```

#### Expected results

#### Actual results

#### Versions

```
Operating System Environment
----------------------------
         Operating System: macOS 14.6.1
                   Kernel: Darwin 23.6.0 x86_64 i386
             System shell: GNU bash, version 3.2.57(1)-release (x86_64-apple-darwin23)
             make's shell: GNU bash, version 3.2.57(1)-release (x86_64-apple-darwin23)

Installed compiler toolchains
-----------------------------
               native gcc: Apple clang version 15.0.0 (clang-1500.3.9.4)
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
                  avr-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
      riscv32-esp-elf-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp32s2-elf-gcc: missing
   xtensa-esp32s3-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: Apple clang version 15.0.0 (clang-1500.3.9.4)

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "4.1.0"
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
   riscv32-esp-elf-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp32s2-elf-newlib: missing
xtensa-esp32s3-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: ccache version 4.9.1
                    cmake: cmake version 3.29.3
                 cppcheck: Cppcheck 2.14.0
                  doxygen: 1.10.0 (GIT-NOTFOUND)
                      git: git version 2.45.0
                     make: GNU Make 3.81
                  openocd: Open On-Chip Debugger 0.12.0
                   python: Python 3.10.6
                  python2: missing
                  python3: Python 3.10.6
                   flake8: error: /Users/donsez/miniforge3/bin/python3: No module named flake8
               coccinelle: missing

```

### Misc

This [LoRa gateway board for cubesat](https://github.com/csu-grenoble/flatsat/tree/main?tab=readme-ov-file#board-1-obc--lora-gateway--can-bus) uses CAN bus for communication with other boards into the INISAT cubesat.

[image]

donsez, 2024-08-28:
> Note: the Nucleo32 boards [`nucleo-f303k8`](https://os.mbed.com/platforms/ST-Nucleo-F303K8/) and [`nucleo-f042k6`](https://os.mbed.com/platforms/ST-Nucleo-F042K6/) have also a CAN bus interface.

krzysztof-cabaj, 2024-08-28:
Hi! I looked at STM32L432KC datasheet and presented config (PA11/12 and AF9) seams reasonable.

I currently don't have nucleo-l432kc but if you would like, I could prepare PR with your changes. Or I could help you with submitting PR.

donsez, 2024-08-30:
Note Bene: I've tried to add `FEATURES_PROVIDED += periph_can` to the `Makefile.features` of the Nucleo32 boards [`nucleo-f303k8`](https://os.mbed.com/platforms/ST-Nucleo-F303K8/) and [`nucleo-f042k6`](https://os.mbed.com/platforms/ST-Nucleo-F042K6/) but the build of `tests/sys/conn_can` failed.

krzysztof-cabaj, 2024-08-30:
Hi! I try compile this code and errors are associated with various names of CMSIS defines in the `candev_conf` struct from `cpu/stm32/include/can_param.h`. For example for f303  `.can` should be `CAN` not `CAN1`, `.rcc_mask` should be `RCC_APB1ENR_CANEN` not `RCC_APB1ENR_CAN1EN` etc.. . To fix this some additional defines should be added - but this could lead to very hard to read/maintence code.

PR #20845 (2024-10-07): boards/nucleo-l432kc: enable CAN support
### Contribution description

This PR adds CAN support to the `nucleo-l432kc`board (see issue #20837) .

### Testing procedure

The testing procedure requires one Nucleo L432KC board, one Nucleo board having CAN interface (F446RE for instance) and 2 CAN drivers (such as [ST L9616](https://www.st.com/en/automotive-analog-and-power/l9616.html)).

```bash
cd tests/sys/conn_can
make BOARD=nucleo-l432kc flash term
```

Nucleo 1 (receiver)
```
init 0
set_mode 0 0
test_can recv 1 0 10000000 100 500
```

Nucleo 2 (sender)
```
init 0
set_mode 0 0
test_can send 0 [... cut]
closed 2024-10-07 by benpicco