RIOT-OS/RIOT #20948: build system: race condition when building and running an app for `native`

maribu, 2024-11-04:
#### Description

```
make BOARD=native -C tests/unittests all test -j
```

Seems to be racy: At least occasionally an old binary seems to be run. This is probably 

#### Steps to reproduce the issue

Run `make BOARD=native -C tests/unittests all test -j`, it should pass. Add e.g. `TEST_ASSERT_EQUAL_INT(1, 0);` to one of the tests and run `make BOARD=native -C tests/unittests all test -j`. This time it should fail, but it will pass again. Run `make BOARD=native -C tests/unittests all test -j` again, this time it will fail as expected.

#### Expected results

`make BOARD=native -C tests/unittests all test -j` should build and run the test using the freshly build executable.

#### Actual results

At least occasionally, the old binary is run instead.

#### Versions

Current `master`.

Enoch247, 2024-11-04:
Is this only true for the native board, or just only known to be true for native, and unproven/tests for other boards?

maribu, 2024-11-04:
I have never observed this on other boards. But there is an explicit `flash` target needed. Probably `tests` does correctly depend on `flash` (if that is among the goals), and `flash` correctly depends on the linked binary. `native` is special in the regard that `flash` is a no-op.

closed 2025-04-03 by maribu