RIOT-OS/RIOT #21079: Rust examples fail to build in local environment

mguetschow, 2024-12-13:
#### Description

On my local environment, installed according to the [documentation](https://doc.riot-os.org/using-rust.html#toolchain) (see Versions below), I am unable to build any of the three Rust examples. Tested for `native`, `native64` and `nrf52840dk` on current `master`.

The build works fine using `BUILD_IN_DOCKER=1`, where the Rust versions are listed below as well.

#### Steps to reproduce the issue
```
$ make -C examples/rust-hello-world BOARD=native clean all
...
  C2Rust binary c2rust, version: C2Rust 0.19.0
  Running C2Rust on /home/mikolai/TUD/Code/RIOT-build/examples/rust-hello-world/bin/native/target/i686-unknown-linux-gnu/release/build/riot-sys-bc2f2079a2410180/out/compile_commands.json
  Transpiling riot-c2rust.h
  cargo:warning=C2Rust failed with error code exit status: 101, exiting

  --- stderr
  riot-c2rust.h:273:33: error: expected expression
      int result = LED0_IS_PRESENT;
                                  ^
  riot-c2rust.h:308:33: error: expected expression
      int result = LED1_IS_PRESENT;
                                  ^
  2 errors generated.
  Error while processing /home/mikolai/TUD/Code/RIOT-build/examples/rust-hello-world/bin/native/target/i686-unknown-linux-gnu/release/build/riot-sys-bc2f2079a2410180/out/riot-c2rust.h.
  thread 'main' panicked at /home/mikolai/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.92/src/lib.rs:849:13:
  unsupported proc macro punctuation character '('
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

(the concrete problem here is likely that the macro is actually an empty define)

```
$ make -C examples/rust-hello-world BOARD=nrf52840dk clean all
...
  C2Rust binary c2rust, version: C2Rust 0.19.0
  Running C2Rust on /home/mikolai/TUD/Code/RIOT-build/examples/rust-hello-world/bin/nrf52840dk/target/thumbv7em-none-eabihf/release/build/riot-sys-61ecc0f43debd7a2/out/compile_commands.json
  Transpiling riot-c2rust.h
  cargo:warning=C2Rust failed with error code exit status: 101, exiting

  --- stderr
  In file included from riot-c2rust.h:80:
  /home/mikolai/TUD/Code/RIOT-build/sys/include/vfs.h:59:10: fatal error: 'sys/statvfs.h' file not found
  #include <sys/statvfs.h> /* for struct statvfs */
           ^~~~~~~~~~~~~~~
  1 error generated.
  Error while processing /home/mikolai/TUD/Code/RIOT-build/examples/rust-hello-world/bin/nrf52840dk/target/thumbv7em-none-eabihf/release/build/riot-sys-61ecc0f43debd7a2/out/riot-c2rust.h.
  thread 'main' panicked at /home/mikolai/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.92/src/lib.rs:849:13:
  unsupported proc macro punctuation character '('
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

(no idea where this comes from)

#### Versions
Local environment
```
rustc 1.83.0 (90b35a623 2024-11-26)
C2Rust 0.19.0
```

Docker environment
```
rustc 1.81.0 (eeb90cda1 2024-09-04)
C2Rust 0.19.0
```

plmorange, 2025-02-20:
Same on my side

chrysn, 2025-02-20:
Could you copy in one of the affected `out/riot-c2rust.h`? (Not sure how GitHub deals with huge texts; a `<details>\n\n'''\npaste\n'''\n\n</details>` with actually backticks usually works well, but that file is like 100KiB, so maybe that's too long for a comment … attachment?)

plmorange, 2025-02-21:
For rust hello world.
[riot-c2rust.zip](https://github.com/user-attachments/files/18903101/riot-c2rust.zip)

chrysn, 2025-02-21:
The errors on `LEDx_IS_PRESENT` appear to be red herrings: C2Rust sadly does print non-fatal "errors" but continues fine.

The actual issue is https://github.com/dtolnay/proc-macro2/issues/475, solved in c2rust after 0.19 as https://github.com/immunant/c2rust/pull/1197.

The workaround I'd recommend is installing c2rust again but passing `--locked` to the cargo install line until c2rust releases again. Docs are being updated in #21236.

closed 2025-02-21 by mguetschow