RIOT-OS/RIOT #21098: sys/usb/usbus/cdc/ecm: inerface thread blocked when connected host interface is down

fabian18, 2024-12-20:
#### Description

USBUS cdcecm thread stalls when sending ping and the receiving interface on Linux is down.

I expected an interrupt to happen after a timeout if a transaction fails. In the MCU datasheet I found the `TRFAIL` interrupt which was not enabled yet. But with this patch when I try to enable it, it is fireing all the time and does not solve the problem.

[cdcecm.patch.txt](https://github.com/user-attachments/files/18211703/cdcecm.patch.txt)

#### Steps to reproduce the issue

Flash `examples/gnrc_networking` to  `same54-xpro` withe `USEMODULE+="usbus_cdc_ecm auto_init_usbus"`

```
2024-12-20 14:53:01,433 # ifconfig
2024-12-20 14:53:01,437 # Iface  6  HWaddr: FC:C2:3D:2F:8E:90  Link: down 
2024-12-20 14:53:01,442 #           L2-PDU:1500  MTU:1500  HL:64  RTR  
2024-12-20 14:53:01,445 #           Source address length: 6
2024-12-20 14:53:01,447 #           Link type: wired
2024-12-20 14:53:01,448 #           
2024-12-20 14:53:01,451 #           Statistics for Layer 2
2024-12-20 14:53:01,454 #             RX packets 0  bytes 0
2024-12-20 14:53:01,458 #             TX packets 0 (Multicast: 0)  bytes 0
2024-12-20 14:53:01,462 #             TX succeeded 0 errors 0
2024-12-20 14:53:01,464 #           Statistics for IPv6
2024-12-20 14:53:01,467 #             RX packets 0  bytes 0
2024-12-20 14:53:01,471 #             TX packets 0 (Multicast: 0)  bytes 0
2024-12-20 14:53:01,475 #             TX succeeded 0 errors 0
2024-12-20 14:53:01,475 # 
2024-12-20 14:53:01,478 # Iface  7  HWaddr: FE:76:BD:F3:19:D8 
2024-12-20 14:53:01,483 #           L2-PDU:1500  MTU:1500  HL:64  RTR  
2024-12-20 14:53:01,486 #           Source address length: 6
2024-12-20 14:53:01,488 #           Link type: wired
2024-12-20 14:53:01,494 #           inet6 addr: fe80::fc76:bdff:fef3:19d8  scope: link  VAL
2024-12-20 14:53:01,497 #           inet6 group: ff02::2
2024-12-20 14:53:01,499 #           inet6 group: ff02::1
2024-12-20 14:53:01,503 #           inet6 group: ff02::1:fff3:19d8
2024-12-20 14:53:01,504 #           
2024-12-20 14:53:01,507 #           Statistics for Layer 2
2024-12-20 14:53:01,510 #             RX packets 14  bytes 2360
2024-12-20 14:53:01,514 #             TX packets 4 (Multicast: 4)  bytes 202
2024-12-20 14:53:01,518 #             TX succeeded 0 errors 0
2024-12-20 14:53:01,520 #           Statistics for IPv6
2024-12-20 14:53:01,524 #             RX packets 11  bytes 1216
2024-12-20 14:53:01,528 #             TX packets 4 (Multicast: 4)  bytes 224
2024-12-20 14:53:01,531 #             TX succeeded 4 errors 0

```

Interface `7` is `cdcecm`.

Set interface on Linux side down:

```
$ sudo ip link set enxfe76bdf305d8 down
9: enxfe76bdf305d8: <BROADCAST,MULTICAST> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether fe:76:bd:f3:05:d8 brd ff:ff:ff:ff:ff:ff
```

Send ping over `cdcecm` interface.

```
> ping ff02::1%7
2024-12-20 14:55:50,706 # ping ff02::1%7
2024-12-20 14:55:53,706 # 
2024-12-20 14:55:53,709 # --- ff02::1%7 PING statistics ---
2024-12-20 14:55:53,714 # 3 packets transmitted, 0 packets received, 100% packet loss
```

Another `ifconfig` stalls and interface `7` doe not show up.

```
2024-12-20 14:56:56,129 # ifconfig
2024-12-20 14:56:56,133 # Iface  6  HWaddr: FC:C2:3D:2F:8E:90  Link: down 
2024-12-20 14:56:56,138 #           L2-PDU:1500  MTU:1500  HL:64  RTR  
2024-12-20 14:56:56,141 #           Source address length: 6
2024-12-20 14:56:56,143 #           Link type: wired
2024-12-20 14:56:56,144 #           
2024-12-20 14:56:56,147 #           Statistics for Layer 2
2024-12-20 14:56:56,150 #             RX packets 0  bytes 0
2024-12-20 14:56:56,154 #             TX packets 0 (Multicast: 0)  bytes 0
2024-12-20 14:56:56,157 #             TX succeeded 0 errors 0
2024-12-20 14:56:56,160 #           Statistics for IPv6
2024-12-20 14:56:56,163 #             RX packets 0  bytes 0
2024-12-20 14:56:56,167 #             TX packets 0 (Multicast: 0)  bytes 0
2024-12-20 14:56:56,170 #             TX succeeded 0 errors 0
2024-12-20 14:56:56,171 # 
```

The second send blocks the interface thread because there
https://github.com/RIOT-OS/RIOT/blob/4f55461e159a1e80e4584c207e81750543d7ffe3/sys/usb/usbus/cdc/ecm/cdc_ecm_netdev.c#L78
a mutex is locked twice without being unlock after a successful transfer, because there is no successful transfer.

#### Expected results

Even thought the Linux interface is down, RIOT should not stall after sending a ping.

#### Actual results

RIOT stalls because a mutex is not unlocked, when sending over a `cdcecn` interface when the Linux interface is down.

#### Versions

fabian18, 2024-12-20:
When the Linux interface is up again the mutex is unlocked.

dylad, 2025-01-03:
Looks like this is a more general issue as I was able to reproduce it on `nrf5340dk`
I'll try to investigate this.