RIOT-OS/RIOT #21106: Build System: Update Doxygen to >=1.10.0 to fix URLs in Headlines

crasbe, 2024-12-23:
#### Description

While updating the `LOSTANDFOUND.md` file, I noticed that the headlines are not correctly formatted. After some (a lot) of searching I found out that this is a Doxygen bug ( https://github.com/doxygen/doxygen/issues/10466 ) which has been fixed in version >=1.10.0 ( https://github.com/doxygen/doxygen/issues/10466#issuecomment-1848646035 ).

I'm not sure if that was a regression introduced to Doxygen at some point or if it never looked right (which I doubt).
Perhaps something in the RIOT documentation will break when updating Doxygen. For testing, I installed the lastest 1.12.0 and it complained about some configuration options being deprecrated. However my Doxygen knowledge is quite limited, so this might have some side effects or additional effort attached to it.

#### Current Behavior

This is a screenshot taken from https://doc.riot-os.org/md_LOSTANDFOUND.html

[image]

#### Expected Behaviour

This was rendered on my machine with Doxygen 1.12.0:

[image]

miri64, 2025-01-21:
I mainly assigned @OlegHahm to figure out, if this somehow conflicts with the cronjob that generates https://doc.riot-os.org/.

crasbe, 2025-01-21:
The main headache with this issue is that several Linux distributions don't ship Doxygen 1.10.0 yet. For example, this is the current state for Ubuntu (which I think the CI uses?) https://launchpad.net/ubuntu/+source/doxygen
Therefore it would have to be installed manually, but I don't know much about how the CI works.

The current version of Doxygen is 1.13.2 by the way and the old 1.9.8 is from late 2023 🫠
For some packages the "old is gold" strategy from Debian-based distributions can be quite annoying, but that's beside the point 😅

OlegHahm, 2025-01-21:
I just updated the docker container for the build job to the latest Alpine version and doxygen is now at 1.12.0. I also re-delivered the latest webhook that triggers building the documentation. As far as I can tell that seems to have worked.

crasbe, 2025-01-21:
It looks fixed indeed. Thank you @OlegHahm :)

miri64, 2025-01-21:
> I just updated the docker container for the build job to the latest Alpine version and doxygen is now at 1.12.0. I also re-delivered the latest webhook that triggers building the documentation. As far as I can tell that seems to have worked.

Does this also fix the generation by Murdock on PRs?

OlegHahm, 2025-01-21:
Not sure I know the entire workflow. There is a webhook called by Github on any push which will trigger to rebuild the documentation. I don't know where (and for what reason) Murdock is involved. Is there some sort of a preview for PRs?

crasbe, 2025-01-21:
Worst case we'll have to wait for a day or two and see once the next PR is merged.

miri64, 2025-01-22:
> Is there some sort of a preview for PRs?

Yes, see e.g. https://github.com/RIOT-OS/RIOT/pull/21154#issuecomment-2605289823

miri64, 2025-01-22:
I _think_ this is the dockerfile for the container of the murdock build https://github.com/RIOT-OS/murdock-scripts/blob/master/Dockerfile. With the actual build happening here https://github.com/RIOT-OS/murdock-scripts/blob/master/build.sh#L279-L283. The webhook is only for pushes to master.

OlegHahm, 2025-01-22:
The Dockerfile in murdock is based on Ubuntu 22.04 where doxygen is apparently still the 1.9 version. Not sure if updating this dockerfile is easily possible.

OlegHahm, 2025-01-22:
But even in newer Ubuntu versions the doxygen package seems to stick to this version.

miri64, 2025-01-22:
Maybe we can delegate the Murdock build to your container somehow?

OlegHahm, 2025-01-22:
On Wed, Jan 22, 2025 at 04:14:35AM -0800, Martine Lenders wrote:
> Maybe we can delegate the Murdock build to your container somehow?

The question might be: is it important that the docs build on developers'
hosts as well.

crasbe, 2025-01-22:
>> On Wed, Jan 22, 2025 at 04:14:35AM -0800, Martine Lenders wrote:
>>  Maybe we can delegate the Murdock build to your container somehow?

> The question might be: is it important that the docs build on developers'
> hosts as well.

The current state is that you can't predict the doxygen version that developers have. It all depends on their distribution, most likely it'll be Doxygen 1.9.8 for Ubuntu users and maybe a little newer for other distributions.

We could add a remark to https://github.com/RIOT-OS/RIOT/blob/master/CONTRIBUTING.md#writing-documentation that `doxygen >= 1.10.0` is highly recommended. Considering the update speed of Ubuntu, that'll be relevant for at least a couple of years.

crasbe, 2025-02-27:
This fixes Issue #17063 (making this technically a duplicate).

Baertig, 2025-03-07:
I spoke with @mguetschow, and we think it would be best, if the preview documentation build in the CI uses the same Doxygen version as the container generating the documentation for the website https://doc.riot-os.org/. 

I think the simplest solution is to add a GitHub workflow, that uses the same container image ([docker-doxygen](https://github.com/OlegHahm/docker-doxygen)) to build a preview of the docs. 
This way we would not have to build the documenation inside murdock (here https://github.com/RIOT-OS/murdock-scripts/blob/master/build.sh#L279-L283) and only need to maintain the doxygen version in one image.

I'll look into it next week.