RIOT-OS/RIOT #21150: GOVERNANCE.md: Do we need a process to appoint moderators for services other than the forum?

miri64, 2025-01-21:
Currently the [GOVERNANCE.md](https://github.com/RIOT-OS/RIOT/blob/master/GOVERNANCE.md) only describes the process to appoint forum moderators since they have special technical moderation powers (editing posts, splitting out discussions into separate topics, blocking users, filtering spam, …). Should we also have also a process for the following services? Should these moderation role also be expanded to the more social responsibilities of moderators? 

- Matrix: Here the moderation powers only go as far as blocking/kicking out users
- Mailing lists: Here the moderation powers only go as far as filtering spam

## Useful links

- https://github.com/RIOT-OS/RIOT/blob/master/GOVERNANCE.md#moderators
- https://forum.riot-os.org/t/moving-from-community-processes-to-governance-md/4433/2
- https://forum.riot-os.org/t/moving-from-community-processes-to-governance-md/4433/9#p-16566-comment-11-2

maribu, 2025-01-21:
I think documenting the process has value. If that just is: "Maintainers decide whom to grant the following roles: Forum moderator, ..." that is little effort to write down. People seeking to gain that role then would know where to knock and ask.

How were those roles handed out so far? Let's just write that down with a sentence or two.

miri64, 2025-01-21:
> If that just is: "Maintainers decide whom to grant the following roles: Forum moderator, ..." that is little effort to write down.

That's picking the worst example. Especially forum moderators (for which we already have a process described) should be granted by the community at large, not just the maintainers.

miri64, 2025-01-21:
> How were those roles handed out so far?

See https://github.com/RIOT-OS/RIOT/pull/21067#discussion_r1878347438. Also added that comment to the Useful links section above.