RIOT-OS/RIOT #21187: I2C Crashes Kernel on Raspberry Pi Pico

tanneberger, 2025-02-04:
#### Description

I want to use the [veml7700](https://www.vishay.com/docs/84286/veml7700.pdf) brightness sensor in combination with the Raspberry Pi PICO. Sadly, RIOT crashes after calling `i2c_init` (see the code below). 

I tested the same code on the `feather-nrf52840-sense` there it works flawlessly. 

```c
int main(void) {
    i2c_t dev = I2C_DEV(0);
    i2c_init(dev); // crashes here

    uint16_t reg_value = 0;
    uint16_t device_addr = 0x10;

    i2c_acquire(dev);

    uint16_t conf_val1 = 6144;
    printf("writing value: %i", conf_val1);

    i2c_write_byte(dev, device_addr, 0, I2C_NOSTOP);
    i2c_write_bytes(dev, device_addr, &conf_val1, 2, 0);

    while (true) {
        i2c_read_regs(dev, device_addr, 0x04, &reg_value, 2, 0);
        printf("%i\n", reg_value);
    }

    i2c_release(dev);
    return 0;
}
```

Output I get by connecting a logic analyzer: 

[image]

#### Steps to reproduce the issue
In the Makefile, I add the line `USEMODULE += periph_i2c`  in order to use i2c. Just compile this program for the `BOARD=rpi-pico`

#### Expected results
Don't crash. 

#### Versions
I run this on RIOT version 2024.10 (`b51f093a395bdd575cff94b7d26be415fa46771d`)

maribu, 2025-02-04:
Thx for the report. Could you try again without `i2c_init()`?

(Alternatively you could disable the auto initialisation of the I2C buses and then do by hand. I'm afk, but I think the module is named `periph_init_i2c` that does the auto initialisation. You can add `DISABLE_MODULE += periph_init_i2c` to the apps Makefile for that; assuming that I correctly nemorized the module name.)

dylad, 2025-02-04:
Hi @tanneberger 
As @maribu suggested, you can try without `i2c_init()` as this part is in general done automatically for you by RIOT.
Except if you use `DISABLE_MODULE += periph_init_i2c`
You can check this with `make -C path/to/your/app info-build ` if you are uncertain. (Look at the `USEMODULE` / `DISABLE_MODULES`).

However, a second call to `i2c_init()` should not hurt that much so there is definitely something wrong underneath.

maribu, 2025-02-04:
> However, a second call to `i2c_init()` should not hurt that much so there is definitely something wrong underneath.

That is not generally true. Most MCU peripherals are initialized in a way that the code is naturally idempotent. But some periph drivers rely on being called exactly once.

Our documentation in again substandard here, but in https://doc.riot-os.org/group__drivers__periph.html#autotoc_md1719 it is hinted.

dylad, 2025-02-04:
Perhaps we could keep track of the state of the peripheral and early return if the initialization was already done or trigger an assert.
A Kernel panic is a high price to pay for a periph reinit.

maribu, 2025-02-04:
@tanneberger: This issue to be auto-closed upon merge, rather than by intent, because I wrote `maybe fixes #21187` in PR https://github.com/RIOT-OS/RIOT/pull/21188.

Could you you report back if the issue was indeed the second call to `i2c_init()`?

And sorry for the doc being wrong here :/

miri64, 2025-02-04:
> Could you you report back if the issue was indeed the second call to `i2c_init()`?

Let's re-open the issue until then.

tanneberger, 2025-02-07:
@dylad @miri64 @maribu 

Removing the `i2c_init` stops the kernel from crashing, but I also cannot find the i2c signal on any pins. 

I also tried adding `DISABLE_MODULE += periph_init_i2c` to my Makefile and then uncommented the `i2c_init` and the kernel then crashed again.

maribu, 2025-02-07:
Thx for reporting. Maybe I can find a bit of time to look into it this weekend.

crasbe, 2026-10-06:
The reason for the crash is simple. The board does not define any pins for the PIO I2C, the comments say that the user has to do it during the build process:
https://github.com/RIOT-OS/RIOT/blob/ca8d9d05f26bd124e232204edb741fcbe4348c13/boards/rpi-pico/include/periph_conf.h#L157-L174

Without that, the assertion will blow because `I2C_NUMOF` is set to zero. It should've used `I2C_PIO_NUMOF`.
https://github.com/RIOT-OS/RIOT/blob/ca8d9d05f26bd124e232204edb741fcbe4348c13/cpu/rpx0xx/periph/i2c.c#L30-L34

I'll provide a PR that fixes that and provides a default build configuration, as we usually do.

Claude Code found that for me.

closed 2026-10-21 by miri64