RIOT-OS/RIOT #21205: dist/tools/uf2: Flashing a board with RIOT seems to prevent flashing other UF2 files afterwards

mguetschow, 2025-02-11:
#### Description

After having flashed `BOARD=feather-nrf52840-sense` at least once with RIOT, I was no longer able to install CircuitPython on it. For a new and untouched board of the same type it works as described on https://learn.adafruit.com/adafruit-feather-sense/circuitpython-on-feather-sense

Also reproduced with `BOARD=seeedstudio-xiao-nrf52840` from #20980.

#### Steps to reproduce the issue
1. Take a board using the UF2 bootloader, e.g. `BOARD=feather-nrf52840-sense`.
2. Flash some RIOT application on it (e.g., `make -C examples/hello-world flash`)
3. Try to flash the UF2 file from https://circuitpython.org/board/feather_bluefruit_sense/ by copying it to the USB device from bootloader mode
4. See no error, but notice that CIRCUITPY does not appear.
5. Also see, that the serial connection via `make term` still shows the RIOT greeting.

#### Expected results
CircuitPython starts and can be used as expected.

#### Actual results
RIOT still remains on the board.

#### Versions
tested with current RIOT master, but also happened with boards previously running some older version of RIOT

crasbe, 2025-02-11:
Oh RIOT and the Adafruit bootloader is not a story of love...

Without looking into it too much I _assume_ the reason for that is that RIOT overwrites the SoftDevice:

https://github.com/RIOT-OS/RIOT/blob/7027fd3bbb5af6e799d3d4338b7d059f88cf5226/boards/feather-nrf52840/Makefile.include#L9-L13

(The `Makefile.include` of the `feather-nrf52840-sense` is just a symlink: https://github.com/RIOT-OS/RIOT/blob/master/boards/feather-nrf52840-sense/Makefile.include )

I played around with this a lot when I tried to get the OTA feature of the Adafruit bootloader working. You have to set the `ROM_OFFSET` to `0x26000` in order not to overwrite the SoftDevice: https://github.com/adafruit/Adafruit_nRF52_Bootloader/?tab=readme-ov-file#making-your-own-uf2

But now that RIOT _did_ overwrite the SoftDevice, you have to reflash the Adafruit bootloader. I'm not sure if you can do that with the bootloader itself or if you have to use SWD.

crasbe, 2025-02-11:
You can verify that the SoftDevice is gone if you look into the `INFO_UF2.TXT` file on the virtual file system, when you enabled the bootloader by writing the `bootloader` command in the serial console:

```
UF2 Bootloader 0.9.2-1-g19b27ab lib/nrfx (v2.0.0) lib/tinyusb (0.12.0-145-g9775e7691) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: xxxxxxxxxxxxxxxxxxx
Board-ID: xxxxxxx
Date: Jan 20 2025
SoftDevice: not found

```

Model and ID are different in my case because I only have the Adafruit bootloader installed on a custom board. I don't have a feather-nrf52840 or similar. I have a bluefruit nrf52832 which neither has the UF2 bootloader nor is supported by RIOT. 😅

The UF2 bootloader _should_ have a serial logging output where it might tell you some details about why the flashing failed. The baud rate is 115200.

crasbe, 2025-02-11:
Thinking about it... The Bootloader probably does flash CircuitPython successfully, but to addresses `0x26000` and higher. The starting position is defined when creating the UF2 file and Adafruit probably keeps the SoftDevice. After finishing the flashing, the bootloader jumps to `0x1000` to start the SoftDevice, but that's long gone and RIOT is still there. If the RIOT binary is small enough, it wouldn't even be damaged by flashing CircuitPython.

crasbe, 2025-02-12:
I remembered that I bought a Xiao nRF52840 board some time ago and yes, I can reproduce the behavior and there is a procedure to "unbrick" it.

You have to reflash the Bootloader, which fortunately can be done via DFU: https://github.com/adafruit/Adafruit_nRF52_Bootloader/?tab=readme-ov-file#flash

For the xiao, you have to use the command `BOARD=xiao_nrf52840_ble make flash-dfu SERIAL=/dev/ttyACM0` (or another serial port if you're using Mac).
In my case, I had to do it twice for some reason. The first time it seemed to have installed the SoftDevice, but the bootloader was still the old version. The second time, it updated the bootloader as well.

After that I was able to upload the CircuitPython UF2 file and it runs successfully.

mguetschow, 2025-02-12:
> In my case, I had to do it twice for some reason.

Wow, yes that was it! I had tested reflashing the bootloader but apparently only once. After the second flashing, it works here too.

Maybe something worth reporting at https://github.com/adafruit/Adafruit_nRF52_Bootloader/ and definitely something to add to our documentation. Thanks a lot for your help! 👍

crasbe, 2025-02-12:
We could think about whether or not it is necessary that RIOT overwrites the SoftDevice in the first place.

If you have some time, you could tinker around with the `ROM_OFFSET` and `ROM_LEN` to see if that fixes the issue. The SoftDevice seems to occupy about 150KB of Flash, but the nRF52840 has 1MB of Flash. So 99% of the applications would still fit.

crasbe, 2025-02-27:
In a previous post, I wrote that the SoftDevice starts at `0x26000`, but that is incorrect. The Xiao uses SoftDevice version 7, so the entry point is at `0x27000`.

During testing with the new Seeedstudio Xiao nRF52840 Sense, it became apparent that removing the SoftDevice also prevents Arduino from working. This really shouldn't be the standard behavior IMO.

closed 2025-03-25 by mguetschow