RIOT-OS/RIOT #21209: cpu/native: Out-of-bounds memory access in startup constructor loop

carl-tud, 2025-02-12:
In [cpu/native/startup.c](https://github.com/RIOT-OS/RIOT/blob/master/cpu/native/startup.c#L690-L709), constructors in the init array following the RIOT constructor are invoked. This is done it two loops: The first tries to find the RIOT startup constructor, the second executes all trailing constructors. However, if the init array is empty (start == end), the first loop will find nothing, yet will still increment the cursor. Both loops do not check array bounds properly (!=) and dereference pointers read from memory, with the second additionally calling that pointer. Result: undefined behavior, segfaults, code execution.

Fix: #....

mguetschow, 2025-02-14:
> the `(uintptr_t)init_array_ptr == (uintptr_t)&__init_array_end` check

Do you mean the _un_equal check?

https://github.com/RIOT-OS/RIOT/blob/8aa08599ccf1572acfac693169be76021400d781/cpu/native/startup.c#L692

> If you do a `(uintptr_t)init_array_ptr < (uintptr_t)&__init_array_end` check instead, a `cmp` instruction is inserted into the binary... Result: undefined behavior, segfaults, code execution.

I guess you mean the opposite: Without the `cmp` instruction you get undefined behavior?

> Might be related to the __init_array_start and end asm references for the linker

Could it be that the root cause lies in the macOS linker script?

carl-tud, 2025-02-15:
>> the (uintptr_t)init_array_ptr == (uintptr_t)&__init_array_end check
> Do you mean the _un_equal check?
Yes, sorry

>> If you do a (uintptr_t)init_array_ptr < (uintptr_t)&__init_array_end check instead, a cmp instruction is inserted into >> the binary... Result: undefined behavior, segfaults, code execution.
> I guess you mean the opposite: Without the cmp instruction you get undefined behavior?
Yes, with `!=` (w/o cmp), the check is optimized away, leading to 'undefined' behavior

>> Might be related to the __init_array_start and end asm references for the linker
> Could it be that the root cause lies in the macOS linker script?
That's what I suspect, too. Apple's DYLD launcher also uses a smaller-than check. However, maybe this issue also applies to GNU..., and you aren't supposed to use equals on symbols generated by the linker (or, more specifically, init array markers)

Basically just wanted to have this written down somewhere in case someone else also stumbles across weird behavior causes be the inequality check...

carl-tud, 2025-03-13:
Fixed in #21283

closed 2025-03-13 by carl-tud