RIOT-OS/RIOT #21220: doc: Use comment blocks correctly in doc.txt files

crasbe, 2025-02-17:
As discovered in #21218, some `doc.txt` files use an incorrectly formatted `doc.txt` file. In these files, the Doxygen documentation is essentially within a C comment. However many files omit the leading asterisk which can cause Doxygen to incorrectly parse Markdown statements for bold and italic at a line start.

Incorrect:
```c
/**
 * @defgroup ...

Markdown here
*Bold ignored* because Doxygen removes leading stars.

*/
```

Correct:
```c
/**
 * @defgroup ...
 *
 * Markdown here
 *
 * *Bold not ignored* because Doxygen has a leading star to remove.
 */
```

This is outlined in the Doxygen documentation as well: 
https://www.doxygen.nl/manual/markdown.html#mddox_emph_spans
https://www.doxygen.nl/manual/markdown.html#mddox_stars

With this shell command `find . -name "doc.txt" -exec grep -lE '^[^ /]' {} +` I compiled a list of *potential* candidates. It searches for files that have lines which don't begin with a space or a slash. This can generate false positives, because some files have `#ifdefs`, but it gives something to start with:

<details><summary>List of potential candidates</summary>

```
RIOT$ find . -name "doc.txt" -exec grep -lE '^[^ /]' {} +
./cpu/msp430/doc.txt
./cpu/stm32/doc.txt
./cpu/esp32/esp-eth/doc.txt
./cpu/esp32/doc.txt
./cpu/cc26xx_cc13xx/doc.txt
./cpu/esp8266/doc.txt
./cpu/qn908x/doc.txt
./cpu/doc.txt
./cpu/rpx0xx/doc.txt
./cpu/kinetis/doc.txt
./cpu/esp_common/esp-wifi/doc.txt
./cpu/esp_common/esp-now/doc.txt
./bootloaders/riotboot_serial/doc.txt
./bootloaders/riotboot_tinyusb_dfu/doc.txt
./bootloaders/riotboot/doc.txt
./bootloaders/riotboot_dfu/doc.txt
./doc.txt
./pkg/uwb-core/doc.txt
./pkg/mynewt-core/doc.txt
./pkg/openwsn/doc.txt
./pkg/lv_drivers/doc.txt
./pkg/lvgl/doc.txt
./pkg/fatfs/doc.txt
./pkg/tinydtls/doc.txt
./boards/remote-reva/doc.txt
./boards/cc1352p-launchpad/doc.txt
./boards/esp32-heltec-lora32-v2/doc.txt
./boards/nucleo-g071rb/doc.txt
./boards/b-u585i-iot02a/doc.txt
./boards/arduino-mkrzero/doc.txt
./boards/feather-m0/doc.txt
./boards/stm32f4discovery/doc.txt
./boards/stm32f3discovery/doc.txt
./boards/seeeduino_arch-pro/doc.txt
./boards/nucleo-f070rb/doc.txt
./boards/spark-core/doc.txt
[... 561 lines cut ...]
- [ ] `./cpu/esp_common/esp-wifi/doc.txt`
- [ ] `./cpu/kinetis/doc.txt`
- [ ] `./cpu/msp430/doc.txt`
- [ ] `./cpu/qn908x/doc.txt`
- [ ] `./cpu/rpx0xx/doc.txt`
- [ ] `./cpu/stm32/doc.txt`
- [ ] `./doc.txt`
- [ ] `./drivers/atwinc15x0/doc.txt`
- [ ] `./drivers/ccs811/doc.txt`
- [ ] `./drivers/lsm303agr/doc.txt`
- [ ] `./drivers/sht3x/doc.txt`
- [ ] `./pkg/fatfs/doc.txt`
- [ ] `./pkg/lv_drivers/doc.txt`
- [ ] `./pkg/lvgl/doc.txt`
- [ ] `./pkg/mynewt-core/doc.txt`
- [ ] `./pkg/openwsn/doc.txt`
- [ ] `./pkg/tinydtls/doc.txt`
- [ ] `./pkg/uwb-core/doc.txt`
- [ ] `./sys/cpp_new_delete/doc.txt`
- [ ] `./sys/cxx_ctor_guards/doc.txt`
- [ ] `./sys/malloc_thread_safe/doc.txt`
- [ ] `./sys/net/ble/bluetil/doc.txt`
- [ ] `./sys/net/ble/doc.txt`
- [ ] `./sys/net/sock/doc.txt`
- [ ] `./sys/shell/doc.txt`

crasbe, 2025-02-17:
I went through the list and unfortunately the number of false positives is quite small 😅

kfessel, 2025-02-28:
we might be able to just change to doc.md 

instead of doxygen-commenting everything

crasbe, 2025-02-28:
That is indeed probably a better approach. Where to start though?

Add the new approach to `doc/doxygen/src/porting-boards.md`?

crasbe, 2025-03-05:
There is an issue...

[image]

As you can see in the screenshot, doxygen will generate an entry in the list of contents for every Markdown file and currently there is no way to disable this: https://github.com/doxygen/doxygen/issues/9437

This is not so clear from the documentation: https://www.doxygen.nl/manual/markdown.html
```
Each file is converted to a page (see the [page](https://www.doxygen.nl/manual/commands.html#cmdpage) command for details).
```

I will ask the Doxygen maintainer to take a look at this issue again, perhaps there is an easy solution in the Doxygen source.

kfessel, 2025-03-05:
could we somehow workaround that without having the c-doxygen-comment in repository, 

keep and create doc.md (without any `/**` ) and preprocess  them to doc.txt whith `*` in every line `/**` at BEGIN and `*/` at EOF (sound like a small AWK script to me)

crasbe, 2025-03-06:
My AWK skills are limited to "asking ChatGPT", but it sounds like a good temporary solution. However I'm not sure how to temporarily change the files and then change it back when `make doc` is complete.

mguetschow, 2025-03-06:
> keep and create doc.md (without any `/**` ) and preprocess them to doc.txt whith `*` in every line `/**` at BEGIN and `*/` at EOF (sound like a small AWK script to me)

The problem with this approach is that we cannot tell doxygen to ignore the existing `doc.md` files.

See #21273 as another temporary fix.