RIOT-OS/RIOT #21232: Add board support for Arduino Nano ESP32

donsez, 2025-02-20:
#### Description

Arduino Nano ESP32 is a ESP32-S3 SoC with the Arduino Nano pinout.

RIOT already supports several ESP32-S3 boards.

### Useful links

* https://doc.riot-os.org/group__boards__esp32s3.html
* https://docs.arduino.cc/hardware/nano-esp32/
* https://docs.arduino.cc/resources/datasheets/ABX00083-datasheet.pdf

shibcreate, 2025-08-10:
Hello @donsez, I'd like to help support this issue. I have the board on hand. Have there been any updates regarding its integration?

gschorcht, 2025-08-11:
I have ordered the fully compatible [Wavshare ESP32-S3-Nano Board] (https://www.waveshare.com/product/arduino/boards-kits/arduino/esp32-s3-nano.htm) that will arrive tomorrow. I will try out a draft version of a board definition over the next few days.