RIOT-OS/RIOT #21288: Doc: `*.doc.md` in addition to to `doc.md`

carl-tud, 2025-03-12:
Currently, we can put documentation in `doc.md` files, and even define groups inside them. I really like this way of documenting code without needing to add a huge documentation block at the start of a given header. See #21273.

I would like to add documentation to XFA without making the header grow to an unmanageable size. Putting a global `doc.md` into `core/lib` doesn't seem like a good fit to me. If there's some other component in `core/lib` someone wants to document more extensively, that would also need to go into the same doc.md there. Hence, I don't think the `doc.md` approach scales well here.

I think this could be solved by also allowing `*.doc.md` files, such as `xfa.doc.md` in `core/lib`.
This would be the necessary change in the doxyfile:
```
FILE_PATTERNS  = \
  doc.md \
  *.doc.md # <- here
```

Does anyone have thoughts on this?

ping @mguetschow

crasbe, 2025-03-12:
I like the idea, especially if it leads to better documentation :)

One thing I am unsure about is how to ensure that everything has a comprehensible hierachy. Currently the group `core_util` is defined in `core/doc.txt` and `xfa` belongs to that group.
Should the same be applied to the `core/lib` subfolder? That there is a `core/lib/doc.md` that defines the group `core_util_xfa` and `core/lib/xfa.doc.md` is part of that group?

Another point is the risk that the documentation goes out of sync. The more files there are, the higher the chance that one file is forgotten.

<hr/>

Perhaps @kfessel has some opinion about this?

kfessel, 2025-03-14:
I also like that idea

we may need to crosscheck with our current doxygen version (as our current workaround only covers "doc.md" but i think it might be extendable)

carl-tud, 2025-03-16:
... or, another idea, we dedicate a folder pattern to documentation, like `ThisIsADocumentationFolder.doc` or `Core.doc` and every markdown file inside these folders ending in `.doc` is considered as input for doxygen.

crasbe, 2025-03-16:
That folder structure would only be worth it for *massive* amounts of documentation IMO.
Also it's usually a good idea to keep the documentation as close to the source as possible.

mguetschow, 2025-03-17:
> Also it's usually a good idea to keep the documentation as close to the source as possible.

I second this. While I do see the point of moving more high-level user-guide-like documentation out of the header files (with the added benefit of being able to write (and preview on Github!) plain markdown files, but I would rather avoid a separate folder structure.

Maybe we could have the file naming convention include/refer to the doxygen group that it belongs to.

kfessel, 2025-03-17:
I think the usefullness of a seperate eg. mutex.doc.md start at cases like the first ~ 120 of https://github.com/RIOT-OS/RIOT/blob/master/core/include/mutex.h 

keeping them close to the related files either header -> or even implementation (close to c file). 

this way the in header docu can focus on api documentation.

i also think we should avoid extra folders for realy over reachin documentation we allready, got that with the doc folder

kfessel, 2025-03-17:
maybe we should go with *doxy.md 
this would indicate that the file is meat to have doxygen marks and doxygen compatible markdown and no other

carl-tud, 2025-03-22:
> maybe we should go with *doxy.md this would indicate that the file is meat to have doxygen marks and doxygen compatible markdown and no other

Okay, but we don't use a documentation generator other than doxygen, so `.doc.md` would imply the same. 

I would also prefer `.doc.md` over `.doxy.md` because it aligns better with the file's purpose imo: it's a Markdown file containing documentation. The majority of it is valid markdown. Only a very tiny parts use 'doxygen' \@nnotations.

Plus, I'd say the 'doxygen' markup or dialect is not doxygen-only... it's a superset of HeaderDoc, and the syntax itself is universally understood. Maybe another documentation generator using the exactly same syntax emerges down the road, so `.doc.md` would also be future-proof. And `.doc.md` would be consistent with our previous approach of having `doc.txt` files. 

I mean, we don't name C files `main.gcc` or `main.clang` either -- even if their contents contain compiler-specific attributes. What I'm trying to say is that a file name should, in my opinion, reflect the _type of content_ rather than the _compiler used to generate another representation from it_. In my view, that's also the reason we don't append `.doxy.c` or `.javadoc.java` to source files just because of the flavor of documentation comments it contains.

chrysn, 2025-03-22:
I don't think we need to point out "doxygen" style markdown. Yes, markdown is a big stinkin' mess, but it's spread throughout all our documentation.

(Frankly I don't see why to go with .doc.md rahter than .md -- it is text intended for human reader consumption inside the source code, what else should it be other than documentation?)

mguetschow, 2025-03-22:
> (Frankly I don't see why to go with .doc.md rahter than .md -- it is text intended for human reader consumption inside the source code, what else should it be other than documentation?)

One argument for going with `*.doc.md` is that it would align well with generic `doc.md` files that have inferred semantic content from the enclosing folder.

(Another minor one is that the CSS hack from #21312  would not work anymore.)

I also favor `*doc.md` in favor of `*doxy.md`.

maribu, 2025-03-23:
+1 for `doc.md` over `doxy.md`. Especially since members of our community actively are experimenting with alternative tools to generation the documentation. If we would ever switch from Doxygen to an alternative, `doxy.md` would look pretty odd.

kfessel, 2025-03-23:
thanks for the input - i will close that PR.

closed 2025-04-01 by mguetschow