RIOT-OS/RIOT #21298: Buildsystem: Makefile of Common Board Directories never called without adding `DIRS`

crasbe, 2025-03-14:
#### Description

During the development of #21281 I noticed that the common board directories that have .c files, explicitly require that the common directory is added to the `DIRS` variable in the board's `Makefile`.

This is for example the case for the `particle-*` boards, that include the `boards/common/particle-mesh` folder:
https://github.com/RIOT-OS/RIOT/blob/685acb9a350de3d45eb73969b083efce18b1ba9b/boards/particle-xenon/Makefile#L1-L5

Even though the `boards_common_particle_mesh` module defined by the `particle-mesh` common board is added in the `Makefile.dep`, the `boards/common/particle-mesh/Makefile` will never be called without the `DIRS +=` definition.
https://github.com/RIOT-OS/RIOT/blob/685acb9a350de3d45eb73969b083efce18b1ba9b/boards/particle-xenon/Makefile.dep#L1-L3

#### Steps to reproduce the issue

Compile an application of your choice (for example `tests/sys/shell`) for the `particle-xenon` and observe the compile output. It should include `boards/common/particle-mesh`:
```
~/RIOTstuff/riot-ada-bootloader/RIOT$ BOARD=particle-xenon make -C tests/sys/shell
make: Entering directory '/home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell'
Building application "tests_shell" for "particle-xenon" with CPU "nrf52".

"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/pkg/cmsis/
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/boards/common/init
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/boards/particle-xenon
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/boards/common/particle-mesh
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/core
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/core/lib
...
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/sys/usb/usbus
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/sys/usb/usbus/cdc/acm
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/sys/usb_board_reset
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/sys/ztimer
   text    data     bss     dec     hex filename
  23816     128    4696   28640    6fe0 /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell/bin/particle-xenon/tests_shell.elf
make: Leaving directory '/home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell'
```

Comment out line 3 of the `boards/particle-xenon/Makefile` (the `DIRS` line) and observe the output again:
```
~/RIOTstuff/riot-ada-bootloader/RIOT$ BOARD=particle-xenon make -C tests/sys/shell
make: Entering directory '/home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell'
Building application "tests_shell" for "particle-xenon" with CPU "nrf52".

"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/pkg/cmsis/
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/boards/common/init
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/boards/particle-xenon
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/core
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/core/lib
...
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/sys/usb/usbus
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/sys/usb/usbus/cdc/acm
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/sys/usb_board_reset
"make" -C /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/sys/ztimer
   text    data     bss     dec     hex filename
  23816     128    4696   28640    6fe0 /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell/bin/particle-xenon/tests_shell.elf
make: Leaving directory '/home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell'
```

The `boards/common/particle-mesh` folder is not included anymore and no error is generated.

#### Expected results

The build system should include the `boards/common/particle-mesh` directory if the `boards_common_particle-mesh` module is used.

The `particle-*` boards are just an example here, the same is true for boards that use the `samdx1-arduino-bootloader`: https://github.com/search?q=repo%3ARIOT-OS%2FRIOT+boards_common_samdx1-arduino-bootloader&type=code

I'm not sure where to look and what to possibly change in the build system, so hints would be very welcome.

#### Actual results

See above.

#### Versions
This happens with the latest RIOT master.

crasbe, 2025-03-14:
Another way to check is the `info-build` target:

<details><summary>Unchanged master:</summary>

```
~/RIOTstuff/riot-ada-bootloader/RIOT$ BOARD=particle-xenon make -C tests/sys/shell info-build
make: Entering directory '/home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell'
APPLICATION: tests_shell
APPDIR:      /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell

supported boards:
adafruit-clue adafruit-grand-central-m4-express adafruit-itsybitsy-m4 adafruit-itsybitsy-nrf52 adafruit-metro-m4-express adafruit-pybadge arduino-mkr1000 arduino-mkrfox1200 arduino-mkrwan1300 arduino-mkrzero arduino-nano-33-ble arduino-nano-33-ble-sense arduino-nano-33-iot bastwan blackpill-stm32f103cb bluepill-stm32f103cb feather-m0 feather-m0-lora feather-m0-wifi feather-nrf52840 feather-nrf52840-sense nrf52840-mdk-dongle nrf52840dongle particle-argon particle-boron particle-xenon seeedstudio-xiao-nrf52840 seeeduino_xiao sensebox_samd21 serpente sodaq-autonomo sodaq-explorer sodaq-one sodaq-sara-aff sodaq-sara-sff wemos-zero yarm

BOARD:   particle-xenon
CPU:     nrf52

RIOTBASE:    /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT
BOARDDIR:    /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/boards/particle-xenon
EXTERNAL_BOARD_DIRS:
RIOTCPU:     /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/cpu
RIOTPKG:     /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/pkg

DEFAULT_MODULE: auto_init auto_init_usbus auto_init_ztimer board board_common_init core core_init core_lib core_msg core_panic core_thread cortexm_fpu cpu libc newlib_nano periph_init periph_init_cpuid periph_init_gpio periph_init_led0 periph_init_led1 periph_init_led2 periph_init_led3 periph_init_led4 periph_init_led5 periph_init_led6 periph_init_led7 periph_init_leds periph_init_pm periph_init_rtt periph_init_usbdev periph_init_usbdev_clk sys test_utils_interactive_sync test_utils_interactive_sync_shell test_utils_print_stack_usage ztimer_init
DISABLE_MODULE:
USEMODULE:      app_metadata boards_common_particle_mesh core_thread_flags cortexm_common cortexm_common_periph cpu_common div event fmt frac isrpipe luid malloc_thread_safe newlib newlib_syscalls_default nrf52_vectors nrf5x_common_periph periph periph_common periph_cpuid periph_gpio periph_gpio_ll_disconnect periph_gpio_ll_input_pull_down periph_gpio_ll_input_pull_up periph_gpio_ll_irq_edge_triggered_both periph_gpio_ll_irq_unmask periph_gpio_ll_open_drain periph_gpio_ll_open_drain_pull_up periph_gpio_ll_open_source periph_gpio_ll_open_source_pull_down periph_pm periph_rtt periph_usbdev periph_usbdev_clk preprocessor preprocessor_successor ps shell shell_builtin_cmd_help_json shell_cmd_app_metadata shell_cmd_pm shell_cmd_ps shell_cmd_sys shell_cmds shell_cmds_default stdin stdio stdio_available stdio_cdc_acm tsrb usb_board_reset usbus usbus_cdc_acm vdd_lc_filter_reg1 ztimer ztimer_convert ztimer_convert_frac ztimer_convert_shift ztimer_core ztimer_extend ztimer_msec ztimer_periph_rtt

ELFFILE: /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell/bin/particle-xenon/tests_shell.elf
HEXFILE: /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell/bin/particle-xenon/tests_shell.hex
BINFILE: /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell/bin/particle-xenon/tests_shell.bin
FLASHFILE: /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/tests/sys/shell/bin/particle-xenon/tests_shell.hex

FEATURES_USED:
         arch_32bit arch_arm cortexm_fpu cpu_core_cortexm highlevel_stdio newlib no_idle_thread periph_cpuid periph_gpio periph_gpio_ll_disconnect periph_gpio_ll_input_pull_down periph_gpio_ll_input_pull_up periph_gpio_ll_irq_edge_triggered_both periph_gpio_ll_irq_unmask periph_gpio_ll_open_drain periph_gpio_ll_open_drain_pull_up periph_gpio_ll_open_source periph_gpio_ll_open_source_pull_down periph_pm periph_rtt periph_usbdev vdd_lc_filter_reg1
FEATURES_REQUIRED:
         arch_32bit arch_arm cpu_core_cortexm highlevel_stdio periph_cpuid periph_gpio periph_pm periph_rtt periph_usbdev
FEATURES_REQUIRED_ANY:
         newlib|picolibc
FEATURES_OPTIONAL_ONLY (optional that are not required, strictly "nice to have"):
         cortexm_fpu no_idle_thread periph_gpio_ll_disconnect periph_gpio_ll_input_pull_down periph_gpio_ll_input_pull_keep periph_gpio_ll_input_pull_up periph_gpio_ll_irq_edge_triggered_both periph_gpio_ll_irq_level_triggered_high periph_gpio_ll_irq_level_triggered_low periph_gpio_ll_irq_unmask periph_gpio_ll_open_drain periph_gpio_ll_open_drain_pull_up periph_gpio_ll_open_source periph_gpio_ll_open_source_pull_down periph_gpio_ll_switch_dir vdd_lc_filter_reg0 vdd_lc_filter_reg1
FEATURES_OPTIONAL_MISSING (missing optional features):
         periph_gpio_ll_input_pull_keep periph_gpio_ll_irq_level_triggered_high periph_gpio_ll_irq_level_triggered_low periph_gpio_ll_switch_dir vdd_lc_filter_reg0
FEATURES_PROVIDED (by the board or USEMODULE'd drivers):
         arch_32bit arch_arm ble_adv_ext ble_nimble ble_nimble_netif ble_phy_2mbit ble_phy_coded board_bat_voltage cortexm_fpu cortexm_mpu cortexm_svc cpp cpu_check_address cpu_core_cortexm cpu_nrf52 dbgpin highlevel_stdio libstdcpp netif netif_openwsn newlib no_idle_thread periph_adc periph_cipher_aes_128_cbc periph_cipher_chacha20 periph_cpuid periph_cryptocell_310 periph_ecc_ed25519 periph_ecc_p192r1 periph_ecc_p256r1 periph_flashpage periph_flashpage_aux periph_flashpage_in_address_space periph_flashpage_pagewise periph_gpio periph_gpio_irq periph_gpio_ll periph_gpio_ll_disconnect periph_gpio_ll_input_pull_down pe [... cut ...] msec ztimer_no_periph_rtt ztimer_periph_timer
        /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/boards/common/samdx1-arduino-bootloader/Makefile.dep

        /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/boards/common/samdx1-arduino-bootloader/Makefile.dep

        /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/boards/common/samdx1-arduino-bootloader/Makefile.dep

        /home/cbuec/RIOTstuff/riot-ada-bootloader/RIOT/boards/common/samdx1-arduino-bootloader/Makefile.include
```

</details>

mguetschow, 2025-03-14:
Thanks for opening the issue! I've also just now experienced this behavior while working on #21242. Would probably need some more digging into the build system to be fixed.

crasbe, 2025-03-14:
I went through the debug log (`BOARD=feather-nrf52840 make -d -n -C tests/sys/shell` on my PR) to see how the current "process" of building the `common/boards/...` is and this is the place where it's called:

https://github.com/RIOT-OS/RIOT/blob/f4d74ff9dd158ac32792c12dbc6728a425849c84/Makefile.base#L30-L31

Looking at the other targets that start with `ALL--` in the log, this is generally the mechanism for the `common` folders as well and used by other architectures as well:
https://github.com/RIOT-OS/RIOT/blob/f4d74ff9dd158ac32792c12dbc6728a425849c84/cpu/sam3/Makefile#L4-L5

https://github.com/RIOT-OS/RIOT/blob/f4d74ff9dd158ac32792c12dbc6728a425849c84/boards/hip-badge/Makefile#L3

https://github.com/RIOT-OS/RIOT/blob/f4d74ff9dd158ac32792c12dbc6728a425849c84/cpu/arm7tdmi_gba/Makefile#L5

and various others.

<hr/>

Considering this is a *giant* can of worms again, I would like to postpone this and not make it a requirement for #21281.

mguetschow, 2025-03-14:
> Considering this is a _giant_ can of worms again, I would like to postpone this and not make it a requirement for [#21281](https://github.com/RIOT-OS/RIOT/pull/21281).

Sure, fine with me. At least we now have a tracking issue :)

crasbe, 2025-03-14:
Apparently using the `DIRS` variable was born by coincidence, or just by the way the change to the `boards/common` folder was done in this PR: #8058

There never was a provision to fetch the `Makefile` of the subfolders in the `boards/common` folder.

closed 2025-03-27 by mguetschow