RIOT-OS/RIOT #21312: doxygen: CSS hack to hide doc breaks on older browsers

mguetschow, 2025-03-21:
#### Description
The fix proposed in #21273  does not work on older browsers such as Firefox 116 since it uses the relatively new `:has` CSS selector: https://caniuse.com/css-has

Not sure if there is a good replacement for it, maybe using less?

carl-tud, 2025-03-22:
tbh, i wouldn't even consider this an issue as 
a, Firefox 116 was released on August 1, 2023
b, `:has` is considered baseline as of 2023: https://developer.mozilla.org/en-US/docs/Web/CSS/:has thanks to Interop 2023
c, with Interop 2024 and Interop 2025 in mind, I'd like to doubt there's going to be any major browser that's not going to support `:has` in the future

crasbe, 2025-03-23:
I'm not sure if it makes sense to make a hack even hackier for browsers that nobody should use anymore. The issue will be pretty much solved by #21300 anyways.

It would be possible to hack it with JavaScript. This is what a friendly AI language model suggested.

In the `doc/doxygen/src/css/riot.css` file:
```css
@supports selector(:has(*)) {
    li:has(> .item a[class^="md_"][class$="doc.html"]),
    tr:has(> .entry a[href^="md_"][href$="doc.html"]) {
        display: none;
    }
}
@supports not selector(:has(*)) {
    /* :has() is not supported */
    body::after {
        content: "fallback";
        display: none;
    }
}
```

In the `doc/doxygen/header.html`:
```html
<head>
...head stuff...
<script>
    document.addEventListener("DOMContentLoaded", function () {
        if (getComputedStyle(document.body, "::after").content.includes("fallback")) {
            document.querySelectorAll("li .item a[class^='md_'][class$='doc.html']").forEach(a => {
                a.closest("li").style.display = "none";
            });

            document.querySelectorAll("tr .entry a[href^='md_'][href$='doc.html']").forEach(a => {
                a.closest("tr").style.display = "none";
            });
        }
    });
</script>
...more head stuff
</head>
```

But having a hack for old and new browsers each would be kind of pointless and it would be possible to just use JS in all cases:
```html
<head>
...head stuff...
<script>
    document.addEventListener("DOMContentLoaded", function () {
        document.querySelectorAll("li .item a[class^='md_'][class$='doc.html']").forEach(a => {
            a.closest("li").style.display = "none";
        });

        document.querySelectorAll("tr .entry a[href^='md_'][href$='doc.html']").forEach(a => {
            a.closest("tr").style.display = "none";
        });
    });
</script>
...more head stuff...
</head>
```

This is untested.

crasbe, 2025-03-23:
Instead of hiding the elements, JS could totally remove them entirely as well:
```
<script>
    document.addEventListener("DOMContentLoaded", function () {
        document.querySelectorAll("li .item a[class^='md_'][class$='doc.html']").forEach(a => {
            a.closest("li")?.remove();
        });

        document.querySelectorAll("tr .entry a[href^='md_'][href$='doc.html']").forEach(a => {
            a.closest("tr")?.remove();
        });
    });
</script>
```