RIOT-OS/RIOT #21315: Question: Why don't we use `__BEGIN_DECLS` and `__END_DECLS` in headers?

carl-tud, 2025-03-22:
hi.

maybe this is a not-so-clever-this-is-completely-obvious question, but why don't we use `__BEGIN_DECLS` and `__END_DECLS` in headers instead of repeating

```c
#ifdef __cplusplus
extern "C" {
#endif

#ifdef __cplusplus
}
#endif
```

every time?

crasbe, 2025-03-23:
Personally I've heard of those macros for the first time. If that's the case for other contributors/maintainers as well, that might be a reason why the "old" style is still used. Also it would be a huge task to change all the relevant header files.

Apparently `__BEGIN_DECLS`/`__END_DECLS` is considered bad style now though and `BEGIN_C_DECLS`/`END_C_DECLS` should be used: https://www.gnu.org/software/libtool/manual/html_node/C-header-files.html

Personally, I don't have an opinion on what's better.

mguetschow, 2025-03-24:
It seems this is not some compiler built-in feature, but just a convention that some (apparently few, according to a quick search online) projects have adopted. I'm kind of in favor of replacing 6 lines of boilerplate with 2 lines of boilerplate, but I don't have a strong opinion either.

crasbe, 2026-10-06:
I don't see the need for another big migration job to save four lines of code for each file by introducing this. Also, this would also have to be defined somewhere and that somewhere has to be included everywhere, so we're down to three or two lines saved.

The discussion didn't really take off either, so I'll close it as not planned.

closed 2026-10-06 by crasbe