RIOT-OS/RIOT #21317: WiFi not working on ESP32-S3-FH4R2

mrninhvn, 2025-03-24:
#### Description
WiFi interface not working on ESP32-S3-FH4R2 (Waveshare ESP32-S3-Zero board), chip get hot and brownout reset.

#### Steps to reproduce the issue
Running `gnrc_border_router` or `paho-mqtt` example on Waveshare ESP32-S3-Zero board.

#### Expected results
WiFi interface up and running, connected to WiFi. Example working fine on ESP32-S3-WROOM module.

#### Actual results
```
2025-03-24 09:34:57,276 # Connect to serial port /dev/ttyACM0
2025-03-24 09:34:58,278 # Serial port disconnected, waiting to get reconnected...
Welcome to pyterm!
Type '/exit' to exit.
2025-03-24 09:34:59,278 # Serial port disconnected, waiting to get reconnected...
2025-03-24 09:35:00,279 # Try to reconnect to /dev/ttyACM0 again...
2025-03-24 09:35:00,283 # Reconnected to serial port /dev/ttyACM0
2025-03-24 09:35:00,910 # DHCPv6: Selecting interface 10 as upstream
2025-03-24 09:35:01,134 # Serial port disconnected, waiting to get reconnected...
2025-03-24 09:35:02,136 # Try to reconnect to /dev/ttyACM0 again...
2025-03-24 09:35:02,142 # Reconnected to serial port /dev/ttyACM0
2025-03-24 09:35:02,926 # Serial port disconnected, waiting to get reconnected...
2025-03-24 09:35:03,927 # Try to reconnect to /dev/ttyACM0 again...
2025-03-24 09:35:03,932 # Reconnected to serial port /dev/ttyACM0
2025-03-24 09:35:04,700 # DHCPv6: Selecting interface 10 as upstreamDHCPv6: Selecting interface 10 as upstream
2025-03-24 09:35:04,718 # Serial port disconnected, waiting to get reconnected...
2025-03-24 09:35:05,718 # Try to reconnect to /dev/ttyACM0 again...
2025-03-24 09:35:05,723 # Reconnected to serial port /dev/ttyACM0
2025-03-24 09:35:06,766 # Serial port disconnected, waiting to get reconnected...
2025-03-24 09:35:07,766 # Try to reconnect to /dev/ttyACM0 again...
```

After add extra power on 3.3V bus, firmware can be launched but can't connect to WiFi, chip is very hot to touch.
```
2025-03-24 09:43:13,327 # Pro cpu up.
2025-03-24 09:43:13,328 # Single core mode
2025-03-24 09:43:13,386 # Initializing. RAM available for dynamic allocation:
2025-03-24 09:43:13,386 # At 3FC9BCA8 len 00044358 (272 KiB): D/IRAM
2025-03-24 09:43:13,386 # At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
2025-03-24 09:43:13,391 # At 3FCF0000 len 00008000 (32 KiB): DRAM
2025-03-24 09:43:13,402 # 
2025-03-24 09:43:13,521 # system_wdt_init TIMERG0 wdtconfig0=b81f8000 wdtconfig1=00500000 main(): This is RIOT! (Version: 2025.01)
2025-03-24 09:43:13,521 # Running mqtt paho example. Type help for commands info
> 2025-03-24 09:43:16,563 # WiFi disconnected from ssid ******, reason 2 (AUTH_EXPIRE)
2025-03-24 09:43:18,613 # WiFi disconnected from ssid ********, reason 205 (CONNECTION_FAIL)
2025-03-24 09:43:21,665 # WiFi disconnected from ssid ********, reason 2 (AUTH_EXPIRE)
2025-03-24 09:43:23,721 # WiFi disconnected from ssid ********, reason 205 (CONNECTION_FAIL)
2025-03-24 09:43:26,774 # WiFi disconnected from ssid ********, reason 2 (AUTH_EXPIRE)
2025-03-24 09:43:28,820 # WiFi disconnected from ssid ********, reason 205 (CONNECTION_FAIL)
```

#### Versions
Operating system: Linux
Build environment: GCC 14.2.1 20250207, clang version 19.1.7 xtensa-esp32s3-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0.

mcr, 2025-03-25:
> After add extra power on 3.3V bus, firmware can be launched but can't connect to WiFi, chip is very hot to touch.

how many amps did you start with, and how much did you add?

gschorcht, 2026-01-14:
I have ordered a Waveshare ESP32-S3-Zero board which works perfectly with WiFi. It takes only about 95 mA with active WiFi connection. I guess your hardware is broken.

<details>

```
2026-01-14 19:54:42,770 # Serial port disconnected, waiting to get reconnected...
2026-01-14 19:54:43,771 # Try to reconnect to /dev/ttyACM0 again...
2026-01-14 19:54:43,772 # Reconnected to serial port /dev/ttyACM0
2026-01-14 19:54:43,772 # 
> 2026-01-14 19:54:43,773 # 
> 2026-01-14 19:54:45,713 # main(): This isWiFi connected to ssid BSHS1, channel 11
> ifconfig
2026-01-14 19:55:19,774 # ifconfig
2026-01-14 19:55:19,776 # Iface  9  HWaddr: 9C:13:9E:F4:0C:18  Channel: 11  Link: up 
2026-01-14 19:55:19,777 #           L2-PDU:1500  MTU:1492  HL:255  RTR  
2026-01-14 19:55:19,778 #           RTR_ADV  
2026-01-14 19:55:19,779 #           Source address length: 6
2026-01-14 19:55:19,779 #           Link type: wireless
2026-01-14 19:55:19,780 #           inet6 addr: fe80::9e13:9eff:fef4:c18  scope: link  VAL
2026-01-14 19:55:19,782 #           inet6 addr: 2003:c1:e73d:26f7:9e13:9eff:fef4:c18  scope: global  VAL
2026-01-14 19:55:19,782 #           inet6 group: ff02::1a
2026-01-14 19:55:19,783 #           inet6 group: ff02::2
2026-01-14 19:55:19,784 #           inet6 group: ff02::1
2026-01-14 19:55:19,785 #           inet6 group: ff02::1:fff4:c18
2026-01-14 19:55:19,785 #           
2026-01-14 19:55:19,786 #           Statistics for Layer 2
2026-01-14 19:55:19,787 #             RX packets 179  bytes 68724
2026-01-14 19:55:19,788 #             TX packets 6 (Multicast: 4)  bytes 587
2026-01-14 19:55:19,789 #             TX succeeded 6 errors 0
2026-01-14 19:55:19,790 #           Statistics for IPv6
2026-01-14 19:55:19,791 #             RX packets 136  bytes 53551
2026-01-14 19:55:19,793 #             TX packets 6 (Multicast: 4)  bytes 503
2026-01-14 19:55:19,793 #             TX succeeded 6 errors 0
2026-01-14 19:55:19,793 # 
> ping fe80::c5b4:388a:60e5:d4ea
2026-01-14 19:56:10,518 # ping fe80::c5b4:388a:60e5:d4ea
2026-01-14 19:56:10,527 # 12 bytes from fe80::c5b4:388a:60e5:d4ea%9: icmp_seq=0 ttl=64 time=6.957 ms
2026-01-14 19:56:11,530 # 12 bytes from fe80::c5b4:388a:60e5:d4ea%9: icmp_seq=1 ttl=64 time=9.118 ms
2026-01-14 19:56:12,524 # 12 bytes from fe80::c5b4:388a:60e5:d4ea%9: icmp_seq=2 ttl=64 time=3.065 ms
2026-01-14 19:56:12,524 # 
2026-01-14 19:56:12,526 # --- fe80::c5b4:388a:60e5:d4ea PING statistics ---
2026-01-14 19:56:12,528 # 3 packets transmitted, 3 packets received, 0% packet loss
2026-01-14 19:56:12,530 # round-trip min/avg/max = 3.065/6.380/9.118 ms
```

</details>

closed 2026-01-14 by gschorcht