RIOT-OS/RIOT #21335: tracking: fixing order of doxygen tags and header guards

KSKNico, 2025-03-28:
### Description

Clang-format requires the header guard #ifndef #define HEADER_GUARD_H to be outside of the doxygen documentation, that means outside the @{ ... @} block. If this is not done, then clang-format will indent all preprocessor directives after the #ifndef.
The problem is that the majority of RIOT's header files don't adhere to this order. They either have the header guard inside the doxygen block or the header guard starts correctly outside the doxygen block but the #endif directive is still inside the doxygen documentation.
To fix this inconsistency, I have written a python script that recursively scans all header files in a directory and rearranges the parts, so that the correct order is established.

### Fixed directories
- [ ] drivers #21325
- [ ] sys
- [ ] cpu
- [ ] core
- [ ] boards
- [ ] examples
- [ ] pkg
- [ ] tests

### Testing procedure

Use my python script and change the path variable 'RIOT_PATH' in the script to whatever directory you want to recursively fix the header files.

The script can be found [here](https://gist.github.com/KSKNico/0c088d957ff170739e7cb9d0a46fe582)
Issues/PRs references

### Application

This is based on the fix in https://github.com/RIOT-OS/RIOT/pull/20905 addressed by @mguetschow but with the added benefit of keeping the blocks balanced.

mguetschow, 2025-03-28:
The order should also be fixed in https://github.com/aabadie/riot-generator

crasbe, 2025-03-31:
There is some documentation in the `dist/tools/ci` folder https://github.com/RIOT-OS/RIOT/tree/master/dist/tools/ci and you can look at how the check-scripts are called by `dist/tools/ci/static_tests.sh` and how they operate.

You don't necessarily have to look into CI for that, the script is executed when you call `make static-test` and that's exactly what the CI system does as well.

benpicco, 2025-04-03:
Should we just move to `#pragma once` instead?

Enoch247, 2025-04-03:
That would be fine by me.

mguetschow, 2025-04-04:
We had that discussion at [VMA 2024.11](https://forum.riot-os.org/t/notes-virtual-maintainer-assembly-vma-2024-11/4428) and had general consensus on switching to `#pragma once`, while having some reservation about the necessary migration work to keep in-tree consistency. Since we are now anyways touching a lot of header files, there was consensus at today's maintainer weekly that we can just as well use that work to switch over to `#pragma once`.

Enoch247, 2025-04-06:
Awesome! I have updated #21344 to make use of the `#pragma once`.

crasbe, 2025-04-06:
For `#pragma once`, the `dist/tools/headerguards/headerguards.py` script has to be modified, otherwise the static tests fail:

[image]

JerelJr, 2025-04-11:
Hello, I am new to the RIOT community and would like to help fix the headers.  I can start in the `core` folder if no one is working on that already. I have a [modified version](https://gist.github.com/JerelJr/e0f2c4fc9d0c42cdbd60063a0cd2595d) of @KSKNico's script to replace the header guard entirely with `#pragma once`. Are there any considerations I should be aware of before I begin?

KSKNico, 2025-05-20:
@mguetschow @crasbe Should vendor specific header files also be modified to use `#pragma once`?

mguetschow, 2025-05-20:
If those are vendor-provided (aka just copied to the RIOT tree from somewhere else) then I wouldn't change them. But all other header files belonging to RIOT which happen to be for a specific vendor should be modified of course.

crasbe, 2025-05-21:
@KSKNico could you extend your script to check that if there is an empty line before *and* after the `#ifndef...#define...#endif` that one of these lines is also removed? That would avoid introducing empty line static errors.

See https://github.com/RIOT-OS/RIOT/pull/21501#issuecomment-2899145625 and #21502.

KSKNico, 2025-05-22:
@crasbe I updated the script to account for that. Should I open another pull request?

closed 2025-10-17 by crasbe

mguetschow, 2025-10-19:
Thanks everyone for pushing and getting this through! 🎉