RIOT-OS/RIOT #21336: nucleo-f446re problem with using the led and spi at the same time

tanneberger, 2025-03-28:
#### Description
@edwardlee and I were trying to program for the `nucleo-f446re` and discovered that when we added the `USEMODULE += periph_spi` the LED macros stopped working. 

#### Steps to reproduce the issue
```c
#include <stdio.h>

#include "clk.h"
#include "board.h"
#include "periph_conf.h"
#include "timex.h"
#include "ztimer.h"

int main(void)
{
    while (1) {
        ztimer_sleep(ZTIMER_USEC, 1 * US_PER_SEC);
        LED0_TOGGLE;
    }

    return 0;
}
```

Now just add `USEMODULE += periph_spi` to your Makefile and this program stops working / led stops blinking. 

#### Expected results

a blinking led

#### Actual results

no blinking led

bergzand, 2025-03-29:
The nucleo boards share the LED GPIO with one of the SPI pins. So either the LED macro stops working (and starts working as SPI activity indicator), or one of the SPI busses stops working.

See https://github.com/RIOT-OS/RIOT/issues/6501 for the original issue around this.

Conclusion there was that it is easier to detect issues with the LED not blinking than finding an issue with a non-working SPI bus. Hence the LED gets disabled and not the SPI bus.

crasbe, 2025-03-29:
@bergzand Is that the case for all Nucleos? That would probably be something worth mentioning in the Doxygen documentation.

bergzand, 2025-03-29:
> [@bergzand](https://github.com/bergzand) Is that the case for all Nucleos? That would probably be something worth mentioning in the Doxygen documentation.

If we can trust the makefile, all nucleo64 boards have this issue.

tanneberger, 2025-03-29:
Okay thanks, this answers my question ... ST is frustrating.

closed 2025-03-29 by tanneberger

crasbe, 2025-03-29:
I created a PR to add a dedicated compiler warning to warn the user when this happens and in another PR I'll add some documentation about this.
While this won't solve your original issue, it might save some headaches in the future and help make RIOT more friendly.

Thank you for raising this issue :)