RIOT-OS/RIOT #21490: doc/doxygen: broken formulas

crasbe, 2025-05-14:
#### Description

The rendered formulas in our documentation are broken. For example: https://doc.riot-os.org/group__drivers__periph__i2c.html

[image]

`make doc` with Doxygen 1.12.0 has the following error messages (among the billion others):
```
error: Problems running latex. Check your installation or look for typos in _formulas.tex and check _formulas.log!
error: Problems running latex. Check your installation or look for typos in _formulas_dark.tex and check _formulas_dark.log!
```

The `doc/doxygen/html/_formulas.log` contains this:
```
cbuec@W11nMate:~/RIOTstuff/riot-gettingstarted/RIOT$ cat doc/doxygen/html/_formulas.log
This is pdfTeX, Version 3.141592653-2.6-1.40.22 (TeX Live 2022/dev/Debian) (preloaded format=latex 2025.3.7)  14 MAY 2025 23:48
entering extended mode
 restricted \write18 enabled.
 %&-line parsing enabled.
**_formulas
(./_formulas.tex
LaTeX2e <2021-11-15> patch level 1
L3 programming layer <2022-01-21> (/usr/share/texlive/texmf-dist/tex/latex/base
/article.cls
Document Class: article 2021/10/04 v1.4n Standard LaTeX document class
(/usr/share/texlive/texmf-dist/tex/latex/base/size10.clo
File: size10.clo 2021/10/04 v1.4n Standard LaTeX file (size option)
)
\c@part=\count185
\c@section=\count186
\c@subsection=\count187
\c@subsubsection=\count188
\c@paragraph=\count189
\c@subparagraph=\count190
\c@figure=\count191
\c@table=\count192
\abovecaptionskip=\skip47
\belowcaptionskip=\skip48
\bibindent=\dimen138
) (/usr/share/texlive/texmf-dist/tex/generic/iftex/iftex.sty
Package: iftex 2020/03/06 v1.0d TeX engine tests
) (/usr/share/texlive/texmf-dist/tex/latex/base/ifthen.sty
Package: ifthen 2020/11/24 v1.1c Standard LaTeX ifthen package (DPC)
) (/usr/share/texlive/texmf-dist/tex/latex/graphics/epsfig.sty
Package: epsfig 2017/06/25 v1.7b (e)psfig emulation (SPQR)
(/usr/share/texlive/texmf-dist/tex/latex/graphics/graphicx.sty
Package: graphicx 2021/09/16 v1.2d Enhanced LaTeX Graphics (DPC,SPQR)
(/usr/share/texlive/texmf-dist/tex/latex/graphics/keyval.sty
Package: keyval 2014/10/28 v1.15 key=value parser (DPC)
\KV@toks@=\toks16
) (/usr/share/texlive/texmf-dist/tex/latex/graphics/graphics.sty
Package: graphics 2021/03/04 v1.4d Standard LaTeX Graphics (DPC,SPQR)
(/usr/share/texlive/texmf-dist/tex/latex/graphics/trig.sty
Package: trig 2021/08/11 v1.11 sin cos tan (DPC)
) (/usr/share/texlive/texmf-dist/tex/latex/graphics-cfg/graphics.cfg
File: graphics.cfg 2016/06/04 v1.11 sample graphics configuration
)
Package graphics Info: Driver file: dvips.def on input line 107.
(/usr/share/texlive/texmf-dist/tex/latex/graphics-def/dvips.def
File: dvips.def 2017/06/20 v3.1d Graphics/color driver for dvips
))
\Gin@req@height=\dimen139
\Gin@req@width=\dimen140
)
\epsfxsize=\dimen141
\epsfysize=\dimen142
) (/usr/share/texlive/texmf-dist/tex/latex/base/inputenc.sty
Package: inputenc 2021/02/14 v1.3d Input encoding file
\inpenc@prehook=\toks17
\inpenc@posthook=\toks18
[... 156 lines cut ...]
  ruby-xmlrpc ruby3.0 rubygems-integration tcl tcl8.6 tex-gyre texlive-fonts-recommended texlive-pictures
  texlive-plain-generic tipa tk tk8.6 zip
Suggested packages:
  libavalon-framework-java libcommons-logging-java-doc libexcalibur-logkit-java liblog4j1.2-java ri ruby-dev bundler
  tcl-tclreadline texlive-fonts-recommended-doc icc-profiles libfile-which-perl libspreadsheet-parseexcel-perl
  texlive-latex-extra-doc dot2tex prerex texlive-pictures-doc vprerex default-jre-headless tipa-doc
The following NEW packages will be installed:
  fonts-lato fonts-texgyre libapache-pom-java libcommons-logging-java libcommons-parent-java libfontbox-java
  libpdfbox-java libruby3.0 libtk8.6 libxss1 preview-latex-style rake ruby ruby-net-telnet ruby-rubygems ruby-webrick
  ruby-xmlrpc ruby3.0 rubygems-integration tcl tcl8.6 tex-gyre texlive-fonts-recommended texlive-latex-extra
  texlive-pictures texlive-plain-generic tipa tk tk8.6 zip
0 upgraded, 30 newly installed, 0 to remove and 0 not upgraded.
Need to get 89.4 MB of archives.
After this operation, 301 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

<hr/>

So now the question is where to go from here... install 340MB of additional packages just to display some formulas?

The other alternative would be `MATHJAX` which does display the forumas correctly.
See: https://www.doxygen.nl/manual/formulas.html

I would like to get some comments about which direction to go here.

mguetschow, 2025-05-15:
> So now the question is where to go from here... install 340MB of additional packages just to display some formulas?
> 
> The other alternative would be `MATHJAX` which does display the forumas correctly. See: [doxygen.nl/manual/formulas.html](https://www.doxygen.nl/manual/formulas.html)
> 
> I would like to get some comments about which direction to go here.

`MATHJAX` sounds like the way more straightforward option for me. However, it will increase the dependence on javascript at the client side, not sure if that is something we wish to avoid. Right now it seems the documentation is pretty much usable without javascript, just missing the navigation menu on the left (but accessible from the top).

But, given that using Mathjax would not worsen things for clients w/o javascript and fix the formulas for everyone else, hopefully at a much lower complexity than installing and running latex in the background, I'd go with that approach.

Would be nice to not load Mathjax from a third-party website though.

crasbe, 2025-05-15:
Fortunately MathJax can be embedded locally, so it would be possible to create another local `dist/tool` script that the Doxygen Makefile depends on...

This however will break the doc preview with Doxygen 1.9.1 even more, since it does not support MathJax v3.
Time to get back to the update thing I guess...