RIOT-OS/RIOT #21515: tracking: replacing old license headers with SPDX headers

KSKNico, 2025-05-27:
### Description

Currently, RIOT OS uses an outdated and not machine-friendly way of declaring the license and the copyright at the start of the file. The exact layout of the copyright headers is also not consistent. For this reason the [SPDX standard](https://spdx.dev/) provides a consistent way to declare license and copyright information that can easily be read by humans and programs such as [REUSE](https://reuse.software/).

Basically, all .c and .h files will have to be converted except for the ones provided by vendors.

Before:
```
/*
 * Copyright (C) 2016 Inria
 * Copyright (C) 2018 Freie Universität Berlin
 *
 * This file is subject to the terms and conditions of the GNU Lesser
 * General Public License v2.1. See the file LICENSE in the top level
 * directory for more details.
 */
```

After (with the SPDX standard for headers):
```
/*
 * SPDX-FileCopyrightText: 2016 Inria
 * SPDX-FileCopyrightText: 2018 Freie Universität Berlin
 * SPDX-License-Identifier: LGPL-2.1-only
 */
```

### Fixed directories
This issue will track the different parts of RIOT that have already been converted to the SPDX standard. Multiple PRs will have to be created.

### Testing procedure
The script can be found [here](https://gist.github.com/KSKNico/1e4dfd31c1d4eba4adf1c15222d3fc5d) but it is still WIP and only prints the new header format. It is able to convert 90% of all old license headers into the new SPDX format. Some edge cases will need to be fixed or manually adjusted.

### References
This issue was brought up by @mguetschow

KSKNico, 2025-06-02:
> IMO it would make sense to also adapt the static test to check for that license format: https://github.com/RIOT-OS/RIOT/blob/master/dist/tools/licenses/check.sh

What the CI script does at the moment is to check whether the file contains the correct string as found in the license directory. Now, SPDX offers parsers that can do just that by processing the header information. I don't know if we want to incorporate that into our CI.

N11cc00, 2025-07-08:
@krzysztof-cabaj The updated script can be found [here](<script src="https://gist.github.com/N11cc00/a6db0252168d0747b40523c65e27b377.js"></script>). Updating would need to be done not only for C files as also other files types have Copyright information.

krzysztof-cabaj, 2025-07-14:
I prepare additional script (which is based on one provided by @N11cc00) which converts licenses in Kconfig files.
Script can be found [here](https://gist.github.com/krzysztof-cabaj/08b68cf2554472cd66a6c2d52c37a659).

As moving licenses from comments to the SPDX format is one time work - I did not decide to provide one universal tool.
I tested it during moving stm32 boards ... and new PR show (or not) if I everything is OK.

N11cc00, 2025-07-14:
One thing to note here is that all licenses also need to be added to the LICENSES directory, otherwise tools like `reuse` are not able to resolve the license. Currently, the LICENSE LGPL-2.1 is not inside the LICENSES directory and it errors:
`../../../examples/networking/cord/cord_ep/main.c: missing license LGPL-2.1-only`

On top of that BSD-3-Clause and Apache-2.0 are also missing.

crasbe, 2025-07-28:
Can someone update the Coding Convention in our Wiki and explain the SPDX headers there?

https://github.com/RIOT-OS/RIOT/blob/master/CODING_CONVENTIONS.md#documentation

krzysztof-cabaj, 2025-08-07:
I prepared a new script which could be used for conversion of `doc.md` files.
The code is available [here](https://gist.github.com/krzysztof-cabaj/a42c4d7afd665e453107dbfb993d1fab).

mguetschow, 2026-09-29:
Is anyone aware of the failing static tests on `master` wrt the licence check (which apparently is executed twice for some reason?): https://github.com/RIOT-OS/RIOT/actions/runs/36534349214/job/109294888123

[image]

crasbe, 2026-09-29:
Yes, but finding the original licenses is a lot of work and nobody has done it so far.

The tests started failing when I fixed the license check, but that was known at the time: #22587 and #22598