RIOT-OS/RIOT #21520: Is there any board support for Texas Instruments cc1352p7-1 or cc1352p7

OmidKandelusy, 2025-06-02:
Unfortunately, there is no support for the CC1352P7-1 or CC1352P7.
I was wondering if there is any effort made for the support on this board?
Or is there any guid/manual for adding build support for other hardware platforms.
(I am wondering if there is any development already made for this board anywhere, any hints would be appreciated)

Thanks.

benpicco, 2025-06-23:
Do you mean [this one](https://doc.riot-os.org/group__boards__cc1352p__launchpad.html)?
There is still a WIP PR for the radio: https://github.com/RIOT-OS/RIOT/pull/18737

OmidKandelusy, 2025-07-01:
No, I already knew of that one, I meant cc1352p7 or cc1352p7-1 boards. Those are different chips.

closed 2025-09-14 by maribu