RIOT-OS/RIOT #21580: [Discussion] IPv6 neighbor discovery with BLE

elenaf9, 2025-07-03:
#### Description

While working on #21574, I noticed that the NIB neighbor cache is always empty when running IP-over-BLE.

The reason is that:
a) RPL control messages are mostly multicast (so just copied onto every unicast link)
b) L2 addresses for unicast are directly resolved from the IP address (reverse [RFC 7668 Section 3.2.2])

After some testing & digging through the code, I observed that (at least with BLE) generally no neighbor discovery from [RFC 4861]/ [RFC 6775] is happening:
- The only sent messages are Router Soliciations (RS), which never get a reply because the Router Advertisements (RA) are only sent if authoritative border routers are present. Nodes apparently don't need to know if other nodes are routers or not, they just use RPL DIO info for routing.
- Neighbor Solicitation messages are only sent when trying to resolve an address, which because of _b)_ never happens. And only then a neighbor cache entry would be created and maintained.

Still, in the docs it is stated that IP-over-BLE does support 6LoWPAN neighbor discovery:
https://github.com/RIOT-OS/RIOT/blob/f7bda59197fdeda59d0ec4464afd638c44d2178f/pkg/nimble/README.ipv6-over-ble.md?plain=1#L9-L10

My questions are:
- Is my understanding above correct?
- Why is a RA only sent if there are authoritative routers present? From a high level it kinda makes sense, especially because it is redundant with the RPL DIS/DIO messages, but neither [RFC 4861] nor [RFC 6775] mention anything about it.
- Does RPL implicitly obsolete neighbor discovery?
- Should we maintain a neighbor cache even if a node's L2 address can be resolved directly from the IP address?
  On one hand it might be be redundant because BLE is a connection-oriented link-layer where we know which nodes are reachable and which aren't. And we can perform address resolution directly based on [RFC 7668 Section 3.2.2].
 On the other hand is neighbor discovery specified in both IP-over-BLE RFCs ([RFC 7668] and [RFC 9159])

Sorry for the many questions, I am still new to IP-over-BLE and might be missing something fundamental 😄 

// cc @miri64 as author of #7479.

#### Motivation

As part of #21574, I need to be able to get the IPv6 address of a connected BLE node. Intuitively, this information should be present in the neighbor cache.
This kind of pushed me down the rabbit hole of trying to understand IPv6 neighbor discovery and its interplay with route-over routing protocols like RPL.

#### Related Issues

- #21574

[RFC 4861]: https://www.rfc-editor.org/rfc/rfc4861
[RFC 4944 Section 6]: https://www.rfc-editor.org/rfc/rfc4944#section-6
[RFC 7668]: https://www.rfc-editor.org/rfc/rfc7668
[RFC 7668 Section 3.2.2]: https://www.rfc-editor.org/rfc/rfc7668l#section-3.2.2
[RFC 6775]: https://www.rfc-editor.org/rfc/rfc6775
[RFC 9159]: https://www.rfc-editor.org/rfc/rfc9159

mcr, 2025-07-03:
Elena Frank ***@***.***> wrote:
    > b) L2 addresses for unicast are
    > directly resolved from the IP address (reverse [RFC 7668 Section
    > 3.2.2])

I didn't understand this statement at first.
While that would make sense when sending across one hop, across two hops,
you'd need a route, and I'm not sure if that would obsolete needing L2
address lookup.
I think that it ought to skip the ND step, but I think it ought to create a
neighbor cache entry anyway.

    > Does RPL implicitly obsolete neighbor discovery?

No.
It might obsolete router discoveries... this was hotly debated 15 years ago.
If all nodes speak RPL, the quite possibly, no RAs needed.
But, if there are leaf-nodes that do not speak RPL, then one needs RA and ND.

    > - Should we maintain a neighbor cache even if a node's L2 address can
    > be resolved directly from the IP address?  On one hand it might be be
    > redundant because BLE is a connection-oriented link-layer where we know
    > which nodes are reachable and which aren't. And we can perform address
    > resolution directly based on [RFC 7668 Section 3.2.2].  On the other
    > hand is neighbor discovery specified in both IP-over-BLE RFCs ([RFC
    > 7668] and [RFC 9159])

I thought BTLE 4.0+ mesh was less connection oriented.

elenaf9, 2025-07-09:
@mcr Thanks for the quick reply!

So I guess one possible solution would be to add during address resolution, even if we can resolve the BLE address directly from the IP address, still the creation of a NC entry and to start the probing process. 
Then we'd at least have entries for parents and children, and potentially also other nodes if we have sent a unicast message to them.

> I think that it ought to skip the ND step, but I think it ought to create a
neighbor cache entry anyway.

I am not sure what you mean by that. Aren't the two steps coupled with each other, in the sense that we create a NC entry, and then use the mechanism from the ND RFC to maintain that entry?

> b) L2 addresses for unicast are
>  directly resolved from the IP address (reverse [RFC 7668 Section
>  3.2.2])
>
> I didn't understand this statement at first.
> While that would make sense when sending [...]

Sorry, I probably wasn't clear. I was only referring to address lookup for single-hop/ direct messages to nodes that we are connected to.

> I thought BTLE 4.0+ mesh was less connection oriented.

My use-case right now targets only BLE multi-hop, not BLE mesh. BLE multi-hop also creates a mesh-topology network, but with connections between the individual nodes. I am using the `nimble_autoconnect` module for this.

elenaf9, 2025-07-09:
> > I think that it ought to skip the ND step, but I think it ought to create a
> neighbor cache entry anyway.
> 
> I am not sure what you mean by that. Aren't the two steps coupled with each other, in the sense that we create a NC entry, and then use the mechanism from the ND RFC to maintain that entry?

Ah, I guess what you meant is to use the BLE connection info to create the NC entry and set it's state, but don't do the ND probing. That would make sense.