RIOT-OS/RIOT #21733: Tutorial for hello_world incomplete/buggy

thoastbrot, 2025-09-20:
#### Description
Trying to follow the tutorial at https://guide.riot-os.org/rust_tutorials/create_project/, I stumble about various issues.

- it says to create a directory (hello_world) and change into it, then checkout RIOT. Few steps later, I create a crate, supposedly _that_ directory. Could have been streamlined: create crate, init git, checkout RIOT as submodule.
- I then should `make info-rust`, which complains that the $(RIOTBASE) is a directory(. Stop.)

#### Steps to reproduce the issue

Follow the guide. 

#### Expected results
Should lead to a buildable hello world project, so I have a stable base to continue exploring RIOT-OS.

#### Actual results
Stuck with errors I shouldn't be dealing with.

#### Versions

```
Operating System Environment
----------------------------
         Operating System: "Arch Linux" 
                   Kernel: Linux 6.16.7-arch1-1 x86_64 unknown
             System shell: GNU bash, Version 5.3.3(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, Version 5.3.3(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (GCC) 15.2.1 20250813
        arm-none-eabi-gcc: missing
                  avr-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
      riscv32-esp-elf-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp32s2-elf-gcc: missing
   xtensa-esp32s3-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 20.1.8

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
   riscv32-esp-elf-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp32s2-elf-newlib: missing
xtensa-esp32s3-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 4.1.1
                 cppcheck: missing
                  doxygen: 1.14.0
                      git: git version 2.51.0
                     make: GNU Make 4.4.1
                  openocd: missing
                   python: Python 3.13.7
                  python2: Python 2.7.18
                  python3: Python 3.13.7
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

AnnsAnns, 2025-09-21:
Hey, thank you for giving RIOT a try. 

Just to make sure everything is working as expected, have you tried the [example linked at the end of the guide](https://github.com/RIOT-OS/RIOT/tree/master/examples/lang_support/official/rust-hello-world)? You can simply run it by going into your RIOT folder at `RIOT/examples/lang_support/official/rust-hello-world` and executing the same commands as in the guide, e.g. `make flash`

thoastbrot, 2025-09-21:
Hey, thanks for the pointer. This ought to help.

But: had to jump through some more hoops. 
- cmake 4 is too new for c2rust. Installed cmake3 from AUR.
- while building c2rust 0.19.0 as described, c2rust-ast-exporter v0.19.0 fails to build. 

```
  Cloning into 'tinycbor_build'...
  HEAD is now at d393c16 Parser: fix reading it->extra on big endian when bytesNeeded == 1
  make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
  make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
  make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
  make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
  /home/thorsten/.cargo/git/checkouts/c2rust-2471865f8518dc65/b9339f4/c2rust-ast-exporter/src/AstExporter.cpp: In lambda function:
  /home/thorsten/.cargo/git/checkouts/c2rust-2471865f8518dc65/b9339f4/c2rust-ast-exporter/src/AstExporter.cpp:2208:60: error: no matching function for call to 'clang::FieldDecl::getBitWidthValue(clang::ASTContext&)'
   2208 |                                  array, D->getBitWidthValue(*this->Context));
        |                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
  /home/thorsten/.cargo/git/checkouts/c2rust-2471865f8518dc65/b9339f4/c2rust-ast-exporter/src/AstExporter.cpp:2208:60: note: there is 1 candidate
  In file included from /usr/include/clang/AST/ASTContext.h:21,
                   from /usr/include/clang/Frontend/ASTUnit.h:17,
                   from /usr/include/clang/Frontend/FrontendAction.h:23,
                   from /usr/include/clang/Frontend/FrontendActions.h:12,
                   from /home/thorsten/.cargo/git/checkouts/c2rust-2471865f8518dc65/b9339f4/c2rust-ast-exporter/src/AstExporter.cpp:15:
  /usr/include/clang/AST/Decl.h:3166:12: note: candidate 1: 'unsigned int clang::FieldDecl::getBitWidthValue() const'
   3166 |   unsigned getBitWidthValue() const;
        |            ^~~~~~~~~~~~~~~~
  /usr/include/clang/AST/Decl.h:3166:12: note: candidate expects 0 arguments, 1 provided
  make[3]: *** [CMakeFiles/clangAstExporter.dir/build.make:79: CMakeFiles/clangAstExporter.dir/AstExporter.cpp.o] Error 1
  make[2]: *** [CMakeFiles/Makefile2:421: CMakeFiles/clangAstExporter.dir/all] Error 2
  make[1]: *** [CMakeFiles/Makefile2:428: CMakeFiles/clangAstExporter.dir/rule] Error 2
  make: *** [Makefile:254: clangAstExporter] Error 2

  thread 'main' panicked at /home/thorsten/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 2

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
error: failed to compile `c2rust v0.19.0 (https://github.com/immunant/c2rust?tag=v0.19.0#b9339f42)`, intermediate artifacts can be found at `/tmp/cargo-installI3Bunz`.

```

From there on, googling doesn't help anymore. I assume some other component is too new.

This has basically nothing to do with the tutorial, but everyone with arch/recent dependencies will stumble over the same issues... what's the learning here? Build in docker?

AnnsAnns, 2025-09-22:
Thank you for investigating the issue further. I agree that esp. in this scenario dependency mismatches can be really annoying to debug, which is why the tutorial also recommends building in docker and even includes BUILD_IN_DOCKER=1 within the makefile.

I will try to make the point more noticeable within the flashing section though. I can confirm that building the rust example without docker also fails for similar reasons for me 😄

PR #21735 (2025-09-22): doc/guides/rust_tutorials: mention BUILD_IN_DOCKER more verbosely

### Contribution description

This should reduce potential errors as happened in #21733 and make it easier to understand when/how to build stuff within our docker container. 

### Testing procedure

`make doc-starlight`

### Issues/PRs references

This closes #21733
closed 2025-09-22 by crasbe